EttusResearch/uhd #493: Error building the CHDR dissectors

NicolasCaseau, 2021-09-24:
Hi, 

I'm trying to build the CHDR dissectors to use them in wireshark. Everything works fine until I try to run the `make` command :

```
~/work/uhd/tools/dissectors$ mkdir build 
~/work/uhd/tools/dissectors$ cd build/~/work/uhd/tools/dissectors$ mkdir build
~/work/uhd/tools/dissectors$ cd build/
~/work/uhd/tools/dissectors/build$ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for one of the modules 'glib-2.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
-- Configuring RFNoC support...
-- Enabling RFNoC support.
-- Override with -DENABLE_RFNOC=ON/OFF
-- Configuring Octoclock support...
-- Disabling Octoclock support.
-- Override with -DENABLE_OCTOCLOCK=ON/OFF
-- the and I have and issue when I run
-- Configuring ZPU support...
-- Disabling ZPU support.
-- Override with -DENABLE_ZPU=ON/OFF
-- ######################################################
-- # Ettus dissectors enabled
-- ######################################################
-- * RFNoC
-- ######################################################
-- # Ettus dissectors disabled
-- ######################################################
-- * Octoclock
-- * ZPU
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bluelab/work/uhd/tools/dissectors/build
~/work/uhd/tools/dissectors/build$ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glib-2.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
-- 
-- Configuring RFNoC support...
--   Enabling RFNoC support.
--   Override with -DENABLE_RFNOC=ON/OFF
-- 
-- Configuring Octoclock support...
--   Disabling Octoclock support.
--   Override with -DENABLE_OCTOCLOCK=ON/OFF
-- the and  I have and issue when I run
-- Configuring ZPU support...
--   Disabling ZPU support.
--   Override with -DENABLE_ZPU=ON/OFF
-- 
-- ######################################################
-- # Ettus dissectors enabled                            
-- ######################################################
--   * RFNoC
-- 
-- ######################################################
-- # Ettus dissectors disabled                           
-- ######################################################
--   * Octoclock
--   * ZPU
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bluelab/work/uhd/tools/dissectors/build

~/work/uhd/tools/dissectors/build$ make 
[ 14%] Generating /home/bluelab/work/uhd/tools/dissectors/build/epan/rfnoc/plugin.c
Updating plugin.c
Scanning dependencies of target rfnoc64
[ 28%] Building C object epan/rfnoc/CMakeFiles/rfnoc64.dir/plugin.c.o
[ 42%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/packet-rfnoc.cpp.o
[ 57%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/home/bluelab/work/uhd/host/lib/exception.cpp.o
[ 71%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/home/bluelab/work/uhd/host/lib/rfnoc/chdr_types.cpp.o
[ 85%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/home/bluelab/work/uhd/host/lib/rfnoc/chdr_packet_writer.cpp.o
make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libgcrypt.so', needed by 'epan/rfnoc/epan/rfnoc64.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:94: epan/rfnoc/CMakeFiles/rfnoc64.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

```

Have you ever seen this issue ? If yes could you please tell me how to solve it ? 

Thanks and have a good day :)

Nicolas
`

mbr0wn, 2021-09-24:
You need to install some dependencies. I think it was libgrypt-dev and something else. The error is not particularly helpful, I know, but I'm also not quite sure where those deps are coming from, or we'd be able to check for them in cmake.

mbr0wn, 2021-09-27:
...closing this, even though we might want to update the README.

closed 2021-09-27 by mbr0wn