EttusResearch/uhd #494: Error building CHDR dissectors

NicolasCaseau, 2021-09-27:
Hi, 

I'm trying to build the CHDR dissectors to use them in wireshark. Unfortunately, I have an error when I try to run the `cmake` command in the terminal : 

```
~/work/uhd/tools/dissectors/build$ cmake ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for one of the modules 'glib-2.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
CMake Error at CMakeLists.txt:93 (message):
  Wireshark was compiled without support for plugins

-- Configuring incomplete, errors occurred!
See also "/home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeOutput.log".
See also "/home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeError.log".

```
As you can see, Wireshark is compiled without the plugins and I don't really understand why. When I first tried to use the `cmake`, the compiler stated that "gregex.h" and "gchecksum.h" couldn't be found in the "/usr/include/glib-2.0/glib" folder so maybe the error comes from there.

Do you know how I could solve this issue ? 

Thank you and have a good day :)

Nicolas

neelpandeya, 2021-09-27:
@NicolasCaseau What is your environment? Which Linux distribution, which version, which version of UHD?

NicolasCaseau, 2021-09-27:
Hi, thanks for the quick response :)

I'm working on ubuntu 18.04 and I just cloned the git of Ettus so I think the version of UHD I must be using is 4.1.0.0.

I also made a mistake earlier, I managed to find  "gregex.h" and "gchecksum.h"  the "/usr/include/glib-2.0/glib" folder so the error I'm getting must no come from there.

wordimont, 2021-09-27:
I'm running 18.04 and I also see those messages about glib, but I am able to build the dissectors. I wonder if this could be due to your Wireshark version. The dissectors in UHD 4 require Wireshark version 3.x but I think 2.x will be installed by default on Ubuntu 18.04.

Also, make sure you've installed all the dependencies listed here for 18.04:
https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

NicolasCaseau, 2021-09-28:
Hi, thanks for your answer :)

I just updated wireshark to the latest version (3.4.8) as you can see below : 

```
~/work/uhd/tools/dissectors/build$ wireshark --version
Wireshark 3.4.8 (Git v3.4.8 packaged as 3.4.8-1~ubuntu18.04.0+wiresharkdevstable1)

Copyright 1998-2021 Gerald Combs <gerald@wireshark.org> and contributors.
License GPLv2+: GNU GPL version 2 or later <https://www.gnu.org/licenses/gpl-2.0.html>
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiled (64-bit) with Qt 5.9.5, with libpcap, with POSIX capabilities (Linux),
with libnl 3, with GLib 2.56.4, with zlib 1.2.11, with SMI 0.4.8, with c-ares
1.14.0, with Lua 5.2.4, with GnuTLS 3.5.18 and PKCS #11 support, with Gcrypt
1.8.1, with MIT Kerberos, with MaxMind DB resolver, with nghttp2 1.30.0, with
brotli, with LZ4, with Zstandard, with Snappy, with libxml2 2.9.4, with
QtMultimedia, without automatic updates, with SpeexDSP (using system library),
with Minizip.

Running on Linux 4.15.0-158-generic, with Intel(R) Xeon(R) Platinum 8280 CPU @
2.70GHz (with SSE4.2), with 1031715 MB of physical memory, with locale
en_US.UTF-8, with libpcap version 1.8.1, with GnuTLS 3.5.18, with Gcrypt 1.8.1,
with brotli 1.0.4, with zlib 1.2.11, binary plugins supported (0 loaded).

Built using gcc 7.5.0.
```
And I also downloaded all the dependencies required for ubuntu 18.04. 

However I still have the same issue as before. 

Do you have any other ideas ?

Have a great day ! 

Nicolas

NicolasCaseau, 2021-09-28:
Also the output of the CmakeError.log file is the following :

```
cat /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeError.log
Determining if files glib/gregex.h exist failed with the following output:
Change Dir: /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_a31c2/fast"
/usr/bin/make -f CMakeFiles/cmTC_a31c2.dir/build.make CMakeFiles/cmTC_a31c2.dir/build
make[1]: Entering directory '/home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_a31c2.dir/CheckIncludeFiles.c.o
/usr/bin/cc  -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include   -o CMakeFiles/cmTC_a31c2.dir/CheckIncludeFiles.c.o   -c /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c
In file included from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:0:
/usr/include/glib-2.0/glib/gregex.h:25:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gregex.h:28:0,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gerror.h:23:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gerror.h:28:0,
                 from /usr/include/glib-2.0/glib/gregex.h:28,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gquark.h:29:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gquark.h:32:0,
                 from /usr/include/glib-2.0/glib/gerror.h:28,
                 from /usr/include/glib-2.0/glib/gregex.h:28,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gtypes.h:29:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:9:0,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/gquark.h:32,
                 from /usr/include/glib-2.0/glib/gerror.h:28,
                 from /usr/include/glib-2.0/glib/gregex.h:28,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gmacros.h:33:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gtypes.h:34:0,
                 from /usr/include/glib-2.0/glib/gquark.h:32,
                 from /usr/include/glib-2.0/glib/gerror.h:28,
                 from /usr/include/glib-2.0/glib/gregex.h:28,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gversionmacros.h:29:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gregex.h:29:0,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gstring.h:29:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gstring.h:33:0,
                 from /usr/include/glib-2.0/glib/gregex.h:29,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gunicode.h:24:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gstring.h:34:0,
                 from /usr/include/glib-2.0/glib/gregex.h:29,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gbytes.h:26:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gbytes.h:30:0,
                 from /usr/include/glib-2.0/glib/gstring.h:34,
[... 51 lines cut ...]
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gchecksum.h:27:0,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/gbytes.h:26:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
In file included from /usr/include/glib-2.0/glib/gbytes.h:30:0,
                 from /usr/include/glib-2.0/glib/gchecksum.h:27,
                 from /home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp/CheckIncludeFiles.c:2:
/usr/include/glib-2.0/glib/garray.h:29:2: error: #error "Only <glib.h> can be included directly."
 #error "Only <glib.h> can be included directly."
  ^~~~~
CMakeFiles/cmTC_8097d.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_8097d.dir/CheckIncludeFiles.c.o' failed
make[1]: *** [CMakeFiles/cmTC_8097d.dir/CheckIncludeFiles.c.o] Error 1
make[1]: Leaving directory '/home/caseau/work/uhd/tools/dissectors/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_8097d/fast' failed
make: *** [cmTC_8097d/fast] Error 2

Source:
/* */
#include <glib/gchecksum.h>

int main(void){return 0;}
```

mbr0wn, 2021-09-28:
I just spun up a Docker to take a quick look. I'm not quite sure why we need to install libgnutls28-dev and libgcrypt20-dev, but when I do (and have the other deps, like Wireshark 3 and Boost installed), this is my output. Note that I see the same messages, but it compiles just fine:

```
root@8bd367f0a5ce:/build# cmake /gnuradio/tools/dissectors/
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for one of the modules 'glib-2.0'
-- Found GLib2: glib-2.0 /usr/include/glib-2.0;/usr/lib/x86_64-linux-gnu/glib-2.0/include
-- Looking for include file glib/gregex.h
-- Looking for include file glib/gregex.h - not found
-- Looking for include file glib/gchecksum.h
-- Looking for include file glib/gchecksum.h - not found
--
-- Configuring RFNoC support...
--   Enabling RFNoC support.
--   Override with -DENABLE_RFNOC=ON/OFF
--
-- Configuring Octoclock support...
--   Disabling Octoclock support.
--   Override with -DENABLE_OCTOCLOCK=ON/OFF
--
-- Configuring ZPU support...
--   Disabling ZPU support.
--   Override with -DENABLE_ZPU=ON/OFF
--
-- ######################################################
-- # Ettus dissectors enabled
-- ######################################################
--   * RFNoC
--
-- ######################################################
-- # Ettus dissectors disabled
-- ######################################################
--   * Octoclock
--   * ZPU
--
-- Configuring done
-- Generating done
-- Build files have been written to: /build
root@8bd367f0a5ce:/build# make
[ 14%] Generating /build/epan/rfnoc/plugin.c
Updating plugin.c
Scanning dependencies of target rfnoc64
[ 28%] Building C object epan/rfnoc/CMakeFiles/rfnoc64.dir/plugin.c.o
[ 42%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/packet-rfnoc.cpp.o
[ 57%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/gnuradio/host/lib/exception.cpp.o
[ 71%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/gnuradio/host/lib/rfnoc/chdr_types.cpp.o
[ 85%] Building CXX object epan/rfnoc/CMakeFiles/rfnoc64.dir/gnuradio/host/lib/rfnoc/chdr_packet_writer.cpp.o
[100%] Linking CXX shared module epan/rfnoc64.so
[100%] Built target rfnoc64
```

mbr0wn, 2021-09-28:
I recommend you clear out the build dir, and make sure that you remove any remnants of old Wireshark 2 from your system. That should do the trick.

NicolasCaseau, 2021-09-29:
Ok thanks, I will try to do that and keep you posted.

mbr0wn, 2021-09-29:
@NicolasCaseau I'm pretty sure that will fix the issue, so I'll close this. But feel free to reopen if it doesn't. FYI, I tried compiling in a Docker container (18.04) without issues.

closed 2021-09-29 by mbr0wn

NicolasCaseau, 2021-09-29:
Hi, I just have an issue to remove the remnants of wireshark 2 ? Should I just run `apt remove wireshark` and `apt autoremove wireshark` ?

mbr0wn, 2021-09-29:
I'm not sure (it's specific to how your system looks like now). You can try using apt purge to remove everything, then reinstall wireshark 3 only.