EttusResearch/uhd #495: Imported target "Boost::chrono" includes non-existent path

marcinsztajn, 2021-09-28:
## Issue Description ##
While installation of the gnuradio using pybombs, the process crashes while UHD library is being installed
## Setup Details ##
My system is:
`Linux ai2n 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
` 
Pybombs version:
`2.3.4`

## Expected Behavior ##
Successful UHD installation
## Actual Behaviour ##
```
Install tree:
|
\- gnuradio
   |
   +- libvolk
   |
   \- uhd
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: uhd
warning: no common commits
remote: Enumerating objects: 104692, done.
remote: Counting objects: 100% (5409/5409), done.
remote: Compressing objects: 100% (2055/2055), done.
remote: Total 104692 (delta 3477), reused 5100 (delta 3321), pack-reused 99283
Receiving objects: 100% (104692/104692), 132.43 MiB | 12.77 MiB/s, done.
Resolving deltas: 100% (80685/80685), done.
From https://github.com/EttusResearch/uhd
 * [new branch]      UHD-3.10                -> uhd/UHD-3.10
 * [new branch]      UHD-3.11                -> uhd/UHD-3.11
 * [new branch]      UHD-3.12                -> uhd/UHD-3.12
 * [new branch]      UHD-3.13                -> uhd/UHD-3.13
 * [new branch]      UHD-3.14                -> uhd/UHD-3.14
 * [new branch]      UHD-3.14.L              -> uhd/UHD-3.14.L
 * [new branch]      UHD-3.15.LTS            -> uhd/UHD-3.15.LTS
 * [new branch]      UHD-3.9.LTS             -> uhd/UHD-3.9.LTS
 * [new branch]      UHD-4.0                 -> uhd/UHD-4.0
 * [new branch]      UHD-4.1                 -> uhd/UHD-4.1
 * [new branch]      amoch/my_fixes          -> uhd/amoch/my_fixes
 * [new branch]      atrnati/ubuntu-20.04-workaround -> uhd/atrnati/ubuntu-20.04-workaround
 * [new branch]      lamsel/fft_playground   -> uhd/lamsel/fft_playground
 * [new branch]      master                  -> uhd/master
 * [new branch]      rfnoc-devel             -> uhd/rfnoc-devel
 * [new branch]      rfnoc-ofdm              -> uhd/rfnoc-ofdm
 * [new branch]      x300_fp_gpio_fix        -> uhd/x300_fp_gpio_fix
 * [new tag]         release_003_010_003_000 -> release_003_010_003_000
 * [new tag]         v3.11.1.0               -> v3.11.1.0
 * [new tag]         v3.14.1.1               -> v3.14.1.1
 * [new tag]         v3.14.1.1.L             -> v3.14.1.1.L
 * [new tag]         v4.1.0.3                -> v4.1.0.3
 * [new tag]             003_007_002_rc1           -> 003_007_002_rc1
 * [new tag]             003_007_003_rc1           -> 003_007_003_rc1
 * [new tag]             003_008_000_rc1           -> 003_008_000_rc1
 * [new tag]             003_008_002_rc1           -> 003_008_002_rc1
 * [new tag]             003_008_003_rc1           -> 003_008_003_rc1
 * [new tag]             003_008_004_rc1           -> 003_008_004_rc1
 * [new tag]             003_008_005_rc1           -> 003_008_005_rc1
 * [new tag]             003_009_000_rc1           -> 003_009_000_rc1
 * [new tag]             003_009_002_rc1           -> 003_009_002_rc1
 * [new tag]             003_009_003_rc1           -> 003_009_003_rc1
 * [new tag]             003_009_004_rc1           -> 003_009_004_rc1
 * [new tag]             003_009_005_rc1           -> 003_009_005_rc1
 * [new tag]             003_009_006_rc1           -> 003_009_006_rc1
 * [new tag]             003_009_007_rc1           -> 003_009_007_rc1
 * [new tag]             003_010_001_000_rc1       -> 003_010_001_000_rc1
 * [new tag]             003_010_001_000_rc2       -> 003_010_001_000_rc2
 * [new tag]             003_010_001_001_rc1       -> 003_010_001_001_rc1
[... 425 lines cut ...]
CMake Error in python/CMakeLists.txt:
  Imported target "Boost::chrono" includes non-existent path

    "/include"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
[ERROR] Configuration failed after running at least twice.
[ERROR] Problem occurred while building package uhd:
Configuration failed
[ERROR] Error installing package uhd. Aborting.
```
`
## Steps to reproduce the problem ##
`pybombs prefix init /home/ubuntu/prefix/gnuradio -R gnuradio-default
`

marcinsztajn, 2021-09-29:
> Which Ubuntu version are you on?

I am running VM Ubuntu 20.04.2 LTS using Oracle VM VirtualBox

`cat /etc/os-release` gives:
```
NAME="Ubuntu"
VERSION="20.04.2 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.2 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

michaelld, 2021-09-29:
@marcinsztajn do both `/bin/cmake` and `/usr/bin/cmake` exist? If so, which is first in `PATH`? (I believe if you do `ls -l` on those, you'll find that the former is a link to the latter ...) Can you try creating a new build directory and then instead of just `cmake ..` do `/usr/bin/cmake ..` ... explicitly call the "correct" cmake binary. There seems to be a bug in some versions of cmake  / boost finding where it doesn't determine the actual absolute path PREFIX to its own binary (should be `/usr`), and thus uses the incorrect PREFIX (here `/`) for referencing some packages during find.

michaelld, 2021-09-29:
Alternatives that should also work are

1) `PATH=/usr/bin:$PATH cmake ..`
    
2) `cmake -DCMAKE_FIND_ROOT_PATH=/usr ..`

marcinsztajn, 2021-09-30:
> @marcinsztajn do both `/bin/cmake` and `/usr/bin/cmake` exist? If so, which is first in `PATH`? (I believe if you do `ls -l` on those, you'll find that the former is a link to the latter ...) Can [...]

@michaelld `/bin/cmake` and `/usr/bin/cmake` both exist but former is not linked to the latter (we are talking about symbolic linking am I right?). My `PATH` is: `/home/ubuntu/radioconda/bin:/home/ubuntu/radioconda/condabin:/sbin:/bin:/usr/bin:/usr/local/bin:/snap/bin`
so `/bin` is first
After deleting the build folder and creating it once again and then running `/usr/bin/cmake ..` build process went successfully. So the problem was with cmake but what the difference if its from `/bin` or `/usr/bin` ?

michaelld, 2021-09-30:
@marcinsztajn I don't know the difference between the 2 cmake binaries. On some systems we see `/bin/cmake` -> `/usr/bin/cmake` and so a soft link. Since there is a simple "fix", I haven't spent quality time with cmake to see if the issue has been fixed upstream either with the cmake binary or the FindBoost.cmake script (whether those supplied by cmake or boost).

michaelld, 2021-09-30:
Since we have a resolution, I'm closing this issue.

closed 2021-09-30 by michaelld

harshit56, 2023-01-20:
> @marcinsztajn do both `/bin/cmake` and `/usr/bin/cmake` exist? If so, which is first in `PATH`? (I believe if you do `ls -l` on those, you'll find that the former is a link to the latter ...) Can [...]

After using /usr/bin/cmake .. in the build directory still i am getting an an error:

This is my error:
-- 
-- Configuring the Python interpreter...
-- Manually determining build Python version...
-- Python interpreter: /usr/local/bin/python3 Version: 3.11.1
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Manually determining runtime Python version...
-- Python runtime interpreter: /usr/local/bin/python3 Version: 3.11.1
-- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
-- Finding Python Libraries...
-- Python Libraries: /usr/local/lib/libpython3.11.a
-- Python include directories: /usr/local/include/python3.11
-- Working off of feature or development branch. Updating version number.
-- Using UHD Images Directory: /usr/local/share/uhd/images
-- Build type not specified: defaulting to release.
-- 
-- Configuring Boost C++ Libraries...
-- 
-- Checking for Boost version 1.65 or greater
--   Looking for required Boost components...
--   Enabling Boost Error Code Header Only
--     Disabling boost::asio use of std::string_view
--   Boost version: 1.71.0
--   Boost include directories: /usr/include
--   Boost library directories: /usr/lib/x86_64-linux-gnu
--   Boost libraries: Boost::chrono;Boost::date_time;Boost::filesystem;Boost::program_options;Boost::serialization;Boost::thread;Boost::unit_test_framework;Boost::system
-- Looking for Boost version 1.65 or greater - found
-- 
-- Python checking for compatible Python version
-- Python checking for compatible Python version - 3.11.1 satisifes minimum required version 3.6
-- 
-- Python checking for Mako templates module
-- Python checking for Mako templates module - "import mako" failed (is it installed?)
-- 
-- Python checking for requests module
-- Python checking for requests module - "import requests" failed (is it installed?)
-- 
-- Python checking for numpy module
-- Python checking for numpy module - "import numpy" failed (is it installed?)
-- 
-- Python checking for ruamel.yaml module
-- Python checking for ruamel.yaml module - "import ruamel.yaml" failed (is it installed?)
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
CMake Error at cmake/Modules/UHDComponent.cmake:59 (message):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:433 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!

Please help me thank you