EttusResearch/uhd #497: thread[thread-per-block[1]: <block gr uhd usrp sink (1)>]: EnvironmentError: IOError: Failed to sync DACs! EnvironmentError:   In TX mode

stackprogramer, 2021-10-02:
when i used usrp x310 with daughterboard UBX 160  for rx mode i had not error for tx had this errors....

[32;1m[INFO] [UHD] [39;0mlinux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.1.0-3
[32;1m[INFO] [X300] [39;0mX300 initialization sequence...
[32;1m[INFO] [X300] [39;0mMaximum frame size: 1472 bytes.
[32;1m[INFO] [X300] [39;0mRadio 1x clock: 200 MHz
[32;1m[INFO] [GPS] [39;0mNo GPSDO found
[32;1m[INFO] [0/DmaFIFO_0] [39;0mInitializing block control (NOC ID: 0xF1F0D00000000000)
[32;1m[INFO] [0/DmaFIFO_0] [39;0mBIST passed (Throughput: 1317 MB/s)
[32;1m[INFO] [0/DmaFIFO_0] [39;0mBIST passed (Throughput: 1292 MB/s)
[32;1m[INFO] [0/Radio_0] [39;0mInitializing block control (NOC ID: 0x12AD100000000001)
[32;1m[INFO] [0/Radio_1] [39;0mInitializing block control (NOC ID: 0x12AD100000000001)
[32;1m[INFO] [0/DDC_0] [39;0mInitializing block control (NOC ID: 0xDDC0000000000000)
[32;1m[INFO] [0/DDC_1] [39;0mInitializing block control (NOC ID: 0xDDC0000000000000)
[32;1m[INFO] [0/DUC_0] [39;0mInitializing block control (NOC ID: 0xD0C0000000000000)
[32;1m[INFO] [0/DUC_1] [39;0mInitializing block control (NOC ID: 0xD0C0000000000000)
[33;1m[WARNING] [X300 RADIO] [39;0mRequesting invalid sampling rate from device: 200 MHz. Actual rate is: 100 MHz.
thread[thread-per-block[1]: <block gr uhd usrp sink (1)>]: EnvironmentError: IOError: Failed to sync DACs! EnvironmentError: IOError: Block ctrl (CE_02_Port_50) no response packet - AssertionError: bool(buff)
  in uint64_t ctrl_iface_impl<_endianness>::wait_for_ack(bool, double) [with uhd::endianness_t _endianness = (uhd::endianness_t)0; uint64_t = long unsigned int]
  at /build/uhd-i9z6iM/uhd-3.13.1.0/host/lib/rfnoc/ctrl_iface.cpp:154

mbr0wn, 2021-10-04:
Thanks! Can you try a more recent version of UHD please to see if it persists?

stackprogramer, 2021-10-05:
Hi, I had a question with Gnuradio 3.7.10 or 3.8, UHD version that is used is 3.13 or 3.15. I used the apt-get install command...
How can install UHD 4 on Debian ubuntu without compiling it...
All work with USRP is in Gnuradio...

mbr0wn, 2024-06-21:
Lack of info here, so closing.

closed 2024-06-21 by mbr0wn