EttusResearch/uhd #518: [INFO] [UHD RF] Tx while waiting for EOB, timed out... 20.6379 >= 0. Starting new burst...

akhila-rao, 2021-11-02:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

michaelld, 2021-11-02:
@akhila-rao a couple thoughts: (1) UHD 4.2 has not been released; it is still in beta. Please use the latest 4.1 release instead and see if that works & if not then update your post here. (2) please provide a run log that includes all of the printouts for UHD and whatever app is using it. What you've provided doesn't provide enough info for us to help.

mbr0wn, 2024-06-25:
Closing this bug report, as it's partly related to srs, partly using a beta version, and partly because it's inactive.

closed 2024-06-25 by mbr0wn