EttusResearch/uhd #522: Documentation of rfnoc clocks

TSlivede, 2021-11-09:
## Issue Description ##

I tried to find an official documentation about what clock speed is used for the `rfnoc_chdr` clock, but couldn't find anything reliable.

## Setup Details ##

current state of the documentation

## Expected Behavior ##

I thought, that there should be some page where the different clocks used by the RFNoC are documented.

## Actual Behaviour ##

I couldn't find any official looking documentation about the `rfnoc_chdr` clock. There are only [few search results](https://www.google.com/search?q="rfnoc_chdr"+clock+site%3Aettus.com) and only [this](https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0#Example:_Adding_an_FFT_Block) KB page seems to say something about the clock ("[...] but `rfnoc_chdr` clock (200 MHz) should also work [...]").

Is it correct, that `rfnoc_chdr` is clocked at 200MHz?

Also am I guessing correctly, that the `radio` clock (`{ srcblk: _device_, srcport: radio,` [...]) is the `master_clock_rate` from the [`device args`](https://files.ettus.com/manual/page_configuration.html)? Is this documented somewhere?

wordimont, 2021-11-16:
The speed of the rfnoc_chdr_clk clock depends on the USRP. So it's not always 200 MHz. You are right that this is not currently documented anywhere besides just looking at the source code. These are the clock rates I found:

- X3xx 187.5 MHz
- N3xx 200 MHz
- E31x 100 MHz
- E320 200 Mhz

You are correct regarding master_clock_rate. So the radio_clk used by RFNoC will change depending on the configured master clock.

mbr0wn, 2024-06-26:
The KB actually does a good job of this: https://kb.ettus.com/RFNoC_Frequently_Asked_Questions#RFNoC_Clocks

closed 2024-06-26 by mbr0wn