EttusResearch/uhd #524: There is one phase difference in UHD 3.15.0.0, but there are four phase differences in UHD 4.1.0.1

zchengyun, 2021-11-12:
## Issue Description ##

      I use four usrp-2945 as synchronous acquisition system, CDA-2990 provides internal reference clock and PPS to four usrp-2945, and pxie-5651 and pxie-5652 are used to provide two-stage local oscillator for four USRPs. Now I use two usrp-2945 to collect the data of two channels (one channel for each) and calculate the phase difference between the two channels. If I use UHD 4.1.0.1, after multiple acquisition, I find that there are four kinds of phase differences, and the phase differences are roughly 90 degrees, 180 degrees and 270 degrees. For example, the four phase differences are 4.5 degrees, 95.53 degrees, 189.51 degrees and 275.49 degrees respectively. If I use UHD 3.15.0.0, there is only one phase difference.
## Setup Details ##

1.CDA-2990 provides internal reference clock and PPS to four usrps.
2.Pxie-5651 and Pxie-5652 are used to provide two-stage local oscillator for four USRPs.
## Expected Behavior ##

Using UHD 4.1.0.1, there is only one phase difference.
## Actual Behaviour ##

There are four phase differences using UHD 4.1.0.1.
## Steps to reproduce the problem ##

After synchronizing the reference clock, PPS and lo, use UHD 4.1.0.1 to collect one channel of different USRP and calculate the phase difference.
## Additional Information ##

mbr0wn, 2021-11-25:
Note to self: USRP-2945 == X310 + 2x TwinRX.

mbr0wn, 2021-11-25:
@zchengyun when you say four phase differences, between which channels are you measuring the phase?

zchengyun, 2021-12-08:
> @zchengyun当您说四个相位差时，您测量的是哪个通道之间的相位？

> @zchengyun when you say four phase differences, between which channels are you measuring the phase?
Sorry to reply to you so late, I used two 2945, each with one channel. The reference source, PPS and two-stage local oscillator are provided externally.

zchengyun, 2021-12-08:
> @zchengyun when you say four phase differences, between which channels are you measuring the phase?

> @mbr0wn Sorry to reply to you so late, I used two 2945, each with one channel. The reference source, PPS and two-stage local oscillator are provided externally.

zchengyun, 2022-01-28:
@mbr0wn I use the first channel of two usrps to collect the same signal source and calculate the phase difference between the two channels. There will be four phase differences, but it has only one phase difference on uhd3.15

jmorton1, 2022-06-23:
Were you able to resolve this issue?  I am having the same issue with the same setup.  I believe it's related to the time_sync signal passed to the fe_control module in x300_core.v:
https://github.com/EttusResearch/uhd/blob/8daa80c05f8006697aea05fea4ac5cd4dabbaef1/fpga/usrp3/top/x300/x300_core.v#L610

In UHD 4.0 and beyond, it's left empty.  In UHD 3.15 it's connected to a time sync signal that synchronizes the frontend when you sync to PPS.  It resets the quarter_rate_downconverter module that mixes down the ADC input to baseband.  The quarter_rate_downconverter cycles through 0, 90, 180, 270 degrees, so if you're not sync'd you'll see a phase offset of one of these vlues between channels.

closed 2022-08-11 by zchengyun

mbr0wn, 2023-12-14:
Definitely fixed since c70b479