EttusResearch/uhd #525: Could UHD for Zed+FMCOMMS2 be possible？

vvverilog, 2021-11-12:
Hi, I want to now is there some guide for porting the uhd into the xilinx zynq board and adi RF frontend?

If this could be done, how to make the boot image for these board？

thanks!

closed 2021-11-24 by mbr0wn

christian-hahn, 2021-11-24:
> Hi @vvverilog <https://github.com/vvverilog>, yes, it's possible. Others
> have ported UHD/RFNoC to other platforms, such as the ZCU111.
>
> However, we don't have any porting guides, and we don't have the resources
> to help. You can peruse the source code. I recommend styling the Zedboard
> like an MPM device (e.g., E310, E320) and follow that model.

Hi Martin,

Do you have a link to the ZCU111 port?

Many thanks,
Christian

mbr0wn, 2021-11-26:
No, someone on the usrp-users mailing list was talking about it.