EttusResearch/uhd #526: I want to compile FPGA source, but i can not open project in xilinx vivado?

stackprogramer, 2021-11-15:
I want to compile the FPGA source, but I can not open the project in Xilinx Vivado?
how can open and build an FPGA source in GUI mode? It is the way I open it like a normal FPGA project...

thanks in advance

stackprogramer, 2021-11-15:
it has not any xpr file or Xilinx project file

wordimont, 2021-11-15:
The build process does not use Xilinx projects. The FPGAs are built in non-project mode. You can read about it in the Xilinx documentation, but it uses the Tcl shell rather than projects and the GUI. Also, Make is used to determine dependencies and to generate the lists of files needed. You can find instructions for running the build process in the UHD manual:

https://files.ettus.com/manual/md_usrp3_build_instructions.html

However, there is a GUI option to run the FPGA build in the Vivado GUI. Note that the IP build process will still run from the command line, but once the IP is built, the GUI will load and the TCL commands will run to build the FPGA in the GUI. Once the build completes, you can then save it as a regular Xilinx project which saves your current design in project mode.

Here's a quick summary of the steps to build it in the GUI, after which you can save the build as a project (example here for X310):

```
cd uhd/fpga/usrp3/top/x300
source setupenv.sh
make X310_XG GUI=1
```

stackprogramer, 2021-11-15:
@wordimont 

thanks but when I want to compile i had error

`ERROR: [Designutils 20-414] HRTInvokeSpec : No Verilog or VHDL sources specified
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.`

wordimont, 2021-11-15:
Can you provide more information?

- What distribution/OS are you running?
- What UHD version are you trying to compile an FPGA for?
- Which FPGA are you trying to compile?
- Did setupenv.sh run without errors?
- What was the command line you used?
- At what point in the build did these errors occur?
- Did you modify anything or are you building from an unmodified master or released version?

stackprogramer, 2021-11-15:
What distribution/OS are you running? **Ubuntu 20.04**
What UHD version are you trying to compile an FPGA for? **UHD 4**
Which FPGA are you trying to compile? **x300   in uhd-master/fpga/usrp3/top/x300**
Did setupenv.sh run without errors?  **Successfully message is showed.....**
What was the command line you used? ` make X300_HG GUI=1   `
At what point in the build did these errors occur? **see all log error and warning**
Did you modify anything or are you building from an unmodified master or released version? **No**

`make X300_HG GUI=1   
make -f Makefile.x300.inc bin NAME=X300_HG ARCH=kintex7 PART_ID=xc7k325t/ffg900/-2 BUILD_1G=1 BUILD_10G=1 SFP0_1GBE=1 SFP1_10GBE=1  X300=1 TOP_MODULE=x300 EXTRA_DEFS="BUILD_1G=1 BUILD_10G=1 SFP0_1GBE=1 SFP1_10GBE=1  X300=1" DEFAULT_RFNOC_IMAGE_CORE_FILE=x300_rfnoc_image_core.v DEFAULT_EDGE_FILE=/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/x300_static_router.hex
make[1]: Entering directory '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300'
BUILDER: Checking tools...
* GNU bash, version 5.0.17(1)-release (x86_64-pc-linux-gnu)
* Python 3.8.10
* Vivado v2020.2 (64-bit)
========================================================
BUILDER: Building IP ten_gig_eth_pcs_pma
========================================================
BUILDER: Staging IP in build directory...
BUILDER: Reserving IP location: /home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma
BUILDER: Retargeting IP to part kintex7/xc7k325t/ffg900/-2...
BUILDER: Building IP...
[00:00:00] Executing command: vivado -mode batch -source /home/sp/Documents/uhd-master/fpga/usrp3/tools/scripts/viv_generate_ip.tcl -log ten_gig_eth_pcs_pma.log -nojournal
WARNING: [IP_Flow 19-2162] IP 'ten_gig_eth_pcs_pma' is locked:
CRITICAL WARNING: [filemgmt 20-1366] Unable to reset target(s) for the following file is locked: /home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci
CRITICAL WARNING: [filemgmt 20-1365] Unable to generate target(s) for the following file is locked: /home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci
[00:00:23] Current task: Initialization +++ Current Phase: Starting
[00:00:23] Current task: Initialization +++ Current Phase: Finished
[00:00:23] Executing Tcl: synth_design -top ten_gig_eth_pcs_pma -part xc7k325tffg900-2 -mode out_of_context
[00:00:23] Starting Synthesis Command
WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Synthesis target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
WARNING: [IP_Flow 19-2162] IP 'ten_gig_eth_pcs_pma' is locked:
ERROR: [Designutils 20-414] HRTInvokeSpec : No Verilog or VHDL sources specified
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xml'
ERROR: [Vivado 12-398] No designs are open
ERROR: [Common 17-69] Command failed: * IP definition '10G Ethernet PCS/PMA (10GBASE-R/KR) (6.0)' for IP 'ten_gig_eth_pcs_pma' (customized with software release 2019.1) has a different revision in the IP Catalog.
[00:00:28] Current task: Synthesis +++ Current Phase: Starting
[00:00:28] Current task: Synthesis +++ Current Phase: Finished
[00:00:28] Process terminated. Status: Failure

========================================================
Warnings:           3
Critical Warnings:  7
Errors:             9

BUILDER: Releasing IP location: /home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma
make[1]: *** [/home/sp/Documents/uhd-master/fpga/usrp3/top/x300/ip/ten_gig_eth_pcs_pma/Makefile.inc:41: /home/sp/Documents/uhd-master/fpga/usrp3/top/x300/build-ip/xc7k325tffg900-2/ten_gig_eth_pcs_pma/ten_gig_eth_pcs_pma.xci.out] Error 1
make[1]: Leaving directory '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300'
make: *** [Makefile:90: X300_HG] Error 2
`

stackprogramer, 2021-11-15:
I too test with out gui

`make X300_HG`

wordimont, 2021-11-15:
Try running `make cleanall` first. The IP is locked for some reason.

What's the output of `vivado -version`?

stackprogramer, 2021-11-15:
@wordimont 

i do clean all  again and again .....i used vivado 2020.2
`
Vivado v2020.2 (64-bit)

`

wordimont, 2021-11-15:
I think setupenv should have reported an error then since that's the wrong Vivado version. You should be using Vivado v2019.1. The error is occurring because the IP versions in Vivado 2020.2 are different.

Can you share the output of setupenv?

stackprogramer, 2021-11-15:
thanks, I in setupenv changed version so it has not any errors...
if I want to compile with  2020.2 I should change IPs?
i don't want to compile with old version vivado

wordimont, 2021-11-15:
I don't recommend using a newer Vivado version because many things could break. It's a fair amount of work and there are a lot of dependencies on specific Vivado versions between different USRPs. This is why we don't upgrade Vivado very frequently. 

You can read the Xilinx documentation to learn how to upgrade IP. We also have some utilities to help, like viv_upgrade_ip. But you're on your own if you want to try a newer Vivado version since it's untested right now and I can't predict what problems you'll run into.

wordimont, 2021-11-16:
I'm closing this issue since I think we found the root cause of the error. Feel free to post again if you have follow up questions.

closed 2021-11-16 by wordimont