EttusResearch/uhd #528: RFNoC: impossible to integrate block

epetragl, 2021-11-16:
## Issue Description ##

I'm trying to deploy an RFNoC design on an Ettus N320 with UHD 4.1.
I've followed the steps described in this tutorial https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0#Example:_Adding_an_FFT_Block
and I was able to obtain a bitstream which implements (by inspecting the netlist) the desired rfnoc architecture.
Since the only UHD example involving an RFNoC custom block is broken (see : https://github.com/EttusResearch/uhd/issues/527), I've decided
to follow this video from UHD's maintainer (https://www.youtube.com/watch?v=fbcxm7f-Tj0) in order to connect the RFNoC blocks using the UHD python API.

Below I'll recap my results :

a) first step, control the loaded FPGA image with the following code lines:

  graph = uhd.rfnoc.RfnocGraph("addr=192.168.0.1")
  for edge in graph.enumerate_static_connections():
    print(edge.to_string())

here is my output:

  0/SEP#0:0==>0/DUC#0:0
  0/DUC#0:0==>0/Radio#0:0
  0/Radio#0:0==>0/DDC#0:0
  0/DDC#0:0==>0/SEP#0:0
  0/SEP#1:0==>0/DUC#1:0
  0/DUC#1:0==>0/Radio#1:0
  0/Radio#1:0==>0/DDC#1:0
  0/DDC#1:0==>0/SEP#1:0
  0/SEP#2:0==>0/Replay#0:0
  0/Replay#0:0==>0/SEP#2:0
  0/SEP#3:0==>0/Replay#0:1
  0/Replay#0:1==>0/SEP#3:0
  0/SEP#4:0==>0/FFT#0:0
  0/FFT#0:0==>0/SEP#4:0

The static connections looks good.

b) second step, create the graph:

    rx_streamer = graph.create_rx_streamer(1, uhd.usrp.StreamArgs("fc32", "sc16"))

    graph.connect("0/Radio#0", 0, "0/DDC#0", 0, False)

    if "0" in add_fft :
        print("Graph without fft \n")
        graph.connect("0/DDC#0", 0, rx_streamer, 0)
    else:
        print("Graph with fft \n")
        graph.connect("0/DDC#0", 0, "0/FFT#0", 0, False)
        graph.connect("0/FFT#0", 0, rx_streamer, 0)

    graph.commit()
    radio_block.set_rx_frequency(2500e6, 0)
    radio_block.set_rx_gain(40, 0)
    radio_block.set_rx_antenna("RX2", 0)
    radio_block.set_rate(125e6)

    ddc_block.set_input_rate(125e6, 0)
    ddc_block.set_output_rate(5e6, 0)

    num_samples = int(ddc_block.get_output_rate(0) * 3)
    radio_data = np.zeros((1, num_samples), dtype="complex64")

    stream_cmd = uhd.types.StreamCMD(uhd.types.StreamMode.num_done)
    stream_cmd.num_samps = num_samples
    stream_cmd.stream_now = True

    rx_streamer.issue_stream_cmd(stream_cmd)
    num_samples_received = rx_streamer.recv(radio_data, uhd.types.RXMetadata(), 15.0)
    print("Nb of samples received = " + str(num_samples_received))

    graph.release()

My outputs are:
- without the FFT: I'm receiving a certain amount of samples, which is the expected result.
- when I insert the FFT block between the DDC and EP0: the N320 stops sending data (in other words
num_samples_received = 0)

The same behavior is described in this issue on the Ettus mailing list
(https://www.mail-archive.com/usrp-users@lists.ettus.com/msg13205.html).

c) In order to test the FFT in a different way I've also tried to connect the FFT to a TX and an RX streamer
(as shown in https://www.youtube.com/watch?v=fbcxm7f-Tj0).
The python code is described below :

    graph = uhd.rfnoc.RfnocGraph("addr=192.168.0.1")
    for edge in graph.enumerate_static_connections():
        print(edge.to_string())

    fft_length = 256
    fft_magnitude = uhd.libpyuhd.rfnoc.fft_magnitude.COMPLEX
    fft_direction = uhd.libpyuhd.rfnoc.fft_direction.FORWARD
    fft_shift = uhd.libpyuhd.rfnoc.fft_shift.NORMAL

    fft = uhd.rfnoc.FftBlockControl(graph.get_block("0/FFT#0"))
    fft.set_length(fft_length)
    fft.set_magnitude(fft_magnitude)
    fft.set_direction(fft_direction)
    fft.set_shift_config(fft_shift)

    sa = uhd.usrp.StreamArgs("fc32", "sc16")
    sa.args = "spp=" + str(fft_length)

    tx_streamer = graph.create_tx_streamer(1, sa)
    rx_streamer = graph.create_rx_streamer(1, sa)

    graph.connect(tx_streamer, 0 , fft.get_unique_id(), 0)
    graph.connect(fft.get_unique_id(), 0, rx_streamer, 0)

    graph.commit()

    input_data = np.zeros((1, 250000), dtype=np.complex64)

    tx_md = uhd.types.TXMetadata()
    tx_md.start_of_burst = True
    tx_md.end_of_burst = True

    num_sent = tx_streamer.send(input_data, tx_md)

    output_data = np.zeros((1, 250000), dtype="complex64")
    recv_md = uhd.types.RXMetadata()
    num_recv = rx_streamer.recv(output_data, recv_md, 0.1)

    print("num sent" + str(num_sent) + "recv" + str(num_recv) + "\n")

    graph.release()

My output : when I execute the script I get this error while creating the graph

    Traceback (most recent call last):
    File "fft_loopback.py", line 16, in <module>
    graph.connect(tx_streamer, 0 , "0/FFT#0", 0)
    RuntimeError: RuntimeError: tx xport timed out wating for a strs packet during initialization

Please note that I had already tried with the OOT gain block from the example,
but I was getting exactly the same results.
I have tried with both the current master branch and with the 4.1.0 branch.

Can anyone please point me what mistakes I am doing (either in the steps I am performing, or in the code)?
Thanks in advance!

## Setup Details ##

Vivado 2019.1
UHD 4.1 and master 
Ettus N320

## Expected Behavior ##

Being able to use UHD example or UHD python API to insert a custom RFNOC block between DDC and EP0.

## Steps to reproduce the problem ##

- generate an n320 bitstream with an FFT block
- execute the python code presented in https://www.youtube.com/watch?v=fbcxm7f-Tj0

atrnati, 2021-11-16:
In your first example, make sure that you are configuring your FFT block with a size value of some integer of two number of samples, like in the second code snippet. Secondly, the FFT block will not respond when it receives a packet that is not the FFT size, so you must ensure that the radio is producing appropriately sized packets at its output. Normally, this is achieved by setting the `spp` parameter when creating a stream as in the second example, but there is an issue with this technique when there are stream endpoints in the middle of the graph. You can try setting the `spp` property directly on the radio:
```
fft_size = <some appropriate value>
radio_block.set_properties(f'spp={fft_size}', 0)
```

epetragl, 2021-11-16:
Hi Aaron,

thank you very much for your quick answer. At the moment I'm unable to try your suggestion.

Nevertheless I'm a bit puzzled because I'm experiencing the same behavior with the OOT  block rfnoc_gain.

The latter shall respond on any packet size if I'm not mistaken.

Moreover what you suggest don't explain the error I get when I'm trying to connect the FFT (or the gain block)

to a TX streamer as suggested in your GRCon video.

The error is :

Traceback (most recent call last):
File "fft_loopback.py", line 16, in <module>
graph.connect(tx_streamer, 0 , "0/FFT#0", 0)
RuntimeError: RuntimeError: tx xport timed out wating for a strs packet during initialization

En passant, your video is really nice and helped me a lot

to understand how to manipulate rfnoc architectures via python.

Regards

Enrico

epetragl, 2021-11-18:
> In your first example, make sure that you are configuring your FFT block with a size value of some integer of two number of samples, like in the second code snippet. Secondly, the FFT block will not [...]

Good news! With the change you've suggested me, the following piece of code correctly inserts the FFT between the DDC and the RX_streamer. :-)

import sys
import re
import uhd
import numpy as np

    
fft_length = 256
fft_magnitude = uhd.libpyuhd.rfnoc.fft_magnitude.COMPLEX 
fft_direction = uhd.libpyuhd.rfnoc.fft_direction.FORWARD 
fft_shift = uhd.libpyuhd.rfnoc.fft_shift.NORMAL 
    
fft = uhd.rfnoc.FftBlockControl(graph.get_block("0/FFT#0"))
fft.set_length(fft_length)
fft.set_magnitude(fft_magnitude)
fft.set_direction(fft_direction)
fft.set_shift_config(fft_shift)  

radio_noc_block = graph.get_block("0/Radio#0")
radio_block = uhd.rfnoc.RadioControl(radio_noc_block)
radio_block.set_properties(f'spp={fft_length}', 0)
    
sa = uhd.usrp.StreamArgs("fc32", "sc16")
sa.args = "spp=" + str(fft_length)

ddc_block = uhd.rfnoc.DdcBlockControl(graph.get_block("0/DDC#0"))
    
rx_streamer = graph.create_rx_streamer(1, sa)

graph.connect("0/Radio#0", 0, "0/DDC#0", 0, False)

print("Graph with fft \n")
graph.connect("0/DDC#0", 0, "0/FFT#0", 0, False)
graph.connect("0/FFT#0", 0, rx_streamer, 0)
    
graph.commit()
    
radio_block.set_rx_frequency(2500e6, 0)
radio_block.set_rx_gain(40, 0)
radio_block.set_rx_antenna("RX2", 0)  
radio_block.set_rate(125e6)

ddc_block.set_input_rate(125e6, 0)
ddc_block.set_output_rate(5e6, 0)

num_samples = int(ddc_block.get_output_rate(0) * 3)
radio_data = np.zeros((1, num_samples), dtype="complex64")

stream_cmd = uhd.types.StreamCMD(uhd.types.StreamMode.num_done) 
stream_cmd.num_samps = num_samples
stream_cmd.stream_now = True
    
rx_streamer.issue_stream_cmd(stream_cmd)
num_samples_received = rx_streamer.recv(radio_data, uhd.types.RXMetadata(), 15.0)
print("Nb of samples received = " + str(num_samples_received))

graph.release()

Unfortunately I'm still having troubles when I'm trying to use an OOT block (rfnoc_gain from Ettus' tutorial).
If I use a bitstream with this topology : 

0/SEP#0:0==>0/DUC#0:0
0/DUC#0:0==>0/Radio#0:0
0/Radio#0:0==>0/DDC#0:0
0/DDC#0:0==>0/SEP#0:0
0/SEP#1:0==>0/DUC#1:0
0/DUC#1:0==>0/Radio#1:0
0/Radio#1:0==>0/DDC#1:0
0/DDC#1:0==>0/SEP#1:0
0/SEP#2:0==>0/Replay#0:0
0/Replay#0:0==>0/SEP#2:0
0/SEP#3:0==>0/Replay#0:1
0/Replay#0:1==>0/SEP#3:0
0/SEP#4:0==>0/Block#0:0
0/Block#0:0==>0/SEP#4:0

and I run my script replacing the FFT with the rfnoc_gain unique ID, I still receive zero samples from the N320. 
Is there an option for the gain block (as it was the case for the FFT example) I need to set in order to get the example work? 

My actual code for rfnoc_gain is listed below :

import sys
import re
import uhd
import numpy as np
   
packet_length = 256

radio_noc_block = graph.get_block("0/Radio#0")
radio_block = uhd.rfnoc.RadioControl(radio_noc_block)
radio_block.set_properties(f'spp={packet_length}', 0)
    
sa = uhd.usrp.StreamArgs("fc32", "sc16")
sa.args = "spp=" + str(packet_length)

ddc_block = uhd.rfnoc.DdcBlockControl(graph.get_block("0/DDC#0"))
    
rx_streamer = graph.create_rx_streamer(1, sa)

graph.connect("0/Radio#0", 0, "0/DDC#0", 0, False)

print("Graph with gain_block \n")
graph.connect("0/DDC#0", 0, "0/Block#0", 0, False)
graph.connect("0/Block#0", 0, rx_streamer, 0)
    
graph.commit()
    
radio_block.set_rx_frequency(2500e6, 0)
radio_block.set_rx_gain(40, 0)
radio_block.set_rx_antenna("RX2", 0)  
radio_block.set_rate(125e6)

ddc_block.set_input_rate(125e6, 0)
ddc_block.set_output_rate(5e6, 0)

num_samples = int(ddc_block.get_output_rate(0) * 3)
radio_data = np.zeros((1, num_samples), dtype="complex64")

stream_cmd = uhd.types.StreamCMD(uhd.types.StreamMode.num_done) 
stream_cmd.num_samps = num_samples
stream_cmd.stream_now = True
    
rx_streamer.issue_stream_cmd(stream_cmd)
num_samples_received = rx_streamer.recv(radio_data, uhd.types.RXMetadata(), 15.0)
print("Nb of samples received = " + str(num_samples_received))

graph.release()

Thanks again! :-)

mbr0wn, 2024-03-22:
@epetragl FYI, we now include an example with the gain block (which I've tested, and it works) so I'm going to close this.

closed 2024-04-17 by mbr0wn