EttusResearch/uhd #531: X300 radio_control_impl.cpp async rx callback fails for odd channels when an OVERFLOW occurs

crroush, 2021-11-23:
UHD Version 4.1.0.4 

## Issue Description ##

Whenever using odd channels, my case 1, 3 and an OVERFLOW occurs, the async message handler  reports:
```Cannot process RX-related async message to invalid chan 1```
The issue occurs [here](https://github.com/EttusResearch/uhd/blob/master/host/lib/rfnoc/radio_control_impl.cpp#L1060)

There is a check that looks to see the number of inputs, and when this message happens, it indicates there are 0 inputs and 2 outputs.  
I changed all the log messages to WARNING in just ```radio_control_impl.cpp``` so that it is easier to see the setup.

```
./benchmark_rate  --rx_channels 1 --rx_rate 200e6 --args addr=192.168.10.2 --duration 30

[INFO] [UHD] linux; GNU C++ version 8.4.0; Boost_107100; UHD_4.1.0.4-0-511b1824
[00:00:00.008306] Creating the usrp device with: addr=192.168.10.2...
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
[WARNING] [RFNOC::GRAPH] One or more blocks timed out during flush!
[WARNING] [0/Radio#0] Loading radio with SPC=1, num_inputs=2, num_outputs=2
[WARNING] [0/Radio#0] Sending async messages to EPID 1, remote port 4, xbar port 1
[WARNING] [0/Radio#0] Calling resolver for samp_rate@1
[WARNING] [0/Radio#0] Calling resolver for samp_rate@0
[WARNING] [0/Radio#0] Calling resolver for spp@0
[WARNING] [0/Radio#0] Calling resolver for spp@1
[WARNING] [0/Radio#0] Calling resolver for samp_rate@0
[WARNING] [0/Radio#0] Calling resolver for spp@0
[WARNING] [0/Radio#0] Calling resolver for spp@1
[WARNING] [0/Radio#0] Calling resolver for samp_rate@1
[WARNING] [0/Radio#0] Setting words_per_pkt to 1996 on chan 0
[WARNING] [0/Radio#0] Setting words_per_pkt to 1996 on chan 1
[WARNING] [0/Radio#1] Loading radio with SPC=1, num_inputs=2, num_outputs=2
[WARNING] [0/Radio#1] Sending async messages to EPID 1, remote port 7, xbar port 1
[WARNING] [0/Radio#1] Calling resolver for samp_rate@1
[WARNING] [0/Radio#1] Calling resolver for samp_rate@0
[WARNING] [0/Radio#1] Calling resolver for spp@0
[WARNING] [0/Radio#1] Calling resolver for spp@1
[WARNING] [0/Radio#1] Calling resolver for spp@0
[WARNING] [0/Radio#1] Calling resolver for spp@1
[WARNING] [0/Radio#1] Calling resolver for samp_rate@0
[WARNING] [0/Radio#1] Calling resolver for samp_rate@1
[WARNING] [0/Radio#1] Setting words_per_pkt to 1996 on chan 0
[WARNING] [0/Radio#1] Setting words_per_pkt to 1996 on chan 1
[WARNING] [0/Radio#1] Calling resolver for spp@0
[WARNING] [0/Radio#1] Calling resolver for samp_rate@0
[WARNING] [0/Radio#1] Calling resolver for spp@1
[WARNING] [0/Radio#1] Calling resolver for samp_rate@1
[WARNING] [0/Radio#0] Calling resolver for spp@0
[WARNING] [0/Radio#0] Calling resolver for samp_rate@0
[WARNING] [0/Radio#0] Calling resolver for spp@1
[WARNING] [0/Radio#0] Calling resolver for samp_rate@1
Using Device: Single USRP:
  Device: X-Series Device
  Mboard 0: X300
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: TwinRX RX0
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: TwinRX RX1
  RX Channel: 2
    RX DSP: 2
    RX Dboard: B
    RX Subdev: TwinRX RX0
  RX Channel: 3
    RX DSP: 3
    RX Dboard: B
[... 235 lines cut ...]
|   |   |   Gain range all: 0.0 to 93.0 step 1.0 dB
|   |   |   Bandwidth range: 80000000.0 to 80000000.0 step 0.0 Hz
|   |   |   Connection Type: II
|   |   |   Uses LO offset: No
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 1
|   |   |   Name: TwinRX RX1
|   |   |   Antennas: RX1, RX2
|   |   |   Sensors: lo_locked
|   |   |   Freq range: 10.000 to 6000.000 MHz
|   |   |   Gain range all: 0.0 to 93.0 step 1.0 dB
|   |   |   Bandwidth range: 80000000.0 to 80000000.0 step 0.0 Hz
|   |   |   Connection Type: QQ
|   |   |   Uses LO offset: No
```

## Expected Behavior ##
async message to handle the overflow correctly
## Actual Behaviour ##
The receiver TIMEOUT errors begin.
## Steps to reproduce the problem ##

See description above
## Additional Information ##

mbr0wn, 2021-11-24:
@crroush Thanks for the great report. This does indeed look like an issue.

mbr0wn, 2021-11-24:
I guess you're using TwinRX? Since you're already modifying UHD, please try changing line 1078:
```cpp
1077         case regmap::SWREG_RX_ERR: {
1078             if (chan > get_num_output_ports()) { // From input to output!
1079                 RFNOC_LOG_WARNING(
1080                     "Cannot process RX-related async message to invalid chan " << chan);
1081                 return;
1082             }
```

mbr0wn, 2021-11-24:
I was able to confirm that the modification mentioned above will fix the issue. We have a fix in the pipeline.

crroush, 2021-11-24:
@mbr0wn while changing it to output ports does resolve the issue as you mentioned above.  There may be some inconsistency with rx/tx  ports and how they correspond to  input/output.  If you look [here](https://github.com/EttusResearch/uhd/blob/master/host/lib/rfnoc/radio_control_impl.cpp#L233-L246) you will see the setup for rx is looping based on ```get_num_input_ports()``` vs ```get_num_output_ports()```

mbr0wn, 2021-11-24:
You're right, excellent find, too! I'll have another look at our usage of get_num_input/output_ports, and see if we have more of those blunders.

crroush, 2021-11-24:
@mbr0wn one interesting thing, is that when that code is executed num_outputs equals num_inputs:
```
[WARNING] [0/Radio#0] Loading radio with SPC=1, num_inputs=2, num_outputs=2
```

But when the async call back occurs
```
[WARNING] [0/Radio#0] Num Input 0
[WARNING] [0/Radio#0] Num Output 2
```

mbr0wn, 2021-11-24:
I found some more comparison issues. We need to internally review the fix.

> @mbr0wn one interesting thing, is that when that code is executed num_outputs equals num_inputs:

This is an idiosyncrasy of the X300 radio block. The FPGA image is generic enough to do 2 TX, 2 RX, or both. On initialization, we first check the capabilities of the block itself (that's the first log statement). When you plug in a daughtercard (like TwinRX, or UBX) you are physically limiting the number of available channels (e.g., 2RX/0TX, or 1RX/1TX). The X300-specific code will, in a later step, check the capabilities of the dboard, and then reduce the number of actual channels. That's also why the second bug you found couldn't be found by testing on our images; it would accidentally work until someone actually reduces the number of channels in the FPGA.

mbr0wn, 2021-11-24:
FYI, the reason the X300 is the only USRP that does this is because we don't know which dboard is used when we build the FPGA image. All other RFNoC devices have fixed numbers of channels, and we don't build channels "to spare" for those.

mbr0wn, 2021-11-24:
This is what we might be publishing:
```patch
commit aa1c1f2abcd81a3b1143ead07592d2de3e023cf8
Author: Martin Braun <martin.braun@ettus.com>
Date:   Wed Nov 24 09:52:00 2021 +0100

    rfnoc: radio: Fix async message handling channel checks
    
    The async message handler and the async message validator would
    erroneously compare channel numbers for RX async messages with the
    number of valid TX channels. On TwinRX, where there are zero TX
    channels, this would always fail. Elsewhere in the code, the comparisons
    for TX and RX channels mixed up input and output ports.
    
    The second issue is that the comparison made was a "greater than" rather
    than "greater or equal".
    
    The effect of these two bugs was that potentially, we could have
    accepted async messages for an invalid port N, where N is the number of
    valid ports of this block, and that for TwinRX/X300 users, async
    messages on channel 1 would not get accepted (they would, however, get
    accepted for channel 0 because of the second issue). This includes
    overrun handling, which was broken for channel 1 and 3 on an X300.
    
    Another effect of the bug was that EPIDs for async messages weren't
    always programmed correctly.

diff --git a/host/lib/rfnoc/radio_control_impl.cpp b/host/lib/rfnoc/radio_control_impl.cpp
index 5bc16f7cd..3c4c28d11 100644
--- a/host/lib/rfnoc/radio_control_impl.cpp
+++ b/host/lib/rfnoc/radio_control_impl.cpp
@@ -105,7 +105,7 @@ radio_control_impl::radio_control_impl(make_args_ptr make_args)
                 return;
             }
             const size_t port = src.instance;
-            if (port > get_num_output_ports()) {
+            if (port >= get_num_output_ports()) {
                 RFNOC_LOG_WARNING("Received stream command to invalid output port!");
                 return;
             }
@@ -126,7 +126,7 @@ radio_control_impl::radio_control_impl(make_args_ptr make_args)
                 get_mb_controller()->get_timekeeper(0)->get_time_now()
                 + uhd::time_spec_t(OVERRUN_RESTART_DELAY);
             const size_t port = src.instance;
-            if (port > get_num_output_ports()) {
+            if (port >= get_num_output_ports()) {
                 RFNOC_LOG_WARNING("Received stream command to invalid output port!");
                 return;
             }
@@ -216,7 +216,7 @@ radio_control_impl::radio_control_impl(make_args_ptr make_args)
     RFNOC_LOG_TRACE("Sending async messages to EPID "
                     << regs().get_src_epid() << ", remote port " << regs().get_port_num()
                     << ", xbar port " << xbar_port);
-    for (size_t tx_chan = 0; tx_chan < get_num_output_ports(); tx_chan++) {
+    for (size_t tx_chan = 0; tx_chan < get_num_input_ports(); tx_chan++) {
         // Set the EPID and port of our regs() object (all async messages go to
         // the same location)
         _radio_reg_iface.poke32(
@@ -230,7 +230,7 @@ radio_control_impl::radio_control_impl(make_args_ptr make_args)
             regmap::SWREG_TX_ERR + regmap::SWREG_CHAN_OFFSET * tx_chan,
             tx_chan);
     }
-    for (size_t rx_chan = 0; rx_chan < get_num_input_ports(); rx_chan++) {
+    for (size_t rx_chan = 0; rx_chan < get_num_output_ports(); rx_chan++) {
         // Set the EPID and port of our regs() object (all async messages go to
         // the same location)
         _radio_reg_iface.poke32(
@@ -973,7 +973,7 @@ bool radio_control_impl::async_message_validator(
         return false;
     }
     if (addr_base == regmap::SWREG_RX_ERR) {
-        if (chan > get_num_output_ports()) {
+        if (chan >= get_num_output_ports()) {
             return false;
         }
         switch (code) {
@@ -986,7 +986,7 @@ bool radio_control_impl::async_message_validator(
         }
     }
     if (addr_base == regmap::SWREG_TX_ERR) {
-        if (chan > get_num_input_ports()) {
+        if (chan >= get_num_input_ports()) {
             return false;
         }
         switch (code) {
@@ -1039,7 +1039,7 @@ void radio_control_impl::async_message_handler(
     }
     switch (addr_base + addr_offset) {
         case regmap::SWREG_TX_ERR: {
-            if (chan > get_num_input_ports()) {
+            if (chan >= get_num_input_ports()) {
                 RFNOC_LOG_WARNING(
                     "Cannot process TX-related async message to invalid chan " << chan);
                 return;
@@ -1075,7 +1075,7 @@ void radio_control_impl::async_message_handler(
             break;
         }
         case regmap::SWREG_RX_ERR: {
-            if (chan > get_num_input_ports()) {
+            if (chan >= get_num_output_ports()) {
                 RFNOC_LOG_WARNING(
                     "Cannot process RX-related async message to invalid chan " << chan);
                 return;
```

closed 2021-11-30 by atrnati