EttusResearch/uhd #532: Repeated calls to rx_streamer->issue_stream_cmd() possible memory leak

skojor, 2021-11-25:
## Issue Description ##
Using mode STREAM_MODE_NUM_SAMPS_AND_MORE (or NUM_SAMPS_AND_DONE) and calling repeatedly issue_stream_cmd() in a receiving thread causes memory usage to slowly increase until embedded OS resources are depleted, whereas the program crashes. If mode STREAM_MODE_START_CONTINUOUS is used and called once, memory usage is stable. Tested on an E310 radio. 

## Setup Details ##
UHD 4.1.0.2-0-g8ce6e64f
FPGA v. 6.0 (git hash d5c2750.clean)
E310_sg3 (s/n 31d7848) using 4.1.0.2 image (downloaded with uhd_images_downloader after building 4.1.0.2 from source)
Cross-compiling with SDK and Ubuntu 20.04.
First tested on custom FPGA with radio and DDC block, later also reproduced the behavior on the original FPGA image without DDC block.

## Expected Behavior ##
Memory usage stable no matter how many times issue_stream_cmd() is called.

## Actual Behaviour ##
Program crashes after some hours, when system runs out of memory.

## Steps to reproduce the problem ##
Below example code to reproduce the problem. If receiverLoop1() is started in thread memory usage increases. receiverLoop2() shows stable memory usage.

## Additional Information ##

```
#include <uhd/exception.hpp>
#include <uhd/rfnoc/defaults.hpp>
#include <uhd/rfnoc/radio_control.hpp>
#include <uhd/rfnoc_graph.hpp>
#include <uhd/types/sensors.hpp>
#include <uhd/utils/thread.hpp>
#include <boost/thread.hpp>
#include <chrono>
#include <complex>
#include <csignal>

uhd::rx_streamer::sptr rx_stream;

void receiverLoop1()
{
    int sampsPerBuff = 4096;
    uhd::rx_metadata_t md;
    uhd::stream_cmd_t stream_cmd(uhd::stream_cmd_t::STREAM_MODE_NUM_SAMPS_AND_MORE);

    stream_cmd.num_samps  = size_t(sampsPerBuff);
    stream_cmd.stream_now = true;
    stream_cmd.time_spec  = uhd::time_spec_t();

    size_t sampsReceived;
    std::vector<std::complex<float>> buffer(sampsPerBuff);

    for ( ;; ) {
        rx_stream->issue_stream_cmd(stream_cmd);
        sampsReceived = rx_stream->recv(&buffer.front(), sampsPerBuff, md, 0.3, false);
        std::this_thread::sleep_for(std::chrono::milliseconds(1));
    }
}

void receiverLoop2()
{
    int sampsPerBuff = 4096;
    uhd::rx_metadata_t md;
    uhd::stream_cmd_t stream_cmd(uhd::stream_cmd_t::STREAM_MODE_START_CONTINUOUS);

    stream_cmd.num_samps  = size_t(sampsPerBuff);
    stream_cmd.stream_now = true;
    stream_cmd.time_spec  = uhd::time_spec_t();

    size_t sampsReceived;
    std::vector<std::complex<float>> buffer(sampsPerBuff);

    rx_stream->issue_stream_cmd(stream_cmd);

    for ( ;; ) {
        sampsReceived = rx_stream->recv(&buffer.front(), sampsPerBuff, md, 0.3, false);
    }
}

int main() {
    int radio_id = 0, radio_chan = 0;
    double rate = 10e6;

    uhd::rfnoc::rfnoc_graph::sptr graph;
    uhd::rfnoc::block_id_t *radio_ctrl_id;
    uhd::rfnoc::radio_control::sptr radio_ctrl;
    boost::thread *receiverThread;

    std::string args = "master_clock_rate=" + std::to_string(rate);
    graph = uhd::rfnoc::rfnoc_graph::make(args);
    radio_ctrl_id = new uhd::rfnoc::block_id_t(0, "Radio", radio_id);
    radio_ctrl = graph->get_block<uhd::rfnoc::radio_control>(*radio_ctrl_id);
    auto edges = graph->enumerate_static_connections();

    uhd::device_addr_t stream_args_args;
    uhd::stream_args_t stream_args("fc32", "sc16");
    stream_args.args        = stream_args_args;
    rx_stream = graph->create_rx_streamer(1, stream_args);

    graph->connect(radio_ctrl->get_block_id(), radio_chan, rx_stream, 0);
    graph->commit();

    radio_ctrl->set_rx_iq_balance(false, radio_chan);
    radio_ctrl->set_rx_agc(false, radio_chan);
    radio_ctrl->set_rx_gain(34, radio_chan);
    radio_ctrl->set_rx_antenna("TX/RX", radio_chan);
    radio_ctrl->set_rate(rate);
    radio_ctrl->set_rx_bandwidth(56e6, radio_chan);
    radio_ctrl->set_rx_frequency(100e6, radio_chan);

    receiverThread = new boost::thread(receiverLoop1);

    for (;;) std::this_thread::sleep_for(std::chrono::milliseconds(10));

    return 0;
}
```

mbr0wn, 2021-11-26:
Many thanks for the report!

mbr0wn, 2022-01-10:
@skojor do you recall if you were getting many overruns?

skojor, 2022-01-11:
@mbr0wn No overruns at all, the next issue_stream_cmd would not be sent until all data was read.

mbr0wn, 2022-01-11:
OK thanks for confirming. I used `massif` to track down the culprit. We'll need to do some work to figure out how to correctly fix this.

Mostly for myself, here's some details:

The massif output is a bit misleading:
[image]
...but the problem is `_resp_queue` in `ctrlport_endpoint.cpp`. It only gets emptied upon request -- and we're not requesting an ACK confirmation. So in this application, it'll fill up.

mbr0wn, 2022-01-11:
I think we can just always call wait_for_ack() with a zero-timeout and allow it to not find the ack in that case.

mbr0wn, 2022-01-11:
If you're feeling adventurous, this is the code we are internally reviewing:
```patch
From 526be01cd252ce53e2ce375ffb0c8d6b89a1c335 Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Tue, 11 Jan 2022 19:56:31 +0100
Subject: [PATCH] rfnoc: Always clear response queue in ctrlport_endpoint

This changes the behaviour of ctrlport_endpoint (the register interface
for block controllers) to always check for an ACK after doing a poke or
poll of any kind. Previously, the behaviour was to only check for an ACK
if the policy was set that way, or if the user requested the ACK to be
received.

The problem with the former approach was that if many pokes were
performed without ever requesting an ACK or a poll, the response queue
would fill up without ever getting emptied, eventually draining the
available heap space. Note that this is not a memory leak in the usual
sense, as the response queue was correctly holding on to the response
packets.

With this change, ctrlport_endpoint::wait_for_ack() now receives
a require_ack parameter. If it is false, the behaviour of wait_for_ack()
is changed as follows:
- If the response queue is empty, immediately return with an empty
  response payload object.
- Otherwise, continue reading elements out of the response queue until
  it is either depleted (in which case the previous rule kicks in), or
  we find the ACK corresponding to the command previously sent out.

Note that this replicates the corresponding behaviour in UHD 3 (see
ctrl_iface_impl::wait_for_ack()).
---
 host/lib/rfnoc/ctrlport_endpoint.cpp | 45 ++++++++++++++++++----------
 1 file changed, 29 insertions(+), 16 deletions(-)

diff --git a/host/lib/rfnoc/ctrlport_endpoint.cpp b/host/lib/rfnoc/ctrlport_endpoint.cpp
index c544a3d4e..a34ea3072 100644
--- a/host/lib/rfnoc/ctrlport_endpoint.cpp
+++ b/host/lib/rfnoc/ctrlport_endpoint.cpp
@@ -67,9 +67,8 @@ public:
         // Send request
         auto request = send_request_packet(OP_WRITE, addr, {data}, timestamp);
         // Optionally wait for an ACK
-        if (ack || _policy.force_acks) {
-            wait_for_ack(request);
-        }
+        const bool require_ack = ack || _policy.force_acks;
+        wait_for_ack(request, require_ack);
     }
 
     void multi_poke32(const std::vector<uint32_t> addrs,
@@ -104,9 +103,8 @@ public:
         // Send request
         auto request = send_request_packet(OP_BLOCK_WRITE, first_addr, data, timestamp);
         // Optionally wait for an ACK
-        if (ack || _policy.force_acks) {
-            wait_for_ack(request);
-        }
+        const bool require_ack = ack || _policy.force_acks;
+        wait_for_ack(request, require_ack);
         */
     }
 
@@ -117,7 +115,7 @@ public:
         auto request = send_request_packet(OP_READ, addr, {uint32_t(0)}, timestamp);
 
         // Wait for an ACK
-        auto response = wait_for_ack(request);
+        auto response = wait_for_ack(request, true);
         return response.data_vtr[0];
[... 54 lines cut ...]
     {
         auto resp_ready = [this]() -> bool { return !_resp_queue.empty(); };
         while (true) {
             std::unique_lock<std::mutex> lock(_mutex);
+            // Allocate return value
+            ctrl_payload rx_ctrl;
             // Wait until there is a response in the response queue
             if (!resp_ready()) {
+                if (!require_ack) {
+                    return rx_ctrl;
+                }
                 // If we're waiting for a timed command or if we have a
                 // command in the queue, use the MASSIVE_TIMEOUT instead
                 auto timeout_time = start_timeout(
@@ -432,7 +446,6 @@ private:
                 }
             }
             // Extract the response packet
-            ctrl_payload rx_ctrl;
             response_status_t resp_status;
             std::tie(rx_ctrl, resp_status) = _resp_queue.front();
             _resp_queue.pop();
-- 
2.25.1
```

skojor, 2022-01-11:
Many thanks for your feedback. Great news! If time permits I will test your
patch early next week.

Anyway I assume this patch will be included in a near future release.

Regards,
Jorn Skorstad

closed 2022-01-13 by atrnati