EttusResearch/uhd #533: Real/Imaginary Components or Magnitude and Phase?

johnwstanford, 2021-11-27:
The names of the registers here suggest that the IQ balance in `rx_frontend_core_3000` should be stored as a magnitude and phase, but we're storing the real and imaginary components.  Do we want to take the phase and magnitude of the complex number or are we basically just using `std::complex` like a `std::pair<double, double>`?  I think if this isn't a mistake, it's at least a confusing choice of names/types.

https://github.com/EttusResearch/uhd/blob/ef94c93a0822beda2f8c691bd4390727f7355322/host/lib/usrp/cores/rx_frontend_core_3000.cpp#L169

mbr0wn, 2021-11-29:
@johnwstanford It's a little more complicated than that. The mag/phase nomenclature is also incorrect. See here:

https://files.ettus.com/manual/page_calibration.html#calibration_fe_corr_iq_offset

However, this is something we've had for so long that at this time, it's more trouble to change than the payoff. FYI, the 'mag' and 'phase' names have some background in IQ impairment correction math, but it's really obscure at this point. See also here: https://github.com/EttusResearch/uhd/blob/master/fpga/usrp3/lib/rfnoc/blocks/rfnoc_block_radio/rx_frontend_gen3.v#L40-L50

So yeah, both names and types are confusing, but we won't fix it. The docs will have to suffice.

closed 2021-11-29 by mbr0wn

johnwstanford, 2021-11-30:
Ok.  No worries.  I'll check out that article.  Thanks.