EttusResearch/uhd #534: Using 4.1.0.4, "lo_locked" not found in RX frontend sensors

pfeatherstone, 2021-11-30:
## Issue Description ##
I'm using libuhd 4.1.0.4 with an E320. I'm running example program ```rx_samples_to_file``` and it's not outputting any samples.
After some debugging, i've found that ```usrp->get_rx_sensor_names(channel)``` returns ``` `ad9361_temperature`, `rssi`, `lo_lock` ```, and so does not include "lo_locked" in one of its sensors.
So the function ```check_locked_sensor``` returns ```false```. So no wonder the capture doesn't work.

Now after carefully reading the output of ```uhd_usrp_probe``` i find in the following:
```
|     _____________________________________________________
|    /
|   |       RX Dboard: 0/Radio#0
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 0
|   |   |   Name: E3xx
|   |   |   Antennas: RX2, TX/RX
|   |   |   Freq range: 70.000 to 6000.000 MHz
|   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   Bandwidth range: 20000000.0 to 40000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
|   |     _____________________________________________________
|   |    /
|   |   |       RX Frontend: 1
|   |   |   Name: E3xx
|   |   |   Antennas: RX2, TX/RX
|   |   |   Freq range: 70.000 to 6000.000 MHz
|   |   |   Gain range PGA: 0.0 to 76.0 step 1.0 dB
|   |   |   Bandwidth range: 20000000.0 to 40000000.0 step 0.0 Hz
|   |   |   Connection Type: IQ
|   |   |   Uses LO offset: No
```

There is no section ```Sensors: lo_locked, ad9361_temperature``` etc.

Do we expect to see "lo_locked" with version 4.1.0.4 ?

pfeatherstone, 2021-11-30:
with 4.1.0.4, here is the output of ```uhd_usrp_probe --tree | grep sensors``` :
```
/mboards/0/sensors
/mboards/0/sensors/ref_locked
/mboards/0/sensors/gps_locked
/mboards/0/sensors/fan
/mboards/0/sensors/temp_fpga
/mboards/0/sensors/temp_internal
/mboards/0/sensors/temp_rf_channelA
/mboards/0/sensors/temp_rf_channelB
/mboards/0/sensors/temp_main_power
/mboards/0/sensors/gps_gpgga
/mboards/0/sensors/gps_sky
/mboards/0/sensors/gps_time
/mboards/0/sensors/gps_tpv
/blocks/0/Radio#0/rx_frontends/0/sensors
/blocks/0/Radio#0/rx_frontends/0/sensors/ad9361_temperature
/blocks/0/Radio#0/rx_frontends/0/sensors/rssi
/blocks/0/Radio#0/rx_frontends/0/sensors/lo_lock
/blocks/0/Radio#0/rx_frontends/1/sensors
/blocks/0/Radio#0/rx_frontends/1/sensors/ad9361_temperature
/blocks/0/Radio#0/rx_frontends/1/sensors/rssi
/blocks/0/Radio#0/rx_frontends/1/sensors/lo_lock
/blocks/0/Radio#0/tx_frontends/0/sensors
/blocks/0/Radio#0/tx_frontends/0/sensors/ad9361_temperature
/blocks/0/Radio#0/tx_frontends/1/sensors
/blocks/0/Radio#0/tx_frontends/1/sensors/ad9361_temperature
```

So "lo_locked" is not found.

With 3.14.1, the output of "uhd_usrp_probe --tree | grep sensors" is:
```
/mboards/0/sensors
/mboards/0/sensors/gps_gpgga
/mboards/0/sensors/temp_internal
/mboards/0/sensors/gps_tpv
/mboards/0/sensors/temp_main_power
/mboards/0/sensors/ref_locked
/mboards/0/sensors/temp_rf_channelB
/mboards/0/sensors/temp_rf_channelA
/mboards/0/sensors/fan
/mboards/0/sensors/gps_sky
/mboards/0/sensors/gps_locked
/mboards/0/sensors/gps_time
/mboards/0/sensors/temp_fpga
/mboards/0/dboards/A/tx_frontends/0/sensors
/mboards/0/dboards/A/tx_frontends/0/sensors/lo_locked
/mboards/0/dboards/A/tx_frontends/0/sensors/ad9361_temperature
/mboards/0/dboards/A/tx_frontends/1/sensors
/mboards/0/dboards/A/tx_frontends/1/sensors/lo_locked
/mboards/0/dboards/A/tx_frontends/1/sensors/ad9361_temperature
/mboards/0/dboards/A/rx_frontends/0/sensors
/mboards/0/dboards/A/rx_frontends/0/sensors/lo_locked
/mboards/0/dboards/A/rx_frontends/0/sensors/ad9361_temperature
/mboards/0/dboards/A/rx_frontends/0/sensors/lo_lock
/mboards/0/dboards/A/rx_frontends/0/sensors/rssi
/mboards/0/dboards/A/rx_frontends/1/sensors
/mboards/0/dboards/A/rx_frontends/1/sensors/lo_locked
/mboards/0/dboards/A/rx_frontends/1/sensors/ad9361_temperature
/mboards/0/dboards/A/rx_frontends/1/sensors/lo_lock
/mboards/0/dboards/A/rx_frontends/1/sensors/rssi
```

pfeatherstone, 2021-11-30:
I imagine the absence of "lo_locked" is what's causing my issue

pfeatherstone, 2021-11-30:
OK, now I'm trying to rollback to a previous version but ```mender -rootfs``` no longer works... Are there any updated docs anywhere ?

pfeatherstone, 2021-11-30:
The options for mender are now:
```
root@ni-e320-XXXXXX:~# mender --help
NAME:
   mender - manage and start the Mender client.

USAGE:
   [global options] command [command options] [arguments...]

VERSION:
   a78b45e  runtime: go1.12.9

DESCRIPTION:
   mender integrates both the mender daemon and commands for manually performing tasks performed by the daemon (see list of COMMANDS below).

Global flag remarks:
  - Supported log levels incudes: 'debug', 'info', 'warning', 'error', 'panic' and 'fatal'.

COMMANDS:
   bootstrap       Perform bootstrap and exit.
   check-update    Force update check.
   commit          Commit current Artifact. Returns (2) if no update in progress.
   daemon          Start the client as a background service.
   install         Mender Artifact to install - local file or a `URL`.
   rollback        Rollback current Artifact. Returns (2) if no update in progress.
   send-inventory  Force inventory update.
   setup           Perform configuration setup - 'mender setup --help' for command options.
   snapshot        Create filesystem snapshot -'mender snapshot --help' for more.
   show-artifact   Print the current artifact name to the command line and exit.
   help            Shows a list of commands or help for one command

GLOBAL OPTIONS:
   --config FILE, -c FILE           Configuration FILE path. (default: "/etc/mender/mender.conf")
   --fallback-config FILE, -b FILE  Fallback configuration FILE path. (default: "/var/lib/mender/mender.conf")
   --data DIR, -d DIR               Mender state data DIRECTORY path. (default: "/var/lib/mender")
   --log-file FILE, -L FILE         FILE to log to.
   --log-level level, -l level      Set logging level. (default: "info")
   --log-modules value, -m value    -log-modules is accepted for compatibility but has no effect
   --trusted-certs FILE, -E FILE    Trusted server certificates FILE path.
   --forcebootstrap, -F             Force bootstrap. (default: false)
   --no-syslog                      Disable logging to syslog. (default: false)
   --skipverify                     Skip certificate verification. (default: false)
   --help, -h                       show help (default: false)
   --version, -v                    print the version (default: false)
```

mbr0wn, 2021-12-01:
@pfeatherstone This is definitely an issue and you're right there should be an lo_locked sensor. I was curious why we had both sensors in UHD 3 -- turns out we hand-forward (in C++) the lo_lock sensor to the other one. The UHD 4 solution is the better engineered solution, but it did do this bad breakage.

FYI, you can skip the check with `--skip-lo` in `rx_samples_to_file`. But that's probably not your main issue.

pfeatherstone, 2021-12-01:
Cool. So do you suggest not using the latest release then? When reverting back to 3.14, i have no issues, except for the occasional "CODEC loopback test failed"  issue which is immensely annoying and requires me to reflash the FPGA image and reboot some arbitrary number of times.

mbr0wn, 2021-12-01:
I'll be honest, I have no great suggestion other than to wait until the release which will have this fix.

One thing you could do is to use a UHD4 image, and modify MPM. It's a hack, of course, don't get me wrong! But in case you like to not reboot because of CODED loopback test failures, but still like an `lo_locked` sensor, you can ssh into your E320 (or use serial console), edit the file `dboard_manager/neon.py` (it's somewhere under `/usr/lib/pythonXXX/site-packages/usrp_mpm`, and modify the sensor map around line 36 to look like this:
```python
    rx_sensor_callback_map = {
        'ad9361_temperature': 'get_catalina_temp_sensor',
        'rssi' : 'get_rssi_sensor',
        # For backward compatibility reasons we have the same sensor with two
        # different names
        'lo_lock' : 'get_lo_lock_sensor',
        'lo_locked' : 'get_lo_lock_sensor',
    }
    tx_sensor_callback_map = {
        'ad9361_temperature': 'get_catalina_temp_sensor',
        # For backward compatibility reasons we have the same sensor with two
        # different names
        'lo_lock' : 'get_lo_lock_sensor',
        'lo_locked' : 'get_lo_lock_sensor',
    }
```

I just typed this up (not tested yet), but if you don't mind doing some minor SW surgery  on the device, it could unlock your problem.

mbr0wn, 2021-12-01:
Note that the hard-coded `lo_locked` sensor name in `rx_samples_to_file` is also something that can be changed, although I think it shouldn't.

mbr0wn, 2021-12-01:
...if you choose to give this a spin ^^^, you need to reboot MPM (or your device) after editing the file for the changes to take effect.

pfeatherstone, 2021-12-01:
Thank you very much. I think for now I'll stick to 3.14 then while waiting for next release. I don't mind hacking software but when there is an expensive piece of hardware involved, I'd rather not.

pfeatherstone, 2022-01-05:
@mbr0wn Does the latest release fix this? I thought i would ask before updating everything as that takes a bit of time if a rollback is required.

mbr0wn, 2022-01-05:
No, sorry. The latest release had other priorities.

mbr0wn, 2022-01-07:
@pfeatherstone While you're right we should have the `lo_locked` sensor, I checked `rx_samples_to_file` again, and it will happily continue running without that sensor. When `check_locked_sensor()` returns `false`, that's not a critical error. Can you maybe increase the log level and paste the entire output?

mbr0wn, 2022-01-07:
Note that we have a fix incoming, but it's for the missing lo_locked sensor. If your issue is different, then I suggest opening a new issue.

closed 2022-01-10 by atrnati