EttusResearch/uhd #535: PATH too long in windows installer

Ivarvdb, 2021-11-30:
## Issue Description ##

[image]
[image]

UDH windows installer crashes after warning display. Can't run any of the example executables in the ln the examples folder.

## Setup Details ##

- Windows 10 machine
- Latest version of UDH (4.1.0.4), happens with other versions as well.

## Expected Behavior ##

Complete install of the UDH software and working examples.

## Actual Behaviour ##

Got the above-mentioned warning and install does not finish properly

## Steps to reproduce the problem ##

Use latest windows installer on windows 10 system

## Additional Information ##

wordimont, 2021-11-30:
How long is the PATH environment variable on your system? Does it exceed what Windows allows?

Ivarvdb, 2021-11-30:
> How long is the PATH environment variable on your system? Does it exceed what Windows allows?

The total size of the user path variable is 1486 characters. But adding new entries to the path works just fine

wordimont, 2021-11-30:
It might be the system PATH variable that's too long. How long is that one?

Ivarvdb, 2021-12-01:
> It might be the system PATH variable that's too long. How long is that one?

That one is 1148 characters long.

wordimont, 2021-12-01:
Thanks for the information and for reporting this issue. I've read that 2048 tends to be a when things start to go wrong with the PATH on Windows. Perhaps the total PATH length (system+user = 2634) is causing the problem. We'll have to look into how the installer PATH update works and see if it can be fixed. 

In the mean time, you might review your PATH variables to see if you can reduce their length and see if that resolves the issue. Look for duplicates between system and user, or old entries that are no longer needed that can be removed. There are also tools you can use and other tricks to shorten and simplify paths that would otherwise be too long. Please let us know if this resolves the issue for you.