EttusResearch/uhd #538: RFNoC OOT block integration problem

epetragl, 2021-12-03:
## Issue Description ##
I'm currently attempting to realize the RFNoC example described in the "Out-of-tree Modules" section of this web page. 
https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0

I've followed step by step the instructions until the make x310_rfnoc_image_core command. Since I'm targeting 
a N320 device I had to do some modifications inside the rfnoc-demo directory. 
Below the additional step I've performed

I've created a n320_rfnoc_core.yml inside rfnoc-demo/icores 
This file was created from the original n320_rfnoc_core.yml located in
uhd/fpga/usrp3/top/n3xx/ with the following modifications: 

-- creation of a new endpoint
ep4: # Stream endpoint name
ctrl: False # Endpoint passes control traffic
data: True # Endpoint passes data traffic
buff_size: 32768 # Ingress buffer size for data

-- definition of the gain block:
gain0:
block_desc: ’gain.yml’

-- connection between the block and the endpoint
- { srcblk: ep4, srcport: out0, dstblk: gain0, dstport: in }
- { srcblk: gain0, srcport: out, dstblk: ep4, dstport: in0 }

Then I've modified the CMakeLists.txt inside rfnoc-demo by 
replacing all the occurrences of X310 with N320. 
I've deleted the content of the rfnoc-demo/build directory 
and I've performed the following commands :

cd ~/rfnoc-demo/build
cmake -DUHD_FPGA_DIR=<UHD4.1 repo>/fpga/ ../
make n320_rfnoc_image_core

I've loaded the generated bitstream inside the n320. At this point 
I've executed the following Python script :

import sys
import re
import uhd
import numpy as np
import matplotlib
from matplotlib import pyplot

graph = uhd.rfnoc.RfnocGraph("addr=192.168.0.1")
for edge in graph.enumerate_static_connections():
    print(edge.to_string())
radio_noc_block = graph.get_block("0/Radio#0")
radio_block = uhd.rfnoc.RadioControl(radio_noc_block)
   
pkt_length = 4

radio_noc_block = graph.get_block("0/Radio#0")
radio_block = uhd.rfnoc.RadioControl(radio_noc_block)
radio_block.set_properties(f'spp={pkt_length}', 0)
    
sa = uhd.usrp.StreamArgs("fc32", "sc16")
sa.args = "spp=" + str(pkt_length)

ddc_block = uhd.rfnoc.DdcBlockControl(graph.get_block("0/DDC#0"))
    
rx_streamer = graph.create_rx_streamer(1, sa)

graph.connect("0/Radio#0", 0, "0/DDC#0", 0, False)
print("Graph with gain_block \n")
graph.connect("0/DDC#0", 0, "0/Block#0", 0, False)
graph.connect("0/Block#0", 0, rx_streamer, 0)
graph.commit()
    
radio_block.set_rx_frequency(2500e6, 0)
radio_block.set_rx_gain(40, 0)
radio_block.set_rx_antenna("RX2", 0)  
radio_block.set_rate(125e6)

ddc_block.set_input_rate(125e6, 0)
ddc_block.set_output_rate(5e6, 0)

num_samples = int(ddc_block.get_output_rate(0) * 3)
radio_data = np.zeros((1, num_samples), dtype="complex64")
    
stream_cmd = uhd.types.StreamCMD(uhd.types.StreamMode.num_done) 
stream_cmd.num_samps = num_samples
stream_cmd.stream_now = True
        
rx_streamer.issue_stream_cmd(stream_cmd)
num_samples_received = rx_streamer.recv(radio_data, uhd.types.RXMetadata(), 15.0)
print("Nb of samples received = " + str(num_samples_received))

matplotlib.pyplot.plot(range(np.size(radio_data[0])), np.real(radio_data[0]), "g", range(np.size(radio_data[0])), np.imag(radio_data[0]), "r")
pyplot.show()

Below you can see the output of this script :

[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.1.0.4-118-g06a9ded7
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.0.1,type=n3xx,product=n320,serial=31A5C61,fpga=HG,claimed=False,addr=192.168.0.1
[INFO] [MPM.PeriphManager] init() called with device args `fpga=HG,mgmt_addr=192.168.0.1,product=n320,clock_source=internal,time_source=internal'.
[INFO] [MPM.Rhodium-0] init() called with args `fpga=HG,mgmt_addr=192.168.0.1,product=n320,clock_source=internal,time_source=internal'
[INFO] [MPM.Rhodium-1] init() called with args `fpga=HG,mgmt_addr=192.168.0.1,product=n320,clock_source=internal,time_source=internal'
[INFO] [MPM.Rhodium-0.init.LMK04828] LMK initialized and locked!
[INFO] [MPM.Rhodium-0.DAC37J82] DAC PLL Locked!
[INFO] [MPM.Rhodium-0.AD9695] ADC PLL Locked!
[INFO] [MPM.Rhodium-0.init] JESD204B Link Initialization & Training Complete
[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2500000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2500000
[WARNING] [RFNOC::BLOCK_FACTORY] Could not find block with Noc-ID 0xb16, 0xffff
0/SEP#0:0==>0/DUC#0:0
0/DUC#0:0==>0/Radio#0:0
0/Radio#0:0==>0/DDC#0:0
0/DDC#0:0==>0/SEP#0:0
0/SEP#1:0==>0/DUC#1:0
0/DUC#1:0==>0/Radio#1:0
0/Radio#1:0==>0/DDC#1:0
0/DDC#1:0==>0/SEP#1:0
0/SEP#2:0==>0/Replay#0:0
0/Replay#0:0==>0/SEP#2:0
0/SEP#3:0==>0/Replay#0:1
0/Replay#0:1==>0/SEP#3:0
0/SEP#4:0==>0/Block#0:0
0/Block#0:0==>0/SEP#4:0
Graph with gain_block 

[WARNING] [UDP] The send buffer could not be resized sufficiently.
Target sock buff size: 2500000 bytes.
Actual sock buff size: 1048576 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.wmem_max=2500000
[WARNING] [0/Radio#0] Coercing requested sample rate from 125 MHz to 200 MHz, the closest possible rate.
[INFO] [MPM.Rhodium-0] init() called with args `fpga=HG,mgmt_addr=192.168.0.1,product=n320,clock_source=internal,time_source=internal,master_clock_rate=200000000.0,ref_clk_freq=25000000.0'
[INFO] [MPM.Rhodium-0.init.LMK04828] LMK initialized and locked!
[INFO] [MPM.Rhodium-0.DAC37J82] DAC PLL Locked!
[INFO] [MPM.Rhodium-0.AD9695] ADC PLL Locked!
[INFO] [MPM.Rhodium-0.init] JESD204B Link Initialization & Training Complete
Nb of samples received = 0

As you can see the static connections inside the FPGA look good, but the gain block seems to prevent any samples to reach the Rx_streamer. By directly connecting the 
DDC to the RX_streamer the N320 starts to produce data. My guess is, there might be something I've missed out on the Python script, but I cannot find any information
about this part of the RFNoC process. I'm running out of ideas, any input is welcome.
Thanks in advance!

## Setup Details ##
Vivado 2019.1
UHD 4.1
Ettus N320

## Expected Behavior ##
Receive samples after introducing the RFNoC OOT gain block between the DDC:0 and an RX_streamer

## Actual Behaviour ##
Zero samples received while using the RFNoC OOT gain block

## Additional Information ##
This issue the continuation of issue 528 (https://github.com/EttusResearch/uhd/issues/528) which was partially resolved. Unfortunately not all the points covered in issue 528 have not been solved. 
The behavior presented in this issue represent a blocking point for getting RFNoC to work fully.

f380cedric, 2021-12-06:
Could you verify that when you request your N320 to record samples, LED next to the RF SMA port RX2 turns green? Does issuing the `issue_stream_cmd` to the `ddc_block` instead of the `rx_streamer` block work? The outcome seems similar to #406.

wordimont, 2021-12-06:
I think @f380cedric is right. Another thing you could try is to change your graph to exclude the gain block (i.e., just do radio -> DDC -> rx_treamer). That will confirm that everything else is at least working.

epetragl, 2021-12-08:
> rx_streamer.issue_stream_cmd(stream_cmd)

Thank you @f380cedric for your suggestion:

In my python script I've replaced : 

rx_streamer.issue_stream_cmd(stream_cmd)

with 

ddc_block.issue_stream_cmd(stream_cmd)

and I started getting samples from my N320. 

Now I can move forward.

closed 2021-12-08 by epetragl