EttusResearch/uhd #539: when i want to build and make UHD in Vivado 2019.1 i faced with errors

stackprogramer, 2021-12-03:
when I want to build and make UHD in Vivado 2019.1 I faced errors 
How can solve it?
Before I built UHD in 2020.2 i had IP core errors now  i had:

```
[00:41:08] Current task: Connectivity Check +++ Current Phase: Starting
[00:41:09] Current task: Connectivity Check +++ Current Phase: Finished
[00:41:09] Starting Netlist Obfuscation Task
[00:41:11] Current task: Netlist Obfuscation +++ Current Phase: Starting
[00:41:11] Current task: Netlist Obfuscation +++ Current Phase: Finished
[00:41:11] Executing Tcl: place_design -directive ExtraNetDelay_high
[00:41:11] Starting Placer Command
[00:41:43] Current task: Placer +++ Current Phase: Starting
[00:41:44] Starting Placer Task
[00:41:44] Current task: Placer +++ Current Phase: Starting
[00:41:44] Current task: Placer +++ Current Phase: 1 Placer Initialization
[00:42:14] Current task: Placer +++ Current Phase: 1.1 Placer Initialization Netlist Sorting
[00:44:21] Current task: Placer +++ Current Phase: 1.2 IO Placement/ Clock Placement/ Build Placer Device
[00:46:57] Current task: Placer +++ Current Phase: 1.3 Build Placer Netlist Model
[00:47:00] Current task: Placer +++ Current Phase: 1.4 Constrain Clocks/Macros
[00:47:00] Current task: Placer +++ Current Phase: 2 Global Placement
[00:47:38] Current task: Placer +++ Current Phase: 2.1 Floorplanning

[00:51:50] Current task: Placer +++ Current Phase: Finished[00:51:50] Current task: Placer +++ Current Phase: 2.2 Global Placement Core
[00:51:50] Process terminated. Status: Failure

========================================================
Warnings:           1035
Critical Warnings:  47
Errors:             0

make[1]: *** [Makefile.x300.inc:127: bin] Error 1
make[1]: Leaving directory '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300'
make: *** [Makefile:90: X300_HG] Error 2

```

wordimont, 2021-12-03:
It looks like you're trying to build the X300_HG bitstream. I don't see any error messages, which sometimes happens if Vivado crashes. Since you mentioned you were using a different Vivado version previously, I suggest doing a clean build. If you made any modifications to the code, you might want to compile the FPGA from an unmodified release tag first just to make sure the build process is working. 

```
cd uhd/fpga/usrp3/top/x300
source setupenv.sh
make cleanall
make X310_HG
```

stackprogramer, 2021-12-05:
@wordimont 
Thanks very much finally I successfully built bitstream for x300.
```
[02:02:19] Current task: Write Bitstream +++ Current Phase: Starting
[02:02:19] Current task: Write Bitstream +++ Current Phase: Finished
[02:02:19] Process terminated. Status: Success

========================================================
Warnings:           1627
Critical Warnings:  111
Errors:             0

make[1]: Leaving directory '/home/sp/Documents/uhd-master/fpga/usrp3/top/x300'
Exporting bitstream files...
Generating LVBITX...
/bin/sh: 1: python: not found
Exporting build report...
Build DONE ... X310_HG

```

closed 2021-12-09 by wordimont