EttusResearch/uhd #543: Confusing Variable Names in `get_freq_and_freq_word`

johnwstanford, 2022-01-01:
The value we pass to `_set_freq` is `samp_rate_out.get()`, which is the output rate of the DUC (a.k.a the input rate to RC, a.k.a. the master clock rate; usually 200 MHz, at least on the X310).  However, the name of this argument in `_set_freq` is `input_rate`, which implies that it's the input rate to the DUC (before interpolation), since we're in the DUC controller.  This is what gets passed to `get_freq_and_freq_word`.

So we're giving it the output rate of the DUC but the variable names suggest that maybe it should be the input rate.  Is it correct as is with the output rate (200 MHz for X310) being passed in?

https://github.com/EttusResearch/uhd/blob/919043f305efdd29fbdf586e9cde95d9507150e8/host/lib/rfnoc/duc_block_control.cpp#L297

johnwstanford, 2022-01-01:
I guess it's a question of whether in the FPGA code, the phase increment happens before interpolation or other way around.  I don't know my way around the FPGA code well enough yet to see for myself.

I found another comment that suggests that the phase increment needs to be recalculated every time the input rate is updated, which suggests that phase increment happens first and passing the output sample rate into `get_freq_and_freq_word` might be a mistake.

mbr0wn, 2022-01-07:
In the DUC, phase increment/mixing happens after interpolation (or we wouldn't have the full frequency range available). The confusing variable name is because it is copy/pasta from the DDC block. Thanks for pointing this out.

closed 2022-01-07 by atrnati