EttusResearch/uhd #544: Freezing on overflow

helmet21, 2022-01-02:
## Issue Description ##
The FFT or whatever software I run will freeze as soon as a single overflow occurs.

## Setup Details ##
Running UHD_4.0.0.0 on a B210 under Fedora 34.

## Expected Behavior ##
Previously with different software versions I have been able to run the B210 at 56 msps without it freezing.

## Actual Behaviour ##
As soon as a single overflow occurs the spectrum will freeze.

## Steps to reproduce the problem ##
I run: uhd_fft -f 2440000000 -s 56000000 -g 34
The freezing happens at different points in time. Sometimes straight away, sometimes when I change the FFT size or sometimes by itself after a few seconds.

## Additional Information ##
If I lower the sample rate it will either stop happening or happen later.

Output from the terminal:
```
$ uhd_fft -f 2440000000 -s 56000000 -g 34
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[INFO] [UHD] linux; GNU C++ version 11.0.1 20210324 (Red Hat 11.0.1-0); Boost_107500; UHD_4.0.0.0
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Asking for clock rate 56.000000 MHz... 
[INFO] [B200] Actually got clock rate 56.000000 MHz.
Ogr::log :ERROR: usrp_source0 - In the last 750 ms, 820637297 overflows occurred.
```
I have tried with newer and older versions of UHD, building from source and from the repos but I keep encountering vast dependency issues and have been unable to install or build successfully. It would be great it someone has a working set up on Fedora 34 and could share the details, like version of UHD (and GNU Radio if they have it installed).

helmet21, 2022-01-02:
So I managed to solve it by building UHD_4.1.0.5 from source together with GNU Radio 3.9.4.0. It was just a big hassle with locating all the missing dependencies for Fedora34 since many of the names given in the instructions were outdated. This is the command I used for the dependencies:

`sudo dnf install fftw-devel cppunit-devel libusb-devel  guile boost-devel alsa-lib-devel numpy gsl-devel python-devel pygsl swig python-sphinx orc-devel orc-compiler doxygen python-docutils python-cheetah python-lxml guile-devel qwt-devel qwtplot3d-qt4-devel cmake xmlto graphviz zeromq-devel python-zmq SDL-devel cmake make gcc gcc-c++ boost-devel thrift-devel python-thrift python-mako gpsd-devel cppzmq-devel`

And because these packages weren't found I had to look up the names:

`sudo dnf install python-pyopengl python3-PyQt4-devel`

I hope that helps someone.

closed 2022-01-02 by helmet21

mbr0wn, 2022-01-10:
@helmet21 Thanks for closing this again (and reporting it in the first place). Yeah, we had fixed this a while ago. FYI your dependency list also includes the GNU Radio deps, and also includes deps for GR 3.8. It can be pared down if you want to optimize.