EttusResearch/uhd #546: Unknown property value "XC6SLX150" while building b205mini FPGA Image

johnwstanford, 2022-01-13:
## Issue Description ##
The FPGA image for the b205mini fails to build.

## Setup Details ##
Commit f23ab721
Ubuntu 21.04, ISE 14.7, Make 4.3, and Python 3.8.5

## Expected Behavior ##
Expect the FPGA image to successfully build

## Actual Behaviour ##
Fails with the following output:
```
ISE Version: Release 14.7 - xtclsh P.20131013 (lin64)
make -f Makefile.b205.inc bin NAME=B205mini DEVICE=XC6SLX150
make[1]: Entering directory '/home/john/github/uhd/fpga/usrp3/top/b2xxmini'
build-B205mini//b205.xise
xtclsh /home/john/github/uhd/fpga/usrp3/top/tcl/ise_helper.tcl ""
>>> Creating project: build-B205mini//b205.xise
Changed current working directory to the project directory:
"/home/john/github/uhd/fpga/usrp3/top/b2xxmini/build-B205mini" 
>>> Setting: Project[family] = Spartan6
>>> Setting: Project[device] = XC6SLX150
ERROR:TclTasksC:project_028: Unknown property value "XC6SLX150" specified for "PROP_DevDevice"
    while executing
"project set $key $opt"
    invoked from within
"if ![string compare $process "Project"] {
				project set $key $opt
			} else {
				project set $key $opt -process $process
			}"
    invoked from within
"if $state {
			set key $opt
			set state 0
		} else {
			puts ">>> Setting: $process\[$key\] = $opt"
			if ![string compare $process "Project"] {
				..."
    ("foreach" body line 2)
    invoked from within
"foreach opt $options {
		if $state {
			set key $opt
			set state 0
		} else {
			puts ">>> Setting: $process\[$key\] = $opt"
			if ![string compare $..."
    (procedure "set_props" line 6)
    invoked from within
"set_props "Project" $env(PROJECT_PROPERTIES)"
    invoked from within
"if [file isfile $env(ISE_FILE)] {
	puts ">>> Opening project: $env(ISE_FILE)"
	project open $env(ISE_FILE)
} else {
	puts ">>> Creating project: $env(..."
    (file "/home/john/github/uhd/fpga/usrp3/top/tcl/ise_helper.tcl" line 29)
make[1]: *** [../Makefile.common:52: build-B205mini//b205.xise] Error 1
make[1]: Leaving directory '/home/john/github/uhd/fpga/usrp3/top/b2xxmini'
make: *** [Makefile:73: B205mini] Error 2
```

## Steps to reproduce the problem ##
- Add `xtclsh` to path
- `source opt/Xilinx/14.7/ISE_DS/settings64.sh`
- `cd ${UHD_REPO}/fpga/usrp3/top/b2xxmini`
- `make B205mini`

wordimont, 2022-01-13:
That's a very early failure, when it's just setting up the ISE project. I'm running Ubuntu 20.04.3 LTS, Make 4.2.1, and Python 3.8.5 without issue.

You shouldn't have to add xtclsh to your path, since settings64.sh does that. Make sure you're running the expected xtclsh and not a different one from what settings64.sh expects.

Could you try running the commands manually to rule out our build process? If the step `project set device XC6SLX150` doesn't work then their might be something wrong with your ISE installation.

Here's an example. You should be able to run these commands by hand without error.
```
$ cd ${UHD_REPO}/fpga/usrp3/top/b2xxmini
$ make clean
$ source /opt/Xilinx/14.7/ISE_DS/settings64.sh
$ xtclsh
% mkdir build-B205mini
% cd build-B205mini
% project new b205.xise
b205
% project set family Spartan6
Spartan6
% project set device XC6SLX150
xc6slx150
% project set package csg484
csg484
% exit
```

johnwstanford, 2022-01-14:
Yeah, when I type them in by hand, I get the same failure message after `project set device XC6SLX150`.  When I open the ISE GUI and create a project through their wizard, I don't see XC6SLX150 as an option.  It only goes up to the XC6SLX75T.  Is the b205mini FPGA not supported by the free WebPack edition of ISE?  I could have sworn I read somewhere that it was.

wordimont, 2022-01-14:
Unfortunately, Xilinx does not support that FPGA in the WebPACK version. That explains the issue.

You can find a list of supported devices in the ISE 14.7 release notes.
https://www.xilinx.com/content/dam/xilinx/support/documentation/sw_manuals/xilinx14_7/irn.pdf

johnwstanford, 2022-01-15:
Ok.  Yeah, the makes sense, unfortunately :/ 

Not to go off topic too far, but do you know what they charge for an individual node locked license?  I can't find a clear answer on their website.  Vivado is about $3k, so I'd assume it's less than that since it's considered an obsolete product.

I've gotten to know the software pretty well lately and I'm trying to get my feet wet with the FPGA code.  I assumed it was best to start small with the b205mini, but what I really want is to build X310 images with custom RFNOC blocks, so maybe it's best to just get a Vivado license and skip straight to that.

johnwstanford, 2022-01-15:
Also, the build process for the B200mini works for me.  It has a XC6SLX75.  I used to not understand the difference between the B200mini and B205mini, but that seems to be the main difference.

wordimont, 2022-01-15:
No, I don't have pricing information. X310 does need a Vivado license. The E310 works with the free version of Vivado, but has a relatively small FPGA. You might also see if you can get an eval license if you want to experiment.

wordimont, 2022-01-19:
I will go ahead and close this issue since we've found the cause.

closed 2022-01-19 by wordimont