EttusResearch/uhd #547: Please support the new DPDK LTS 21.11

cpaelzer, 2022-01-13:
## Issue Description ##

Hello UHD project,
I wanted to report a build issue against the new [DPDK LTS release 21.11](https://doc.dpdk.org/guides/rel_notes/release_21_11.html) fails.

```
/<<PKGBUILDDIR>>/host/lib/include/uhdlib/transport/dpdk/udp.hpp:61:13: warning: "PKT_TX_IP_CKSUM" is deprecated
   61 |     mbuf->ol_flags = PKT_TX_IP_CKSUM | PKT_TX_IPV4;
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/<<PKGBUILDDIR>>/host/lib/include/uhdlib/transport/dpdk/udp.hpp:61:13: warning: "PKT_TX_IPV4" is deprecated
   61 |     mbuf->ol_flags = PKT_TX_IP_CKSUM | PKT_TX_IPV4;
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/<<PKGBUILDDIR>>/host/lib/transport/udp_dpdk_link.cpp: In member function ‘virtual void uhd::transport::udp_dpdk_link::release_send_buff(uhd::transport::frame_buff::uptr)’:
/<<PKGBUILDDIR>>/host/lib/transport/udp_dpdk_link.cpp:173:52: error: ‘struct rte_ether_hdr’ has no member named ‘d_addr’; did you mean ‘dst_addr’?
  173 |         rte_ether_addr_copy(&_remote_mac, &l2_hdr->d_addr);
      |                                                    ^~~~~~
      |                                                    dst_addr
/<<PKGBUILDDIR>>/host/lib/transport/udp_dpdk_link.cpp:174:50: error: ‘struct rte_ether_hdr’ has no member named ‘s_addr’; did you mean ‘dst_addr’?
  174 |         rte_ether_addr_copy(&local_mac, &l2_hdr->s_addr);
      |                                                  ^~~~~~
      |                                                  dst_addr
```

[Full logs for all architectures](https://launchpad.net/ubuntu/+source/uhd/4.1.0.5-1build1) can be found in this try to rebuild against the new DPDK.

## Setup Details and Steps to reproduce ##

No special setup needed in regard to hardware.
Just try to build it using [DPDK 21.11 in Ubuntu 22.04](https://launchpad.net/ubuntu/+source/dpdk/21.11-1) - right now that means [Jammy](https://cdimage.ubuntu.com/daily-live/current/) with [proposed](https://wiki.ubuntu.com/Testing/EnableProposed) enabled.

When building UHD in there it will fail to build as reported.

## Additional Information ##

IMHO we'd need something like [this change for 20.11](https://github.com/EttusResearch/uhd/commit/b17d2dfa988344debe63f1b5d7e8556fef87218f), but to now also include 21.11.

FYI: For now in Ubuntu UHD is on the tail end of the DPDK transition and for now I have to interim-disable DPDK support in those builds to get everything else moving for the upcoming release of Ubuntu 22.04

manderseck, 2022-01-20:
Thanks for reporting this issue @cpaelzer. We are aware of that and had an update on our list anyway, so we used the opportunity to leverage PR https://github.com/EttusResearch/uhd/pull/548. It still has to go through our internal review process and your issue will be closed automatically as soon as the changes will have been merged into our internal master. A little bit later it will arrive in the public repo, too.

cpaelzer, 2022-01-20:
Yeah I know about #548 as @bluca is a friend of mine working on DPDK in Debian.
Thanks for the interim feedback @manderseck - looking forward to that release then ...

closed 2022-01-20 by atrnati