EttusResearch/uhd #550: X310 not responding after load custom RFNoC FPGA image

ArtisticZhao, 2022-01-25:
## Issue Description ##
I've built an RFNoC OOT module following the video of [RFNoC 4 Workshop - GRCon 2020](https://www.youtube.com/watch?v=M9ntwQie9vs&t=3308s).

And I build the FPGA image on target X300_HG. But when I load the image to X310, the X310 network is not responding.
But I try to load the image from the JTAG via Xilinx Vivado toolchain, the image seems worked. I can run the `uhd_usrp_probe` and get the image details.

## Setup Details ##

UHD 4.1.0.5

Now I can use my RFNoC OOT module by the Jtag loaded image. How can I burn image to Flash? 

Thank you.

wordimont, 2022-01-25:
You mentioned X300_HG but X310 device. Make sure you use the correct bitstream type for your device (X300_HG vs X310_HG).

Start by loading a known-good bitstream via JTAG, like the one downloaded by `uhd_images_downloader`. Make sure you can run `uhd_usrp_probe`.

Then you should be able to use `uhd_image_loader` to write a bitstream to flash, then cycle the power on the USRP to load the bitstream from flash. For example, for X310 you would run something like this:

`uhd_image_loader --args "type=x300,addr=192.168.10.2" --fpga-path ./build-X310_HG/x300.bit`

Make sure --fpga-path points to the correct working bitstream file you want to flash. Here are additional instructions on restoring the flash to a working state:

https://kb.ettus.com/X300/X310_Device_Recovery

ArtisticZhao, 2022-01-26:
@wordimont Thank you for your reply.

I make sure I had built the image use the correct target X310_HG. The following is my `gain_x310_rfnoc_image_core.yml`:

```
# General parameters
# -----------------------------------------
schema: rfnoc_imagebuilder_args         # Identifier for the schema used to validate this file
copyright: ''                           # Copyright information used in file headers
license: 'SPDX-License-Identifier: LGPL-3.0-or-later' # License information used in file headers
version: '1.0'                            # File version
rfnoc_version: '1.0'                      # RFNoC protocol version
chdr_width: 64                          # Bit width of the CHDR bus for this image
device: 'x310'
default_target: 'X310_HG'

# A list of all stream endpoints in design
# ----------------------------------------
stream_endpoints:
  ep0:                       # Stream endpoint name
    ctrl: True                      # Endpoint passes control traffic
    data: True                      # Endpoint passes data traffic
    buff_size: 32768                # Ingress buffer size for data
  ep1:                       # Stream endpoint name
    ctrl: False                     # Endpoint passes control traffic
    data: True                      # Endpoint passes data traffic
    buff_size: 0                    # Ingress buffer size for data
  ep2:                       # Stream endpoint name
    ctrl: False                     # Endpoint passes control traffic
    data: True                      # Endpoint passes data traffic
    buff_size: 32768                # Ingress buffer size for data
  ep3:                       # Stream endpoint name
    ctrl: False                     # Endpoint passes control traffic
    data: True                      # Endpoint passes data traffic
    buff_size: 0                    # Ingress buffer size for data
  ep4:                       # Stream endpoint name
    ctrl: False                     # Endpoint passes control traffic
    data: True                      # Endpoint passes data traffic
    buff_size: 32768                # Ingress buffer size for data

# A list of all NoC blocks in design
# ----------------------------------
noc_blocks:
  duc0:                      # NoC block name
    block_desc: 'duc.yml'    # Block device descriptor file
    parameters:
      NUM_PORTS: 1
  ddc0:
    block_desc: 'ddc.yml'
    parameters:
      NUM_PORTS: 2
  radio0:
    block_desc: 'radio_2x64.yml'
  duc1:
    block_desc: 'duc.yml'
    parameters:
      NUM_PORTS: 1
  ddc1:
    block_desc: 'ddc.yml'
    parameters:
      NUM_PORTS: 2
  radio1:
    block_desc: 'radio_2x64.yml'
  gain0:
    block_desc: 'gain.yml'

# A list of all static connections in design
# ------------------------------------------
# Format: A list of connection maps (list of key-value pairs) with the following keys
#         - srcblk  = Source block to connect
[... 38 lines cut ...]
#         - dstblk  = Destination block to connect
#         - dstport = Clock domain on the destination block to connect
clk_domains:
    - { srcblk: _device_, srcport: radio, dstblk: radio0, dstport: radio }
    - { srcblk: _device_, srcport: ce,    dstblk: ddc0,   dstport: ce    }
    - { srcblk: _device_, srcport: ce,    dstblk: duc0,   dstport: ce    }
    - { srcblk: _device_, srcport: radio, dstblk: radio1, dstport: radio }
    - { srcblk: _device_, srcport: ce,    dstblk: ddc1,   dstport: ce    }
    - { srcblk: _device_, srcport: ce,    dstblk: duc1,   dstport: ce    }
    - { srcblk: _device_, srcport: ce,    dstblk: gain0, dstport: ce }

```
I built the image via:
```
rfnoc_image_builder -F ~/Downloads/tmp/uhd/fpga -I ~/Playground/rfnoc/rfnoc-demogain -y ./rfnoc/icores/gain_x310_rfnoc_image_core.yml -l debug
```
I find image file on `/home/lilacsat/Downloads/tmp/uhd/fpga/usrp3/top/x300/build-X310_HG/x300.bit`

And I upload the image .bit file by Vivado hardware manger through the JTAG. It worked (I can `uhd_usrp_probe` and run GNURadio flowgraph which contain the RFNoC block I create).

But I burn the image use the `uhd_image_loader --args="type=x300,addr=192.168.40.2" --fpga-path=/home/lilacsat/Downloads/tmp/uhd/fpga/usrp3/top/x300/build-X310_HG/x300.bit`. When I cycled the power on X310, I seems like SFP not work. My PC cannot detect the connection from the X310.

Do you have any ideas? Thank you again.

Regards.

wordimont, 2022-01-26:
I don't see a problem with the commands you ran. It seems like the USRP isn't successfully loading your bitstream from flash on power up. Perhaps something went wrong when you flashed your bitstream. I suggest flashing a known-good bitstream onto X310 (like the one downloaded by uhd_images_downloader) and make sure that one works from the flash.  If that bitstream works then you could try to flash your custom one again.

ArtisticZhao, 2022-01-26:
Yes, I had try to upload the official image to flash. It works well.
Very strange.

wordimont, 2022-01-26:
Were there any error messages or anything suspicious when you flashed your image? Does it happen when you try to flash it the second time?

ArtisticZhao, 2022-01-27:
```
➜ uhd_image_loader --args="type=x300,addr=192.168.40.2" --fpga-path=/home/lilacsat/Downloads/tmp/uhd/fpga/usrp3/top/x300/build-X310_HG/x300.bit
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.1.0.HEAD-0-g6bd0be9c
Unit: USRP X310 (3191F5E, 192.168.40.2)
FPGA Image: /home/lilacsat/Downloads/tmp/uhd/fpga/usrp3/top/x300/build-X310_HG/x300.bit
-- Initializing FPGA loading...successful.
-- Loading FPGA image: 100% (121/121 sectors)
-- Finalizing image load...successful.
Power-cycle the USRP X310 to use the new image.
```
No error reported by uhd_image_loader.

Then I try to probe the X310.
```
➜ uhd_usrp_probe --args "addr=192.168.40.2"
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_4.1.0.HEAD-0-g6bd0be9c
Error: LookupError: KeyError: No devices found for ----->
Device Address:
    addr: 192.168.40.2

```
In my network page of settings, all ethenet is unplugged, It seems like the 10GE of X310 not configured. And all LEDs on X310 panel are off.

[image]

BTW, Is there any way to get X310 boot message for debug? 

Thank  you.

wordimont, 2022-01-27:
It sounds like the FPGA is unable to configure itself from the flash with your bistream. Do you think you could post the bitstream somewhere I could download it? I'd like to see if there's anything unusual about it (size, header, etc.).

ArtisticZhao, 2022-01-28:
[x300.zip](https://github.com/EttusResearch/uhd/files/7955236/x300.zip)
I upload the FPGA image in the attachment.

Thanks alot for your help.

wordimont, 2022-02-02:
I tried out your bitstream and it seems to work fine for me when loaded from flash. Both Ethernet ports work as expected after power cycling. I can't think of good a reason why the default bitstream would work on your flash but your custom one wouldn't, especially when it works for me.

Any chance you have another X310 you could try it on with an identical setup? I suppose it's possible the flash chip is damaged, but if that were the case then I wouldn't expect the default bitstream to also fail.

wordimont, 2022-02-02:
You could also try downloading and testing using the 1 GbE port to rule out any signal integrity issues over 10 GbE.

ArtisticZhao, 2022-02-17:
@wordimont 
Hi,
I use another x310 to load my custom image. It worked!
Maybe some strange bug of flash chip :joy:

closed 2022-02-17 by ArtisticZhao