EttusResearch/uhd #551: uhd-4.1.0.5: tries to load the firmware from the /share/uhd/images instead of the /usr/share/uhd/images

yarda, 2022-01-28:
## Issue Description ##
It tries to load the firmware from the /share/uhd/images instead of the /usr/share/uhd/images.

## Setup Details ##
Fedora distro package uhd-4.1.0.5-2.fc34.x86_64. It's also reproducible with the versions.

## Expected Behavior ##
Firmware should be loaded from the `/usr/share/uhd/images`.

## Actual Behaviour ##
Firmware is loaded from the `/share/uhd/images`.

## Steps to reproduce the problem ##
- Deploy Fedora
- dnf install uhd
- uhd_usrp_probe

## Additional Information ##
The problem is caused by the `uhd::get_images_dir()` method from the `/host/lib/utils/paths.cpp`. It tries three methods to get the directory. The third method tries to construct the path as the `"UHD package path" / share / uhd / images`. It does it by the following code:
```
fs::path(uhd::get_pkg_path()) / "share" / "uhd" / "images";
```
The `get_pkg_path()` does:
```
fs::path(uhd::get_lib_path()).parent_path().lexically_normal();
```
The `get_lib_path()` does:
```
fs::path runtime_libfile_path = boost::dll::this_line_location();
```
The `boost::dll::this_line_location()` is just wrapper to the `dlinfo` which uses the linker `ld` configuration.

From the manual page of the `ld.so`, i.e. `man ld.so`:
>  On some 64-bit architectures, the default paths for 64-bit shared  objects  are /lib64, and then /usr/lib64

And this seems to be the case in Fedora. It has `/lib64` symlink to the `/usr/lib64` directory, thus the linker uses the shorter path, i.e. for the `/usr/lib64/libuhd.so.4.1.0` the `this_line_location()` returns `/lib64/libuhd.so.4.1.0`. Which when processed by the code described above resulted in the `/share/uhd/images` instead of the `/usr/share/uhd/images`.

I think it should try first the compiled-in path constructed from the cmake variables like e.g. other tools do, e.g. the `uhd_images_downloader.py.in`:
```
_BASE_DIR_STRUCTURE_PARTS = ["share", "uhd", "images"]
_DEFAULT_INSTALL_PATH     = os.path.join("@CMAKE_INSTALL_PREFIX@", *_BASE_DIR_STRUCTURE_PARTS)
```
And if the compiled-in path doesn't exists then it could fallback to the runtime detection. Or the runtime detection should somehow cope with it. In fact, there could be arbitrary INSTALL_PREFIX, not just the `/usr`.

mbr0wn, 2022-02-01:
Thanks again for the detailed report.
This is interesting:

> The `boost::dll::this_line_location()` is just wrapper to the `dlinfo` which uses the linker `ld` configuration.

That's not what the docs say: https://www.boost.org/doc/libs/1_78_0/doc/html/boost/dll/this_line_location.html

Boost calls `dladdr` to figure out where libuhd.so is. So it's not clear why/how it's returning `/lib64` instead of `/usr/lib64`, unless that's actually where the .so is. I run Fedora on one of my machines with custom prefixes, and it works -- but I also have some more env vars set for the dynamic linker, so maybe that's why it works.

yarda, 2022-02-01:
Check the /etc/ld.so.cache. On the cleanly provisioned Fedora 34 it contains /lib64/... records, not the /usr/lib64/... records.

Also simple reproducer:
```
$ cat foo.cpp 
#include <iostream>
#include <boost/dll/runtime_symbol_info.hpp>

void foo(void)
{
  std::cout << boost::dll::this_line_location() << std::endl;
}

$ cat main.cpp 
extern void foo();

int main()
{
    foo();
    return 0;
}

$ g++ -fpic -shared -o libfoo.so foo.cpp
$ mv libfoo.so /usr/lib64/
$ g++ -o test main.cpp -lfoo -ldl
$ ./test
"/lib64/libfoo.so"
```
The same is reproducible on the fedora-rawhide (f36).

Regarding the dladdr, from the dladdr manual page:
```
       The function dladdr() determines whether the address specified in addr is located in one of the shared objects loaded by the
       calling application.  If it is, then dladdr() returns information about the shared object and  symbol  that  overlaps  addr.
       This information is returned in a Dl_info structure:

           typedef struct {
               const char *dli_fname;  /* Pathname of shared object that
                                          contains address */
               void       *dli_fbase;  /* Base address at which shared
                                          object is loaded */
               const char *dli_sname;  /* Name of symbol whose definition
                                          overlaps addr */
               void       *dli_saddr;  /* Exact address of symbol named
                                          in dli_sname */
           } Dl_info;
```
This is the dlinfo I meant, i.e. Dl_info not dlinfo. Sorry, I wasn't precise about it.

closed 2022-05-02 by mbr0wn