EttusResearch/uhd #556: Clock or time source can be silently altered by UHD on N3xx

draeman-synoptic, 2022-02-07:
## Issue Description ##
When calling set_time_source or set_clock_source with a combination that isn't present in the list of valid sync sources, the usrp_mpm/periph_manager implementation for n3xx.py silently changes clock or time sources to coerce a valid combination.

I want to use `clock_source=external` and `time_source=gpsdo` because I planned to use an external OCXO but wanted the internal GPS receiver's PPS for coarse time sync across distributed radios. I was calling set_clock_source("external") followed by set_time_source("gpsdo"), both of which seem to succeed. It took a long while to realize the second call was silently changing the clock_source to also be gpsdo.

The x4xx.py implementation is somewhat better because it at least logs a warning. However, my perspective is this case is better handled by throwing an exception - if a user application is explicitly asking for an unsupported clock/time combination, the user likely must revisit the planned use case, and automatic coercion is more likely to be confusing rather than helpful.

## Setup Details ##
N320 running UHD 4.1.0.1 and using the multi_usrp API.

## Expected Behavior ##
Attempting to set unsupported clock/time combination using sequential calls to set_clock_source/set_time_source throws an exception.

## Actual Behaviour ##
Attempting to set unsupported clock/time combination using sequential calls to set_clock_source/set_time_source seems successful, but actually performs silent coercion of whichever value was set first.

## Steps to reproduce the problem ##
1. Create a multi_usrp object to an N320 that has an external clock reference present.
2. Call set_clock_source("external").  It succeeds and readback confirms the value was set.
3. Call set_time_source("gpsdo").  It succeeds and readback confirms the value was set.
4. At this point, clock_source was silently changed to gpsdo.

## Additional Information ##
An additional question: Why is` clock_source=external,time_source=gpsdo` not a valid combination? It's just a PPS that is asynchronous with respect to refclk, ostensibly no different than `clock_source=external,time_source=external` which is supported.

closed 2025-02-05 by joergho

SunflowerHaf, 2025-04-26:
It would be great to have the solution for this - if one was reached?

joergho, 2025-04-27:
> It would be great to have the solution for this - if one was reached?

The implemented change is available in `master`: https://github.com/EttusResearch/uhd/commit/ba257139faaefc57ec05b8eaf56cdea37d908bcc. With this change, warnings are generated when time/clock sources are coerced. See the commit message for more details.