EttusResearch/uhd #557: Why RFNOC gain example building FPGA, I faced with module 'rfnoc_block_gain' not found?

stackprogramer, 2022-02-09:
I copied RFNOC gain example from UHD folder, I installed it successfully in
Gnuradio, But I can not see RFNOC Gain block in GRC blocks? why RFNOC gain example not work correctly?
I used Gnuradio 3.8.1,UHD-4.1.0.5,
for building FPGA  was faced with these errors.
**How can I solve this problem?** is there any example for creating custom RFNOC block that works correctly....
Thanks in advance

_ERROR: [Synth 8-439] module 'rfnoc_block_gain' not found
[/home/sp/Documents/rfnoc-tutorial/rfnoc/icores/x310_rfnoc_image_core.v:1055]
ERROR: [Synth 8-6156] failed synthesizing module 'rfnoc_image_core'
[/home/sp/Documents/rfnoc-tutorial/rfnoc/icores/x310_rfnoc_image_core.v:24]
ERROR: [Synth 8-6156] failed synthesizing module 'bus_int'
[/home/sp/Documents/uhd-4.1.0.5/fpga/usrp3/top/x300/bus_int.v:9]
ERROR: [Synth 8-6156] failed synthesizing module 'x300_core'
[/home/sp/Documents/uhd-4.1.0.5/fpga/usrp3/top/x300/x300_core.v:9]
ERROR: [Synth 8-6156] failed synthesizing module 'x300'
[/home/sp/Documents/uhd-4.1.0.5/fpga/usrp3/top/x300/x300.v:20]
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the
console or run log file for details
[00:04:05] Current task: Synthesis +++ Current Phase: Starting
[00:04:05] Current task: Synthesis +++ Current Phase: Finished
[00:04:05] Process terminated. Status: Failure_

wordimont, 2022-02-11:
Hi @stackprogramer, I see that your question was already answered on the mailing list. Feel free to reply there if those instructions don't work for you. A working example is here:

https://github.com/EttusResearch/uhd/tree/master/host/examples/rfnoc-example

You can build an FPGA like this:

```
cd rfnoc-example
mkdir build
cd build
cmake -DUHD_FPGA_DIR=<path_to_uhd_repo>/fpga ../
make x310_rfnoc_image_core
```

Or, you could build the FPGA like this:

```
rfnoc_image_builder -F <path_to_uhd_repo>/fpga -I ./rfnoc-example -y ./rfnoc-example/icores/x310_rfnoc_image_core.yml
```

closed 2022-02-11 by wordimont

stackprogramer, 2022-02-12:
I did these works, but the same before I was faced with error. I think there is a bug in UHD 4.1.0.5

ERROR: [Synth 8-439] module 'rfnoc_block_gain' not found