EttusResearch/uhd #558: UHD 4.1.0.5 breaks support for DPDK on 10GbE links

draeman-synoptic, 2022-02-10:
## Issue Description ##
Unfortunately DPDK support was broken by commit a629ce3a46. Here's a quick trace of the problem:
1. The commit increased MPMD_10GE_DATA_FRAME_MAX_SIZE from 7972 to 8016.
2. In mpmd_link_if_ctrl_udp.cpp, this constant is passed as the max frame size to discover_mtu(), which then tries to send a 8016-byte buffer via dpdk_simple::send.
3. dpdk_simple::send asserts if provided a buffer size >8000. This is because its constructor always makes a udp_dpdk_link object with default link parameters, which hard-codes send_frame_size and recv_frame_size to be to 8000 (toward the bottom of dpdk_simple.cpp).

## Setup Details ##

UHD 4.1.0.5 and DPDK 18.11.11.  Connected to an N320 over QSFP+ (using XQ image) and use_dpdk=1.

## Expected Behavior ##

Can operate connect to the radio via DPDK over a QSFP+ link (this worked in UHD 4.1.0.1).

## Actual Behaviour ##

Assertion is triggered:
```
[WARNING] [MPMD::XPORT::UDP] Error during MTU discovery on address 192.168.10.2: AssertionError: nbytes <= _link->get_send_frame_size()
  in virtual size_t uhd::transport::dpdk_simple_impl::send(const boost::asio::const_buffer&)
  at /home/rflab/prefix-4.1.0.5/src/uhd/host/lib/transport/dpdk_simple.cpp:98
```

alynch-ni, 2022-02-11:
This behavior was already fixed in the master branch: https://github.com/EttusResearch/uhd/commit/4439497afc8e9c9dfb982859c6b438d72c775bca

That change should be applied to the the UHD-4.1 branch as well.

draeman-synoptic, 2022-02-11:
Great, thank you. Apologies I didn't look beyond that was committed to the UHD-4.1 branch.

wordimont, 2022-02-11:
@draeman-synoptic Thanks for reporting! Please let us know if the fix in master failed resolve your issue.

closed 2022-02-11 by wordimont

mbr0wn, 2022-02-14:
We're cherry-picking this fix into UHD-4.1 also.