EttusResearch/uhd #564: UHD Python API Crash if usrp.recv_num_samps() called more than 90 times

Abdo-Gaber, 2022-02-17:
## Issue Description ##
UHD python API crashed if usrp.recv_num_samps() called more than 95 Times. In our project, we need to get large number of RX snapshots. We used the following example as a reference:
https://github.com/EttusResearch/uhd/blob/2c7ce2dbf72414b64f8a477be614e23bc12f086d/host/examples/python/rx_to_file.py

The process is:
- Open the session to usrp: **usrp = uhd.usrp.MultiUSRP(args.args)**
- Start for loop:
-  Get RX data: **samps = usrp.recv_num_samps(num_samps, args.freq, args.rate, args.channels, args.gain)**
-  Write RX Data to file
- repeat the previous two lines

## Setup Details ##
- UHD master: commit 2c7ce2dbf
- X310 USRP
- Ubuntu 20.04

## Expected Behavior ##
Get RX data con. without any crash for a big number of snapshots, 100, or even 1000

## Actual Behaviour ##
Crash happens in our for loop at n larger than approx. 94

## Steps to reproduce the problem ##
- Use this example as a referene: https://github.com/EttusResearch/uhd/blob/2c7ce2dbf72414b64f8a477be614e23bc12f086d/host/examples/python/rx_to_file.py
- Add for loop for line 38

## Additional Information ##
Crash output from terminal, line 59 in our code is the line 38 in above reference code:
[UHD_crash.txt](https://github.com/EttusResearch/uhd/files/8088112/UHD_crash.txt)

mbr0wn, 2022-02-18:
@Abdo-Gaber This may be a nice improvement regardless: Try creating a streamer before calling `recv_num_samps()` and keep it:

```py
st_args = lib.usrp.stream_args("fc32", "sc16")
st_args.channels = [0, 1, 2, 3] # If you're only using one channel, then this is simply [0]
rx_streamer = usrp.get_rx_stream(st_args)
for _ in range(100):
    data = usrp.recv_num_samps(num_samps, freq, channels=[0, 1, 2, 3], streamer=rx_streamer)
```

Abdo-Gaber, 2022-03-01:
@mbr0wn : Thanks for the input, I can run the code for 1000 captures without any crash in UHD Python API.

closed 2022-03-01 by Abdo-Gaber