EttusResearch/uhd #568: When I make fpga project in vivado2019.1 I faced errors

Qixibi, 2022-02-23:
## Issue Description ##

## Setup Details ##

**UHD:  fpga-UHD-3.15.LTS
Vivado : 2019.1
OS: win11 + cygwin64**

**follow the USRP Hardware Driver and USRP Manual 
source setupenv.sh --vivado-path=<VIVADO_PATH> (where VIVADO_PATH is a non-default installation path)**

`$ source setupenv.sh --vivado-path="/cygdrive/c/Xilinx/Vivado/2019.1/bin"`
**I got **
dirname: unknown option -- b
Try 'dirname --help' for more information.

**So, I confige the .bashrc and add**
`export PATH=$PATH:/cygdrive/c/Xilinx/Vivado/2019.1/bin`
**and skip the above step**
```
make cleanall
make x310_1G
```
 
**And， get output as**

qixib@DESKTOP-89E2NGG ~/fpga/usrp3/top/x300
$ make X310_1G
make -f Makefile.x300.inc bin NAME=X310_1G ARCH= PART_ID= BUILD_1G=1 SFP0_1GBE=1 SFP1_1GBE=1  X310=1 TOP_MODULE=x300 EXTRA_DEFS="BUILD_1G=1 SFP0_1GBE=1 SFP1_1GBE=1  X310=1"
make[1]: Entering directory '/home/qixib/fpga/usrp3/top/x300'
BUILDER: Checking tools...
* GNU bash, version 4.4.12(3)-release (x86_64-unknown-cygwin)
* Python 3.9.10
* Vivado v2019.1 (64-bit)
========================================================
BUILDER: Building IP one_gig_eth_pcs_pma
========================================================
BUILDER: Staging IP in build directory...
BUILDER: Reserving IP location: /home/qixib/fpga/usrp3/top/x300/build-ip/one_gig_eth_pcs_pma
BUILDER: Retargeting IP to part /...
**ERROR: Invalid target format. Must be <arch>/<device>/<package>/<speedgrade>/<tempgrade>/<silicon revision>
BUILDER: Building IP...
[00:00:00] Executing command: vivado -mode batch -source /home/qixib/fpga/usrp3/tools/scripts/viv_generate_ip.tcl -log one_gig_eth_pcs_pma.log -nojournal
[00:00:03] Current task: Initialization +++ Current Phase: Starting
couldn't read file "/home/qixib/fpga/usrp3/tools/scripts/viv_generate_ip.tcl": no such file or directory**
[00:00:03] Current task: Initialization +++ Current Phase: Finished
[00:00:03] Process terminated. Status: Failure
========================================================
Warnings:           0
Critical Warnings:  0
Errors:             1
BUILDER: Releasing IP location: /home/qixib/fpga/usrp3/top/x300/build-ip/one_gig_eth_pcs_pma
make[1]: *** [/home/qixib/fpga/usrp3/top/x300/ip/one_gig_eth_pcs_pma/Makefile.inc:46: /home/qixib/fpga/usrp3/top/x300/build-ip/one_gig_eth_pcs_pma/one_gig_eth_pcs_pma.xci.out] Error 1
make[1]: Leaving directory '/home/qixib/fpga/usrp3/top/x300'
make: *** [Makefile:62: X310_1G] Error 2

**actually  the file /home/qixib/fpga/usrp3/tools/scripts/viv_generate_ip.tcl exists**

I have searched and the link can not solve my issues. 
https://github.com/EttusResearch/uhd/issues/539

## Additional Information ##

wordimont, 2022-02-24:
Hi @Qixibi. The `dirname: unknown option -- b` error is a known issue with Cygwin and is due to the fact that it's case insensitive. The error usually occurs because you're running the Xilinx "setupEnv.sh" in /cygdrive/c/Xilinx/Vivado/2019.1/bin, which you added to your path, not the one in the x300 directory.

Try sourcing it with the path, like this: `source ./setupenv.sh --vivado-path="/cygdrive/c/Xilinx/Vivado`
(note the `./` that was added)

The setupenv.sh needs to complete successfully because it configures variables that contain things like the target FPGA, and it looks like those variables aren't set in your build, causing the "Invalid target format" error.

wordimont, 2022-02-24:
As an alternative to Cygwin, you might also consider WSL. It's less likely to run into Windows issues like this, or the long file paths that Vivado likes to generate. The down side is that you have to install the Linux version of Vivado on WSL.

closed 2022-04-12 by wordimont