EttusResearch/uhd #576: Error installing UHD

mvl5023, 2022-03-24:
I am attempting to install UHD on Ubuntu 20.04.3 using the instructions provided at the knowledge base.  I clone the repository, select a branch, and when I attempt to run cmake I get the following:

        -- The CXX compiler identification is GNU 9.4.0
        -- The C compiler identification is GNU 9.4.0
        -- Check for working CXX compiler: /usr/bin/c++
        -- Check for working CXX compiler: /usr/bin/c++ -- works
        -- Detecting CXX compiler ABI info
        -- Detecting CXX compiler ABI info - done
        -- Detecting CXX compile features
        -- Detecting CXX compile features - done
        -- Check for working C compiler: /usr/bin/cc
        -- Check for working C compiler: /usr/bin/cc -- works
        -- Detecting C compiler ABI info
        -- Detecting C compiler ABI info - done
        -- Detecting C compile features
        -- Detecting C compile features - done
        -- 
        -- Configuring the Python interpreter...
        -- Python interpreter: /usr/local/bin/python3.9 Version: 3.9.9
        -- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
        -- Python runtime interpreter: /usr/local/bin/python3.9 Version: 3.9.9
        -- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
        -- Finding Python Libraries...
        -- Python Libraries: /usr/local/lib/libpython3.9.a
        -- Python include directories: /usr/local/include/python3.9
        -- Working off of feature or development branch. Updating version number.
        -- Using UHD Images Directory: /usr/local/share/uhd/images
        -- Build type not specified: defaulting to release.
        -- Performing Test HAVE_VISIBILITY_HIDDEN
        -- Performing Test HAVE_VISIBILITY_HIDDEN - Success
        -- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN
        -- Performing Test HAVE_VISIBILITY_INLINES_HIDDEN - Success
        -- 
        -- Configuring Boost C++ Libraries...
        -- 
        -- Checking for Boost version 1.65 or greater
        --   Looking for required Boost components...
        --   Enabling Boost Error Code Header Only
        --     Disabling boost::asio use of std::string_view
        --   Boost version: 1.71.0
        --   Boost include directories: /usr/include
        --   Boost library directories: /usr/lib/x86_64-linux-gnu
        --   Boost libraries: Boost::chrono;Boost::date_time;Boost::filesystem;Boost::program_options;Boost::serialization;Boost::thread;Boost::unit_test_framework;Boost::system
        -- Looking for Boost version 1.65 or greater - found
        -- 
        -- Python checking for Python version 3.6 or greater
        -- Python checking for Python version 3.6 or greater - found
        -- 
        -- Python checking for Mako templates 0.4.2 or greater
        -- Python checking for Mako templates 0.4.2 or greater - "import mako" failed
        -- 
        -- Python checking for requests 2.0 or greater
        -- Python checking for requests 2.0 or greater - "import requests" failed
        -- 
        -- Python checking for numpy 1.11 or greater
        -- Python checking for numpy 1.11 or greater - "import numpy" failed
        -- 
        -- Python checking for ruamel.yaml 0.15 or greater
        -- Python checking for ruamel.yaml 0.15 or greater - "import ruamel.yaml" failed
        -- 
        -- Configuring LibUHD support...
        --   Dependency Boost_FOUND = TRUE
        --   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
        CMake Error at cmake/Modules/UHDComponent.cmake:59 (message):
          Dependencies for required component LibUHD not met.
        Call Stack (most recent call first):
          CMakeLists.txt:420 (LIBUHD_REGISTER_COMPONENT)
        
        
        -- Configuring incomplete, errors occurred!
        See also "/home/mlipski/workarea/uhd/host/build/CMakeFiles/CMakeOutput.log".
        See also "/home/mlipski/workarea/uhd/host/build/CMakeFiles/CMakeError.log".

It looks like it can't find some Python packages?  All of the packages mentioned should be installed, so maybe it's a path/version mismatch problem.  I searched around and found some old posts from people who encountered similar issues, but I couldn't find a solution that worked.

Any help would be appreciated;

Thanks!

michaelld, 2022-03-24:
The [KB build instructions use `apt` for all dependencies](https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux#Update_and_Install_dependencies), which provides Python 3.8:
```
$ uname -a
Linux uhd-dev-mld-beast 5.13.0-37-generic #42~20.04.1-Ubuntu SMP Tue Mar 15 15:44:28 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
$ python3
Python 3.8.10 (default, Nov 26 2021, 20:14:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> ^D
```

You have installed (somehow) Python 3.9:
```
    -- Configuring the Python interpreter...
    -- Python interpreter: /usr/local/bin/python3.9 Version: 3.9.9
    -- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
    -- Python runtime interpreter: /usr/local/bin/python3.9 Version: 3.9.9
    -- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
    -- Finding Python Libraries...
    -- Python Libraries: /usr/local/lib/libpython3.9.a
    -- Python include directories: /usr/local/include/python3.9
```
Thus, the `mako` installed by the KB may or not be found depending on how/where it is installed.

You can of course use your Py39 to install Mako, which one would hope would work well given how it would be installed!

closed 2022-03-24 by michaelld

mvl5023, 2022-03-24:
Great, I'll give it a try, thanks!