EttusResearch/uhd #577: Python ModuleNotFoundError on Ubuntu

nsbruce, 2022-03-25:
## Issue Description ##
Following install from source instruction at kb, the manual, and pysdr.org yields `ModuleNotFoundError` in Python 3.8.10 on Ubuntu 20.04.4.

## Setup Details ##

Ubuntu 20.04.4
UHD v4.1.0.5 (but can switch to something else if it will fix this problem)
Python 3.8.10
cmake 3.16.3
make 4.2.1

## Expected Behavior ##

No error on

```python
import uhd
```

## Actual Behaviour ##

```python
In [1]: import uhd
---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
Input In [1], in <cell line: 1>()
----> 1 import uhd

ModuleNotFoundError: No module named 'uhd'
```

## Steps to reproduce the problem ##

Clone this repo,

```bash
cd uhd/host && mkdir build && cd build
git checkout v4.1.0.5
cmake -DENABLE_PYTHON_API=ON -DENABLE_C_API=ON ..
make
sudo make install
ldconfig
```

No errors are thrown. It says it has all successfully compiled. Afterwards, there is no `LD_LIBRARY_PATH` variable on my system so I found `libuhd.so` at `/usr/local/lib/libuhd.so` and created a file at `/etc/ld.so.conf.d/uhd.conf` containing `/usr/local/lib` and reran `ldconfig`.

## Additional Information ##

nsbruce, 2022-03-27:
I don't have a `libpyuhd.so` on my system but I do have the following:

```bash
$ sudo find / -name "*libpyuhd*"
        /usr/local/lib/python3/dist-packages/uhd/libpyuhd.cpython-38-x86_64-linux-gnu.so
        /usr/local/lib/python3.8/site-packages/uhd/libpyuhd.cpython-38-x86_64-linux-gnu.so
        /home/nsbruce/uhd/host/build/python/libpyuhd.cpython-38-x86_64-linux-gnu.so
        /home/nsbruce/uhd/host/build/python/build/lib/uhd/libpyuhd.cpython-38-x86_64-linux-gnu.so
        /home/nsbruce/uhd/host/build/python/uhd/libpyuhd.cpython-38-x86_64-linux-gnu.so 
```

michaelld, 2022-03-28:
A couple thoughts: (1) You have (somehow) installed Python3 (probably 3.8) into `/usr/local` (PREFIX), which is non-standard but might work still; I believe that Ubuntu 20.04.4 provides Python 3.8 by default. That said, wondering why you're not using the `apt` provided Python (whatever version it might actually be)? (2) What does the following return: `/usr/local/bin/python3.8 -c "import sys ; print(sys.path)` ... I'm assuming that the `python3.8` executable is correctly located in the noted PREFIX.

nsbruce, 2022-03-30:
Thanks for the reply @michaelld . This is the default apt installed python 3.8 - I'm on a fresh machine where I'll I've done is tried to install uhd. `which python3` and `which python3.8` return `/usr/bin/python3` and `/usr/bin/python3.8` respectively. Running `apt install python3` does nothing saying it's already the newest version.

Running your command returns an error since that's not the python prefix

```bash
$ /usr/local/bin/python3.8 -c "import sys ; print(sys.path)"
zsh: no such file or directory: /usr/local/bin/python3.8
```

While running it on the /usr/bin/python3.8 returns:

```bash
$ /usr/bin/python3.8 -c "import sys ; print(sys.path)" 
['', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/<MY-USERNAME>/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
```

michaelld, 2022-03-30:
@nsbruce OK progress! Ah ... so you're using the build-in python3.8 but installing UHD from source. Gotcha. Here's the rub:

1) You noted:
```
sudo find / -name "*libpyuhd*"
        /usr/local/lib/python3/dist-packages/uhd/libpyuhd.cpython-38-x86_64-linux-gnu.so
        /usr/local/lib/python3.8/site-packages/uhd/libpyuhd.cpython-38-x86_64-linux-gnu.so
```
I'm guessing if you do `ls -lAF /usr/local/lib/python3/  /usr/local/lib/python3.8/` there will be no symlink between the entries. Also note that one of the subdirectories is `dist-packages` while the other is `site-packages`.

2) You also noted:
```
$ /usr/bin/python3.8 -c "import sys ; print(sys.path)" 
..., '/usr/local/lib/python3.8/dist-packages',...
```

Finally: Note that the directories from (1) and (2) are inconsistent! I know this is not your fault; I'm just noting it.

The fastest way through is to do something like (for BASH; you're using ZSH & I'm not sure what to do there, hopefully you do ...):
```
export PYTHONPATH=${PYTHONPATH}:/usr/local/lib/python3/dist-packages:/usr/local/lib/python3.8/site-packages
```

Another option is when configuring UHD (in `cmake`) to set its Python install directory to one known by the Python being used.

nsbruce, 2022-03-31:
Setting the pythonpath worked! Thank you for the help!

closed 2022-03-31 by nsbruce

michaelld, 2022-04-01:
Glad that worked & you're welcome!