EttusResearch/uhd #581: E310 not found using "uhd_find_devices"

mvl5023, 2022-04-07:
## Issue Description ##
I have an E310 connected to my host computer via USB, but when I run "uhd_find_devices", I get "No UHD devices found."

## Setup Details ##
Host system is a Dell XPS13 running Ubuntu 20.04 LTS.  I have UHD 4.1.0 installed, and I updated the SD card on the E310 to UHD 4.1.0 as well.  When I plug the E310 into the host using the USB cable, there's no apparent change or system message to indicate that something has been plugged in (not sure if there is supposed to be).   However, I can open a serial connection to the E310 using "sudo screen /dev/ttyUSB0 115200", can navigate the file system and run example programs.

I tried the commands under the "Configuring USB" section of the UHD Linux build instructions on the knowledge base, but this did not help.

## Expected Behavior ##
From everything I have read so far, the E310 should be set up and good to go...running "uhd_find_devices" should show the E310.

## Actual Behaviour ##
Host system shows no UHD device connected.

## Steps to reproduce the problem ##
Unclear.

## Additional Information ##

wordimont, 2022-04-10:
@mvl5023 The micro USB port on the E310 is for the serial console, and that seems to be working as expected for you. For uhd_find_devices to work, you need to connect the Ethernet port to the same network as the computer running UHD. You can also SSH into the device once it's on the network.

mvl5023, 2022-04-10:
So uhd_find_devices doesn't work with a USB connected USRP?  Interesting.  I'll try connecting the E310 over the network and see what happens.  Thanks!

wordimont, 2022-04-10:
uhd_find_devices over USB should work for Bus (B) Series devices, but Embedded (E) Series devices use the network interface. They are meant to run standalone (without a host computer). The instructions for configuring USB that you read were probably for B-series devices.

mvl5023, 2022-04-12:
I was able to connect to the E310 directly using a USB to ethernet adapter, and I can SSH into it, but uhd_find_devices still says it can't find any devices.  Since I'm not running a DHCP server or anything, I set a static IP on the E310; would this affect the ability for uhd_find_devices to work?

wordimont, 2022-04-12:
Make sure the E310 and the IP address for your Eth adapter connected to it are on the same IP subnet. Then I would expect it to work.

mvl5023, 2022-04-12:
Okay, I set up a static IP on both the E310 and the ethernet adapter such that they are both on the same subnet.  The USRP is now showing up correctly when I run uhd_find_devices.

Awesome, thank you so much for the help!

closed 2022-04-12 by mvl5023