EttusResearch/uhd #582: USRP X300 + UBX 160 bandwith only 100MHz

cn0xroot, 2022-04-08:
Hi,Ettus.

I'm using X300+ubx160 on ubuntu 18.04   UHD-3.10  (10G ETH SPF

```
uhd_find_devices 
linux; GNU C++ version 7.5.0; Boost_106501; UHD_003.010.003.UHD-3.10-0-gef157678

--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    type: x300
    addr: 192.168.40.2
    fpga: HG
    name: 
    serial: 
    product: X300

```

when I try to use ubx160  and set bandwith  >100MHz:

```

gr-osmosdr 0.1.5 (0.1.5) gnuradio 3.7.14.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
-- Using subdev spec 'A:0 B:0'.

UHD Warning:
    The hardware does not support the requested RX sample rate:
    Target sample rate: 108.000000 MSps
    Actual sample rate: 100.000000 MSps
```

GNURadio and gqrx get the same Warning

[image]

[image]

[image]

any ideal to fix it? Thx!

cn0xroot, 2022-04-08:
change fpga from HG to XG,get the same Warning

michaelld, 2022-04-08:
@cn0xroot A few thoughts: 

(1) Your UHD version is quite old. We really don't support that old UHD any longer. Please update to UHD 4.1 and if the problem persists then post new updates showing the recent UHD.

(2) Officially we support only UHD directly, not the other interfaces that use UHD such as OSMOCOM or, currently, gr-uhd (we will be supporting gr-uhd soon, once gr-ettus is merged into it & just for [I think] starting with GR 4). If you can use one of the UHD-installed examples or utilities and show the same issue, that would be a much stronger case for there being an issue.

(3) I think you're confusing sample rate with bandwidth. To get higher BW, you need to get the sample rate >= BW. The X3x0 USRP [supports "master clock rate" of 200.0 MHz and 184.32 MHz](https://kb.ettus.com/X300/X310#Clocking_and_Sampling_Rates). Hence to get > 100 MHz BW, you'll need to use full sample rate for whichever MCR is in use. Given that you're seeing "Actual sample rate: 100.000000 MSps", this means that the MCR is 200 MHz. So, try 200e6 for the sample rate, and yes you'll need the XG FPGA image for that amount of data.

michaelld, 2022-04-08:
I'm going to close this issue since I think once you've worked over my thoughts you'll figure out how to do what you want. If I'm mistaken, then please reopen and post updates showing the updated UHD as well as MCR at 200 MHz.

closed 2022-04-08 by michaelld