EttusResearch/uhd #585: cmd time errors with UHD USRP sink

zhanght34, 2022-04-23:
I'm playing with gnuradio and an USRP X410 device.
I connect pc with USRP X410 by QFSP.I want to test how fast the QFSP port transfer rate is.
Then in gnuradio grc,i have connected two block:Sig Source and USRP Sink. Details as shown
[image]

When i run this grc,and set samp_ate to 30MHz，gnuradio printfs"cmd time errors".How to slove it?
[image]

System Information
i9-11900F @ 2.50GHz
Ubuntu20.04
GNU Radio 3.10.1.1
Python 3.8.10
UHD 4.2.00-rc1

USRP X410 fpga image
X4_200

GNU Radio Version
3.10 (maint-3.10)

Specific Version
3.10.1.1

Error message printfs"
Susrp_sink :error: In the last 28292 ms, 1 cmd time errors occurred.
SSSSSSSSUusrp_sink :error: In the last 28298 ms, 1 underflows occurred.
"
At the same time, the light on the usrp x410 showing the RF transmit go off.

raefYoussef, 2023-09-01:
I have the same issue. Did you ever get it solved?

mbr0wn, 2023-12-12:
We have too little data, so I'm closing this. In general, using the USRP Sink works with X410 and other devices.

closed 2023-12-12 by mbr0wn

Shirleyue, 2024-07-12:
> I have the same issue. Did you ever get it solved?

I have the same issue. Did you ever get it solved? Please contact me if you get it solved.  My email address is shirleyuue@foxmail.com.