EttusResearch/uhd #586: When FPGA is CG_ 400, the connection between x410 and network card（E810） fails

zchengyun, 2022-04-28:
## Issue Description ##

I updated the mender of x410 to 4.2.0.0. I burn the FPGA into X4_200, I use QSFP28 cable to connect QSFP28-0 of x410, connect E810 at the other end, and use win10 to run uhd_ usrp_probe it works normally and detects device information. But when I burn FPGA into CG_400,and modify the speed and duplex range to 100Gbps in the network connection of the control panel ,the E810 indicator doesn't light up after inserting the cable. It seems that the connection failed. What do I need to do to make it jobs?

## Setup Details ##

E810 supports: 100/50/25/10GbE (200GbE when ports bifurcated).
When I connect the two ports of E810 with QSFP28 cable, it is displayed as 100Gbps on windows.
## Expected Behavior ##

When I burning CG_400 to X410, USRP x410 and E810 can be connected.
## Actual Behaviour ##

When I burning CG_400 to X410, USRP x410 and E810 are not connected.
## Steps to reproduce the problem ##

## Additional Information ##

wordimont, 2022-04-28:
I think a NVM update is required to change the mode of the E810 card. Try using the Intel Ethernet Port Configuration Tool (EPCT) to reconfigure the Intel E810 card for 100 GbE, then completely power cycle your system and reboot. This tool is included with the Intel Ethernet Adapter Complete Driver Pack or can be downloaded separately.

https://www.intel.com/content/www/us/en/download/15084/intel-ethernet-adapter-complete-driver-pack.html

zchengyun, 2022-05-05:
> 我认为需要NVM更新来更改E810卡的模式。尝试使用英特尔以太网端口配置工具 （EPCT） 将英特尔 E810 卡重新配置为 100 GbE，然后完全重启系统并重新启动。此工具包含在英特尔以太网适配器完整驱动程序包中，也可以单独下载。
> 
> https://www.intel.com/content/www/us/en/download/15084/intel-ethernet-adapter-complete-driver-pack.html

Thank you for your reply. I have updated NVM to the latest version 3.20, and then restart the system completely. In addition, the E810 network port connected to qsfp28-0 is set to 192.168.10.3, and the network card light is still not on. It seems that the connection is not successful.

wordimont, 2022-05-06:
Were you able to confirm using the EPCT tool that the E810 is correctly configured for 100 GbE operation?

What type of cable are you using (model/part number)?

zchengyun, 2022-05-07:
I set the speed and duplex to 100GbE in Intel PROSet, and then connect the two ports of E810. The connection status is 100GbE, so it seems that there should be no problem with the cable and E810.

zchengyun, 2022-05-07:
> Were you able to confirm using the EPCT tool that the E810 is correctly configured for 100 GbE operation?
> 
> What type of cable are you using (model/part number)?

I set the speed and duplex to 100GbE in Intel PROSet, and then connect the two ports of E810. The connection status is 100GbE, so it seems that there should be no problem with the cable and E810.

closed 2022-08-11 by zchengyun