EttusResearch/uhd #587: RFNOC : GRAPH ERROR x310 + TVRX2

himanshu10140, 2022-04-29:
USRP X310 with 01 TVRX2 daughterboard
Latest uhd (v4.2.0.0)
X310 + SBX = working without any issue

uhd(v3.14) = TVRX2 gets detected without any issue

[image]

mbr0wn, 2022-05-02:
Is only uhd_usrp_probe failing, or also other utilities/examples?

Did you build UHD from source? If so, can you please rebuild with `-DUHD_LOG_MIN_LEVEL=trace`.

In any case, can you please set the log level to max (`export UHD_LOG_LEVEL=trace`, `export UHD_LOG_CONSOLE_LEVEL=trace`) and post the output here?

mbr0wn, 2024-04-17:
Closing due to lack of feedback.

closed 2024-04-17 by mbr0wn