EttusResearch/uhd #588: CMake and compile issues: 4.2.0.0 with c++std=20

CJCombrink, 2022-05-12:
## Issue Description ##

We can't compile UHD tag 4.2.0.0 due to the following issues:
- Template parameter specified in constructor of property_impl
- "host/cmake/Modules/UHDAtomics.cmake" does not find boost if boost is specified using "BOOST_ROOT" cmake definition (not in system installs)

## Setup Details ##

Clean Debian 11 environment, like docker.
1. Install GCC11 (build from source probably)
1. Download and build boost from source
1. For UHD run cmake with BOOST_ROOT set to path where boost was downloaded to, with c++std=20
1. Cmake fails to detect atomics 
1. Using the property tree from an application also using c++std=20 fails. 

## Expected Behavior ##
UHD Compiles and works

## Actual Behaviour ##
CMake fails to configure uhd and build.

## Steps to reproduce the problem ##

```
docker run -it --rm gcc:11
apt update
apt install -y cmake python3-mako
# get boost
cd /
wget https://boostorg.jfrog.io/artifactory/main/release/1.79.0/source/boost_1_79_0.tar.gz
tar -xf boost_1_79_0.tar.gz
cd boost_1_79_0
./bootstrap.sh --prefix=/boost/1.79/
./b2
./b2 install
# get uhd
cd /
git clone https://github.com/EttusResearch/uhd.git
cd uhd 
git checkout v4.2.0.0
# run cmake
mkdir -p host/build && cd host/build
cmake -DBOOST_ROOT=/boost/1.79/ ..
```

Error from CMAKE:
```
...
--   Boost version: 1.79.0
--   Boost include directories: /boost/1.79/include
--   Boost library directories: /boost/1.79/lib
--   Boost libraries: /boost/1.79/lib/libboost_chrono.so;/boost/1.79/lib/libboost_date_time.so;/boost/1.79/lib/libboost_filesystem.so;/boost/1.79/lib/libboost_program_options.so;/boost/1.79/lib/libboost_serialization.so;/boost/1.79/lib/libboost_thread.so;-lpthread;/boost/1.79/lib/libboost_unit_test_framework.so;/boost/1.79/lib/libboost_system.so;/boost/1.79/lib/libboost_atomic.so
-- Looking for Boost version 1.65 or greater - found
...
-- Configuring atomics support...
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_BOOST_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_BOOST_ATOMICS_WITHOUT_LIB - Failed
-- Looking for __atomic_fetch_add_4 in atomic
-- Looking for __atomic_fetch_add_4 in atomic - found
-- Performing Test HAVE_CXX_BOOST_ATOMICS_WITH_LIB
-- Performing Test HAVE_CXX_BOOST_ATOMICS_WITH_LIB - Failed
CMake Error at cmake/Modules/UHDAtomics.cmake:102 (message):
  Host compiler must support std::atomic!
Call Stack (most recent call first):
  lib/utils/CMakeLists.txt:187 (CHECK_ATOMICS_LIB_REQUIRED)
  lib/CMakeLists.txt:46 (include)
  lib/CMakeLists.txt:93 (INCLUDE_SUBDIRECTORY)
```

## Additional Information ##
If I apply the following patch cmake works:
```
diff --git a/host/cmake/Modules/UHDAtomics.cmake b/host/cmake/Modules/UHDAtomics.cmake
index eb2800f..560bfce 100644
--- a/host/cmake/Modules/UHDAtomics.cmake
+++ b/host/cmake/Modules/UHDAtomics.cmake
@@ -10,0 +11,2 @@
+set(CMAKE_REQUIRED_INCLUDES ${BOOST_ROOT}/include) 
+
```

clean build folder and rerun cmake:
```
-- Configuring atomics support...
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_BOOST_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_BOOST_ATOMICS_WITHOUT_LIB - Success
--   Atomics support is built-in, no linking required.
```

We also need to apply the following patch to be able to use C++20 and the property tree:
```
diff --git a/host/include/uhd/property_tree.ipp b/host/include/uhd/property_tree.ipp
index e9dd307..20b8453 100644
--- a/host/include/uhd/property_tree.ipp
+++ b/host/include/uhd/property_tree.ipp
@@ -24 +24 @@
-    property_impl<T>(property_tree::coerce_mode_t mode) : _coerce_mode(mode)
+    property_impl(property_tree::coerce_mode_t mode) : _coerce_mode(mode)
```
This is related to a fix that was done in the past but did not fix all of the cases, see this fix in UHD: https://github.com/EttusResearch/uhd/commit/a5f32a7
SO on the issue: https://stackoverflow.com/a/71978837/991000

CJCombrink, 2022-05-12:
I have tested with the master branch (commit 5333d3d12ffc21229ec4203a9ea1c7f68d82e57f (HEAD -> master, origin/master, origin/HEAD) and the issue is still there.

The patch above for property_tree.ipp does not work on the master branch but the issue is still present

CJCombrink, 2022-05-12:
I updated the issue after creation to remove an unrelated issue, will file a separate issue for that one.

b00ga, 2022-05-16:
I ran into this issue today as well.
* UHD 4.2.0.0 release
* AlmaLinux 8
* Boost 1.69 from Alma boost169 family of packages (`dnf install boost169-devel`) (installs includes to /usr/include/boost169/ and libs to /usr/lib64/boost169/)
* Hinting cmake with `-DBoost_INCLUDE_DIR=/usr/include/boost169 -DBoost_LIBRARY_DIR=/usr/lib64/boost169/` command line options (for use by _host/cmake/Modules/UHDBoost.cmake_).

I made a similar-ish fix. Based on these docs: https://cmake.org/cmake/help/latest/module/CheckCXXSourceCompiles.html
I set `CMAKE_REQUIRED_INCLUDES`.

Here's my diff (from master to my modified cmake file in 4.2.0.0 release).
```
$ diff -urN uhd/host/cmake/Modules/UHDAtomics.cmake uhd-4.2.0.0/host/cmake/Modules/UHDAtomics.cmake 
--- uhd/host/cmake/Modules/UHDAtomics.cmake	2022-05-16 18:01:59.170208258 +0000
+++ uhd-4.2.0.0/host/cmake/Modules/UHDAtomics.cmake	2022-05-16 18:04:03.809489339 +0000
@@ -36,6 +36,7 @@
 # Note: If we reach this, we have already checked for the existence of Boost,
 # and Boost.Lockfree is a header-only library so no linker options required.
 function(CHECK_WORKING_CXX_BOOST_ATOMICS varname)
+    list(APPEND CMAKE_REQUIRED_INCLUDES ${Boost_INCLUDE_DIRS})
     CHECK_CXX_SOURCE_COMPILES("
         #include <boost/lockfree/queue.hpp>
         boost::lockfree::queue<int> queue(1);
```