EttusResearch/uhd #596: The FPGA build is not compatible with the host code build

VDelport, 2022-05-20:
Hello,

I want to run a B200 with GNU Radio Version 3.7 on Linux Mint 19 Cinnamon 3.8.9. Since the package installation from UHD-Host didn't work, I translated the sources. When starting the GNU Radio program I get the following error message:

RuntimeError: RuntimeError: Expected FPGA compatibility number 14, but got 16:
The FPGA build is not compatible with the host code build.
Please run:

  "/usr/lib/x86_64-linux-gnu/uhd/utils/uhd_images_downloader.py"

  I had previously run images_downloader.py successfully.

Which version of UHD do I need to clone and compile for FPGA compatibility?

wordimont, 2022-05-20:
@VDelport, FPGA compat number 16 is the latest, so that's good. To work with compat number 16, you need UHD 3.13 or later.

wordimont, 2022-05-21:
Hmm, I'm not sure. I suggest you uninstall the old UHD version if you're planning to use the one you compiled. In my experience, having one from the package manager and one installed from source doesn't mix well. You're probably better off using the newer versions of UHD and GNU Radio anyway.

closed 2022-06-02 by michaelld