EttusResearch/uhd #598: Clock incompatibility in UHD example UHD-4.1.0.5,Can any developer examine it?[possible be a bug]

stackprogramer, 2022-05-21:
When I want to build the rfnoc-example that is in the UHD-4.1.05 folder I
am faced with the  ce_clock error...
In this example, there is an Incompatibility clock domain. In file gain.yml
we see that there is not any ce_clock domain...

### //gain.yml  section
clocks:

name: rfnoc_chdr
freq: "[]"
name: rfnoc_ctrl
freq: "[]"
But in file x310_rfnoc_image_core.yml we have not defined any clock for
gain blocks.
Why is this RFNOC example not built successfully? Does anyone have an idea? when I
added ce clock domain in file image core and defined
a wire ce_clk, Even bitstream is generated but block not work correctly in
Gnuradio and we have bad packet errors...

### //x310_rfnoc_image_core.yml
clk_domains:
- { srcblk: device, srcport: radio, dstblk: radio0, dstport: radio }
- { srcblk: device, srcport: ce,    dstblk: ddc0,  dstport: ce    }
- { srcblk: device, srcport: ce,    dstblk: duc0,  dstport: ce    }
- { srcblk: device, srcport: radio, dstblk: radio1, dstport: radio }
- { srcblk: device, srcport: ce,    dstblk: ddc1,  dstport: ce    }
- { srcblk: device, srcport: ce,    dstblk: duc1,  dstport: ce    }

### Error ce_clk:
```
[00:00:22] Starting Synthesis Command
ERROR: [Synth 8-448] named port connection 'ce_clk' does not exist for
instance 'b_gain0_6' of module 'rfnoc_block_gain'
[/home/sp/Documents/rfnoc-transceiver/rfnoc/icores/x300_rfnoc_image_core.v:1058]
ERROR: [Synth 8-6156] failed synthesizing module 'rfnoc_image_core'
[/home/sp/Documents/rfnoc-transceiver/rfnoc/icores/x300_rfnoc_image_core.v:24]
ERROR: [Synth 8-6156] failed synthesizing module 'bus_int'
[/home/sp/Documents/uhd-4.1.0.5/fpga/usrp3/top/x300/bus_int.v:9]
ERROR: [Synth 8-6156] failed synthesizing module 'x300_core'
[/home/sp/Documents/uhd-4.1.0.5/fpga/usrp3/top/x300/x300_core.v:9]
ERROR: [Synth 8-6156] failed synthesizing module 'x300'
[/home/sp/Documents/uhd-4.1.0.5/fpga/usrp3/top/x300/x300.v:20]
ERROR: [Common 17-69] Command failed: Synthesis failed - please see the
console or run log file for details
[00:03:42] Current task: Synthesis +++ Current Phase: Starting
[00:03:43] Current task: Synthesis +++ Current Phase: Finished
[00:03:43] Process terminated. Status: Failure

---=======================
Warnings:          315
Critical Warnings:  1
Errors:            6

```

wordimont, 2022-05-21:
@stackprogramer I'm not able to reproduce your "Original error example". Please ensure you are using the unmodified example. If you have been experimenting with changes, then I suggest you revert the changes or download a fresh copy of the example to try it again.

Your example shows this;
```
/home/sp/Documents/uhd-4.1.0.5/fpga     -y /home/sp/Documents/rfnoc-example/icores/x310_rfnoc_image_core.yml  --vivado-path=/home/sp/xilinx/Vivado
```

But I don't see what command you ran. Can you share the whole command line you ran to cause the original error?

Can you also run that command with `-l debug` and share the output?

stackprogramer, 2022-05-21:
Hi, Thank you very much. but I am sure that file is not edited....In the below link is respond me that rfnoc-exampe does not have any ce-clk.

https://lists.ettus.com/empathy/thread/7635SR5K7QXNVV3KBUPRWY76S42KJAFV

wordimont, 2022-05-21:
The gain RFNoC block example does not have ce_clk, but the x310 has other RFNoC blocks that do. But all the ce_clk ports in the example code are assigned, so you should not be getting the unresolved clock domain error unless one of the files you are using is different form the original provided in the example.

For example, when you ran the image builder, the info message says it used `/usr/local/share/uhd/rfnoc/blocks/gain.yml`. But that is not the location of `gain.yml` from the example code. Try running the build like this and see if you get the same error:
```
source <path_to_repo>/uhd/fpga/usrp3/top/x300/setupenv.sh
cd <path_to_repo>/uhd/host/examples/rfnoc-example
rfnoc_image_builder -y ./icores/x310_rfnoc_image_core.yml -I ./ --generate-only -l debug
```
The `-I` option tells it where to find the OOT example. Make sure it uses the right gain.yml file.

stackprogramer, 2022-05-22:
I did it but it can not build I faced with strange errors.
```
rfnoc_image_builder -F /home/sp/Documents/uhd-4.1.0.5/fpga  -y ./icores/x310_rfnoc_image_core.yml   --vivado-path=/home/sp/xilinx/Vivado  -I ./ --generate-only -l debug 
[DEB] Load configuration ./icores/x310_rfnoc_image_core.yml.
[DEB] Configuration successful loaded.
[DEB] Validate against schema rfnoc_imagebuilder_args.
[DEB] Using schema file /usr/local/share/uhd/rfnoc/core/rfnoc_imagebuilder_args.json.
[DEB] Configuration successful validated.
[INF] Using FPGA directory /home/sp/Documents/uhd-4.1.0.5/fpga
[INF] Selected device x310
[DEB] Image core output file: /home/sp/.local/share/Trash/files/rfnoc-example/icores/x310_rfnoc_image_core.v
[DEB] Image core header output file: /home/sp/.local/share/Trash/files/rfnoc-example/icores/x310_rfnoc_image_core.vh
[DEB] Edge output file: /home/sp/.local/share/Trash/files/rfnoc-example/icores/x310_static_router.hex
[INF] Using io_signatures.yml from /usr/local/share/uhd/rfnoc/core.
[INF] Using x310_bsp.yml from /usr/local/share/uhd/rfnoc/core.
[DEB] Looking for block descriptors in:
[DEB]     /usr/local/share/uhd/rfnoc/blocks
[DEB]     blocks
[INF] Adding block description from ddc.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from fft_1x64.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from window.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from keep_one_in_n.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from split_stream.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from null_src_sink.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from fosphor.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from siggen.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from radio_1x64.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from fir_filter.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from switchboard.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from correlate.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from axi_ram_fifo_4x64.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from duc.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from addsub.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from axi_ram_fifo_2x64.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from axi_ram_fifo.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from radio.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from gain.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from logpwr.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from replay.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from radio_2x64.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from moving_avg.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from vector_iir.yml (/usr/local/share/uhd/rfnoc/blocks).
[INF] Adding block description from gain.yml (blocks).
[INF] Writing static routing table to /home/sp/.local/share/Trash/files/rfnoc-example/icores/x310_static_router.hex
[DEB] ep0-out0 (1,0) => duc0-in_0 (6,0)
[DEB] duc0-out_0 (6,0) => radio0-in_0 (8,0)
[DEB] radio0-out_0 (8,0) => ddc0-in_0 (7,0)
[DEB] ddc0-out_0 (7,0) => ep0-in0 (1,0)
[DEB] radio0-out_1 (8,1) => ddc0-in_1 (7,1)
[DEB] ddc0-out_1 (7,1) => ep1-in0 (2,0)
[DEB] ep2-out0 (3,0) => duc1-in_0 (9,0)
[DEB] duc1-out_0 (9,0) => radio1-in_0 (11,0)
[DEB] radio1-out_0 (11,0) => ddc1-in_0 (10,0)
[DEB] ddc1-out_0 (10,0) => ep2-in0 (3,0)
[DEB] radio1-out_1 (11,1) => ddc1-in_1 (10,1)
[DEB] ddc1-out_1 (10,1) => ep3-in0 (4,0)
[DEB] ep4-out0 (5,0) => gain0-in (12,0)
[DEB] gain0-out (12,0) => ep4-in0 (5,0)
[INF] Writing image core to /home/sp/.local/share/Trash/files/rfnoc-example/icores/x310_rfnoc_image_core.v
[INF] Writing image core header to /home/sp/.local/share/Trash/files/rfnoc-example/icores/x310_rfnoc_image_core.vh
[INF] Skip build (generate only option given)
```

stackprogramer, 2022-05-22:
Even though I changed image core USRP and I added RFNOC block to USRP,
without custom block, error same is in block,So block is not work in
Gnuradio:I attached image core file
gr::log :DEBUG: rfnoc_rx_streamer0 - Committing graph...
[WARNING] [0/Radio#0] Attempting to set tick rate to 0. Skipping.
gr::log :DEBUG: rfnoc_rx_streamer0 - Sending start stream command...
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOOO

see this link.https://lists.ettus.com/empathy/thread/6J4H6LQ5JW5AQESLS4XKJZS23YBUOFU7

wordimont, 2022-05-22:
Great, it sounds like your original issue of unresolved clock error is resolved. You may want to remove the custom gain.yml from your UHD installation so that it doesn't get used by accident in the future.

closed 2022-05-22 by wordimont

stackprogramer, 2022-05-23:
But I removed it and I installed gain.yml again this not solve my problem...

wordimont, 2022-05-23:
Your current problem sounds like a different one from the one you described in this issue. We use GitHub Issues for reporting bugs in the Ettus code, and we only want one bug per GitHub issue.

This second problem doesn't sound like a bug in the Ettus code and is more likely something wrong with the code you wrote or changes you made to the examples. We don't have the resources to debug your applications for you. If you have a specific question, I suggest you post it on the mailing list. If you think you have found a bug in the Ettus code, and you can explain exactly how to reproduce it, then you can create a new issue.