EttusResearch/uhd #600: Update Gen3 USRP FPGA Build Instructions with more information

zminot, 2022-05-26:
## Issue Description ##
Ran through the x410 (x400) setup recently, and there were a couple things that I felt could be added to the documentation to help devs get through the process.

One of the main things was that the x410 actually requires 2019.1.1_AR73068, not just 2019.1. The 2019.1.1 is pretty self-explanatory, but finding that patch for AR73068 was a bit construed and we could just directly link it there ([link to the patch](https://support.xilinx.com/s/article/73068?language=en_US)).

Also, one needs to make sure they install RFSoC XCZU28DR when installing Xilinx as it does not come by default (you have to select it).

Might want to also include which operating systems are supported by the required versions of Xilinx (not necessary).

Not necessary for it to be specifically Python 3.5, but Python 3.9+ (afaik).

I think as well the documentation seems like it needs a reorganization as it constantly switches between Xilinx ISE and Vivado. Realistically, I think it would be easier to read if you found which program you needed to use via the SDR you're working with, then find the operating system you are planning to user, and then finally you get all of the directions for it.

## Setup Details ##
Open https://github.com/EttusResearch/uhd/blob/master/fpga/docs/usrp3/build_instructions.md

## Expected Behavior ##
To include the above.

## Actual Behaviour ##
It isn't haha.

## Steps to reproduce the problem ##
Read ig

## Additional Information ##
Planning on making a pull request to update this a bit. Let me know if we should include anything else.

Also, would making this pull request sync [this documentation page](https://files.ettus.com/manual/md_usrp3_build_instructions.html)?

wordimont, 2022-05-26:
@zminot Thanks for the feedback.

I agree regarding the AR and which families to install.

Regarding Python version, I think it should just say `Python 3.5+` instead of just `Python 3.5`, right? Or are you saying 3.9 is the minimum?

I'm not sure how to fix the "switches between Xilinx ISE and Vivado" issue. Maybe we could put the instructions for the ISE-based devices in a separate page.

Yes, when we merge a pull request for the manual, the online manual automatically updates. Automation is great. 😄

wordimont, 2022-08-26:
Fixed by https://github.com/EttusResearch/uhddev/pull/6198

closed 2022-08-26 by wordimont