EttusResearch/uhd #604: How does USRP B210 control the tx speed on the host pc

vvverilog, 2022-06-18:
## Issue Description ##

I have design a amateur radio board,  which is a is a combination of AD9361+ARTIX7 + 1G Ethernet.
I designed the fpga project based on the usrp b205mini project, changed the FX3 interface into udp  to communicate with host uhd driver.
I have been able to use uhd_usrp_probe  to send control and response packet over udp, and the uhd_usrp_probe works fine for me, it can print the radio device info.

while I start to send radio data to my board over udp, I found the ethernet speed of tx  reached almost 120MB/s. This udp tx speed is fixed no matter how I changed the tx sample rate. Even though I have mapped my 256MB DDR into a huge fifo for tx chain, the fifo will overflow due to the ethernet speed is too fast, I hope you can tell me how the USRP B210 limits the tx rate, the tx rate should be 4 times the sampling rate in 1R1T mode I think.

By the way, the rx works fine for me, and the rx udp rate is 4 times  to the samp rate. I can using gnu radio to receive radio signal.

## Setup Details ##

UHD version 3.15-LTS
FPGA project based on  USRP B205mini
transport data between host and fpga using udp_zero_copy like below code segment.
```c
...
data_xport_args["recv_frame_size"] = std::to_string(recv_frame_size);
        data_xport_args["num_recv_frames"] = device_addr.get("num_recv_frames", "32");
        data_xport_args["send_frame_size"] = device_addr.get(
            "send_frame_size", std::to_string(B200_USB_DATA_DEFAULT_FRAME_SIZE));
        data_xport_args["num_send_frames"] = device_addr.get("num_send_frames", "32");

        udp_zero_copy::buff_params ignored_out_params;
        device_addr_t fi_hints;
        _data_transport = udp_zero_copy::make(
            addr, BOOST_STRINGIZE(UDP_DATA_PORT), default_buff_args,
            ignored_out_params, fi_hints);
        while (_data_transport->get_recv_buff(0.0)) {
        } // flush ctrl xport
        _demux = recv_packet_demuxer_3000::make(_data_transport);

        ////////////////////////////////////////////////////////////////////
        // create time and clock control objects
        ////////////////////////////////////////////////////////////////////
        _spi_iface = b200_local_spi_core::make(_local_ctrl);
        if (not(_product == B200MINI or _product == B205MINI)) {
            _adf4001_iface = boost::make_shared<b200_ref_pll_ctrl>(_spi_iface);
        }
....
```

wordimont, 2022-06-20:
Hi @vvverilog. You need flow control. On b205mini this is handled by the FX3 chip in combination with the FIFO mode it uses. You can read up on the FX3 and take a look at gpif2_slave_fifo32.v to understand how it works. Flow control is built into USB and the FX3 chip.

When using UDP, there's no flow control built in. Our other network-based USRPs (300 series) use a higher-level protocol called CHDR that runs over UDP to implement flow control. You need to have some kind of flow control mechanism for TX or else the host will inevitably send too fast or too slow relative to the radio's constant clock rate. RX is easier because the host just needs to be faster than the data rate coming from the USRP.

black-pigeon, 2022-06-20:
Thanks for your reply. It seems like that the b205mini is also using CHDR packet. while I am using the uhd 3.15 LTS branch, I found a board called N230, but I can't find any information for this board. The driver for N230 is also based on udp, is it possible for me to implement flow control based on this n230's interface?

vvverilog, 2022-06-20:
Thanks for your reply. It seems like that the b205mini is also using CHDR packet. while I am using the uhd 3.15 LTS branch, I found a board called N230, but I can't find any information for this board. The driver for N230 is also based on udp, is it possible for me to implement flow control based on this n230's interface?

wordimont, 2022-06-21:
I think it is possible. You can look at the N230 code to see how it does it. You have to have both the driver side and the FPGA side. The N230 was never released, which is probably why you're having trouble finding information on it.

vvverilog, 2022-06-21:
Thanks for your sharing, I will try this method.