EttusResearch/uhd #609: Custom RFNoC Block to add metadata results in no packets sent to UHD

zminot, 2022-07-11:
## Issue Description ##

I've been trying for a bit now to add a custom RFNoC block for the x410 that simply just overwrites the metadata field in the incoming CHDR packet to a specific value (in the shared code, it's `0xDEADBEEFDEADC0DE` for simplicity). I integrated a similar testing schema as the addsub testing bench so I'm pretty sure the block functions correctly.

However, when actually adding the add_metadata block to a core yml file and creating the image, UHD sends an error of overflow (the "O" to stdout) when running the example rx_samples_to_file. When looking at Wireshark, I simply receive no packets at all from the x410.

I have been following the directions stated in the [RFNoC UHD 4.0 Guide](https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0#:~:text=Available%20RFNoC%20Blocks%20%20%20%20Block%20Name,Digital%20Up%20Converter%20%2014%20more%20rows%20).

On a similar note, on page 14 of the [RFNoC specification](https://files.ettus.com/app_notes/RFNoC_Specification.pdf), it says that "The framework will provide the ability for the user logic to extract and insert metadata into a packet...", but I haven't found any such tool as of yet. If there already the ability to do so in the PL of the FPGA, then I can likely just use that instead of adding this.

[Here is a GitHub Gist with the block files and the image core yml file I used to generate the file.](https://gist.github.com/zminot/8810545721c7d36f61ff72c02fe1c946) I know GitHub issues aren't really for debugging custom additions, but since I believe the custom code is correct and the RFNoC specification says there is an ability to do so, I thought to make this an issue. Let me know if there is anything else I should add!

## Setup Details ##

Ubuntu 20.04.4 LTS 64-bit
UHD 56f3aab (was just the latest commit at the time of installation)
USRP x410
Connection to the device is via the Ethernet port.

## Expected Behavior ##

Adding the custom RFNoC block to add metadata will send packets with the added metadata to the host machine.

## Actual Behaviour ##

No packets are sent at all, and UHD gives the message of overflow ("O").

## Steps to reproduce the problem ##

Add the custom RFNoC block shared in the gist as directed within the [RFNoC UHD 4.0 Guide](https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0#:~:text=Available%20RFNoC%20Blocks%20%20%20%20Block%20Name,Digital%20Up%20Converter%20%2014%20more%20rows%20).
Make sure all CMakeList files are updated with the new files if necessary.
To run the tests, it should be `make rfnoc_block_add_metadata_tb`
To generate the image core, it should be `make x410_200_rfnoc_image_core_metadata` after adding the yml file to `icores`.
Then load the image core to the fpga using `uhd_image_loader` and the appropriate arguments as described in the RFNoC UHD 4/0 Guide.

## Additional Information ##

Maybe there is something I'm not testing for?

wordimont, 2022-07-11:
Hi @zminot. First, quick disclaimer. The metadata feature isn't fully supported in UHD. Because the software doesn't fully support it yet, it hasn't received much testing. That being said, I'm not aware of a reason why you'd be having this issue.

zminot, 2022-07-11:
Should be okay from my front, I only want to deal with metadata from within the FPGA RFNoC pipeline and don't have a need for it within the UHD app.

If there is any way to monitor the signals within the FPGA during a recv call, I can definitely do so and analyze that too.