EttusResearch/uhd #620: DSP frequency offset not working with timed command

adriengit, 2022-08-11:
## Issue Description ##
DSP frequency offset not working with timed command.

The need is to change tx DSP freq only at a specific time.

## Setup Details ##

Performing uhd_find_devices, one can get the following setup details:

[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; DPDK_21.11; UHD_4.3.0.0-79-g56f3aab2
--------------------------------------------------
-- UHD Device 0
--------------------------------------------------
Device Address:
    serial: 3166XXX
    addr: 192.168.140.2
    fpga: HG
    name: 
    product: X310
    type: x300

more precisely it is a NI USRP 2944R (X310 + UBX).

The PC is a Zbook with ubuntu 22.04.

## Expected Behavior ##

When using timed command with a command to change dsp freq (using manual tune request policy), it is expected to change the frequency at the time specified by the timed command (or executed just after the reception of the command in case of late command).

## Actual Behaviour ##

The RF frequency do no change. tune_result provide wrong results about the actual frequency.
 
## Steps to reproduce the problem ##

I provide a python3 script to reproduce the issue. Tests with cpp did not work either.

```python3
#!/usr/bin/env python3
import uhd
import numpy as np

CHANNEL = 1

U = uhd.usrp.MultiUSRP("")
U.set_tx_rate(1e6)
U.set_tx_freq(uhd.types.TuneRequest(400e6, 0e6), CHANNEL)
U.set_tx_gain(10, CHANNEL)
stream_arg = uhd.usrp.StreamArgs("fc32", "sc16")
stream_arg.channels = [CHANNEL]
tx_stream = U.get_tx_stream(stream_arg)
md = uhd.types.TXMetadata()
md.has_time_spec = True
t0 = U.get_time_now() + 0.5
md.time_spec = t0

spb = 10000
n = 500
duration = (spb*n)/U.get_tx_rate()
print("t0:", t0.get_real_secs())
print("burst duration (s)", duration) 
repetition = 2
dsp_freq = [-10e6, 5e6, 10e6, 20e6] * repetition

n_freq = len(dsp_freq)
for j in range(n_freq):
    for i in range(n):
        nb_send = tx_stream.send(np.ones((spb,), dtype=np.complex64), md)
        if i==0:
            md.start_of_burst = False
            md.has_time_spec = False
        if i==(n-2) and j==(n_freq-1):
            md.end_of_burst = True
    U.set_command_time(U.get_time_now()+0.5)
    tr = uhd.types.TuneRequest()
    tr.rf_freq_policy = uhd.types.TuneRequestPolicy.none
    tr.dsp_freq_policy = uhd.types.TuneRequestPolicy.manual
    tr.rf_freq = 400e6
    tr.dsp_freq = dsp_freq[j]
    #U.set_tx_freq(400e6+dsp_freq[j],CHANNEL) # working correctly
    U.set_tx_freq(tr, CHANNEL) # not changing frequency !!
```
The last 2 lines need to be commented or not depending on the case.

## Additional Information #

This test requires a spectrum analyzer to check the actual transmit frequency.

The tune_result struct do not provide the correct information, i.e. tune_result indicates that frequency switched but checks with spectrum analyser proved the other way.

mbr0wn, 2025-02-06:
Thanks for the excellent bug report. Can confirm this issue.

mbr0wn, 2025-02-10:
Good news -- it's not actually broken. The downside is: This could definitely be more user friendly.

If you update the timestamp on your `send()` call, it works fine:

```python
rate = U.get_tx_rate()
n_freq = len(dsp_freq)
for j in range(n_freq):
    for i in range(n):
        nb_send = tx_stream.send(np.ones((spb,), dtype=np.complex64), md)
        if i==0:
            md.start_of_burst = False
        if i==(n-2) and j==(n_freq-1):
            md.end_of_burst = True
        md.time_spec += spb/rate
    U.set_command_time(U.get_time_now()+0.5)
    tr = uhd.types.TuneRequest()
    tr.rf_freq_policy = uhd.types.TuneRequestPolicy.none
    tr.dsp_freq_policy = uhd.types.TuneRequestPolicy.manual
    tr.rf_freq = 400e6
    tr.dsp_freq = dsp_freq[j]
    #U.set_tx_freq(400e6+dsp_freq[j],CHANNEL) # working correctly
    U.set_tx_freq(tr, CHANNEL) # not changing frequency !!
```

The reason is a bit complicated. Essentially, the DUC tracks timestamps based on timestamps, but if it gets a packet with no time, it resets its internal counter. This way, you will queue up the timed tunes, but you will never reach the time.

closed 2025-02-10 by mbr0wn