EttusResearch/uhd #628: UHD custom Python install path includes `local` on Ubuntu 22.04

jdemel, 2022-09-02:
## Issue Description ##

If i use a custom install path, e.g. `~prefix/gnuradio`, the Python install path ends up being:
```
~/prefix/gnuradio/local/lib/python3.10/site-packages/uhd/
~/prefix/gnuradio/local/lib/python3.10/dist-packages/usrp_mpm/
```
The `local` part should not be there.
## Setup Details ##

I use the latest UHD-4.2 branch. I try to use a custom prefix and I'm on Ubuntu 22.04.

## Expected Behavior ##

all Python files should end up in these folders:
```
~/prefix/gnuradio/lib/python3.10/site-packages/uhd/
~/prefix/gnuradio/lib/python3.10/dist-packages/usrp_mpm/
```

## Actual Behaviour ##

Incorrect install path:
```
~/prefix/gnuradio/local/lib/python3.10/site-packages/uhd/
~/prefix/gnuradio/local/lib/python3.10/dist-packages/usrp_mpm/
```

## Steps to reproduce the problem ##

```bash
git clone .....
cd uhd/host 
mkdir build
cd build
`cmake -DCMAKE_INSTALL_PREFIX=~/prefix/gnuradio ..`
make -j ...
make install
```

## Additional Information ##

One place to modify would be:

https://github.com/EttusResearch/uhd/blob/dfc7b31cfb9a5a001f0786a1ad963b93aad871a6/host/python/CMakeLists.txt#L166-L173

This fixed it:
```cmake
        execute_process(COMMAND ${PYTHON_EXECUTABLE} -c
            "import os,sysconfig;\
            platlib = sysconfig.get_path('platlib','posix_prefix');\
            prefix = sysconfig.get_path('data','posix_prefix');\
            print(os.path.relpath(platlib, prefix));"
            OUTPUT_VARIABLE UHD_PYTHON_DIR OUTPUT_STRIP_TRAILING_WHITESPACE
        )
```

mbr0wn, 2023-01-09:
@jdemel I think we're going with your suggestion for now, although I think that there might be other install schemes viable for some other scenarios. Thanks for pointing that out!

mbr0wn, 2023-02-27:
FYI, we're having issues making this solution work for embedded builds. Hang in there!

closed 2023-04-03 by michael-west

jdemel, 2023-04-04:
Sounds like good news that this issue is "completed". Though, is there a commit with this change?

mbr0wn, 2023-04-05:
There is! Same story as usual. It takes a few days for the commits to go public.

mbr0wn, 2023-05-11:
c385d20eeea717b3859ac6a2bcc247b69fc66003