EttusResearch/uhd #631: error importing uhd into python broke a perfect install somehow!!

alexxmarsh44, 2022-09-20:
## Issue Description ##
I have been using the python UHD API for a solid couple of years with no problems.  I am running Arch Linux and installed uhd from the main package repository.  I have been coding with python in a python ide called pycharm.  Basically I am running python in a virtual environment that mirrors the site packages directory from the python system directory.  All is well and good until today.  I installed a package called gpsd a package I have used many times before.  It instantly breaks my uhd import.  Here is the following screenshot.  
[image]
Any idea why this sudden error?  I really am in trouble here I was on the verge of finishing a huge direction finding project when this happened.  please help!

## Setup Details ##
usrp b210, libuhd latest version from pacman repo, and I use a copy of uhd from the site packages python3 directory and copy that to my virtual environment.

## Expected Behavior ##
clean import of uhd

## Actual Behaviour ##
throws error

## Steps to reproduce the problem ##
import uhd 

## Additional Information ##

alexxmarsh44, 2022-09-20:
Also it appears that the missing file has a file by the same name but with version 80 instead of 79

mbr0wn, 2023-07-12:
Sorry, can't reproduce. I'm guessing this must be specific to your setup. Maybe some .so couldn't be found.

closed 2023-07-12 by mbr0wn