EttusResearch/uhd #635: rfnoc-example FPGA build fails in UHD-4.3

ptrkrysik, 2022-10-18:
## Issue Description ##
Build of rfnoc-example (uhd/host/examples/rfnoc-example) fails for UHD-4.3 branch.
It fails on building of cmplx_mul ip block that is in the tree of the example. Probably uhd/host/examples/rfnoc-example/fpga/ip/cmplx_mul/cmplx_mul.xci wasn't updated after migration from Vivado 2019.1 to 2021.1 as Vivado reports that it is locked.

## Setup Details ##
Tip of UHD-4.3 branch and Vivado 2021.1.

## Expected Behavior ##

FPGA bitstream for the rfnoc-example should be built correctly.

## Actual Behaviour ##

## Steps to reproduce the problem ##
`
git clone https://github.com/EttusResearch/uhd.git
cd uhd/host/examples/rfnoc-example
rfnoc_image_builder -F $(pwd)/../../../fpga -y $(pwd)/icores/x310_rfnoc_image_core.yml -I $(pwd)
`
## Additional Information ##

Tail of log with failure to build cmplx_mul:

BUILDER: Releasing IP location: /home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/dds_sin_cos_lut_only
========================================================
BUILDER: Building IP cmplx_mul
========================================================
BUILDER: Staging IP in build directory...
BUILDER: Reserving IP location: /home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul
BUILDER: Retargeting IP to part kintex7/xc7k410t/ffg900/-2...
BUILDER: Building IP...
[00:00:00] Executing command: vivado -mode batch -source /home/inusx/uhd_4.3/fpga/usrp3/tools/scripts/viv_generate_ip.tcl -log cmplx_mul.log -nojournal
WARNING: Default location for XILINX_HLS not found
[00:00:05] Current task: Initialization +++ Current Phase: Starting
WARNING: [IP_Flow 19-2162] IP 'cmplx_mul' is locked:
CRITICAL WARNING: [filemgmt 20-1366] Unable to reset target(s) for the following file is locked: /home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xci
CRITICAL WARNING: [filemgmt 20-1365] Unable to generate target(s) for the following file is locked: /home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xci
[00:00:05] Current task: Initialization +++ Current Phase: Finished
[00:00:05] Executing Tcl: synth_design -top cmplx_mul -part xc7k410tffg900-2 -mode out_of_context
[00:00:05] Starting Synthesis Command
WARNING: [Vivado_Tcl 4-391] The following IPs are missing output products for Synthesis target. These output products could be required for synthesis, please generate the output products using the generate_target or synth_ip command before running synth_design.
WARNING: [IP_Flow 19-2162] IP 'cmplx_mul' is locked:
ERROR: [Designutils 20-414] HRTInvokeSpec : No Verilog or VHDL sources specified
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
ERROR: [Common 17-53] User Exception: No open design. Please open an elaborated, synthesized or implemented design before executing this command.
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xml'
CRITICAL WARNING: [IP_Flow 19-4739] Writing uncustomized BOM file '/home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul/cmplx_mul.xml'
ERROR: [Vivado 12-398] No designs are open
[00:00:27] Current task: Synthesis +++ Current Phase: Starting
[00:00:27] Current task: Synthesis +++ Current Phase: Finished
[00:00:27] Process terminated. Status: Failure

========================================================
Warnings:           4
Critical Warnings:  7
Errors:             8

BUILDER: Releasing IP location: /home/inusx/uhd_4.3/fpga/usrp3/top/x300/build-ip/xc7k410tffg900-2/cmplx_mul
make[1]: *** [/home/inusx/uhd_4.3/host/examples/rfnoc-example/fpga//ip/cmplx_mul/Makefile.inc:21: LIB_IP_CMPLX_MUL_TRGT] Błąd 1

wordimont, 2022-10-19:
@ptrkrysik Thanks for reporting this! I've submitted a pull request to fix this issue. Here's a patch, in case that helps.

[upgrade_cmplx_mul_2021.1.patch.txt](https://github.com/EttusResearch/uhd/files/9816387/upgrade_cmplx_mul_2021.1.patch.txt)

closed 2022-10-20 by skooNI

skooNI, 2022-10-20:
@wordimont's patch is merged into master:
https://github.com/EttusResearch/uhd/commit/1e718d9be88e2b69f6aacecc6e9a31310f35621d