EttusResearch/uhd #643: uhd.lib cannot be opened

Slaih, 2022-10-26:
## lib\Release\uhd.lib entrance file cannot be opened ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

manderseck, 2022-11-08:
I cannot reproduce the issue here on Win10 and VS2019. @Slaih Please provide more information, otherwise I'll close this issue. Thanks!

Slaih, 2022-11-08:
I solved this issue by adding include directories to vs2022 project properties. Thanks your consideration.

Paneran, 2024-02-12:
Hi, I am also going through this issue, can you elaborate how you fixed it? @Slaih

Slaih, 2024-02-12:
Hello,

I actually gave up with setting uhd at windows because it is originally developed for linux

mkoop-ni, 2024-02-13:
> Hi, I am also going through this issue, can you elaborate how you fixed it? @Slaih

Unfortunately, I can not read the error message from the original problem description. I would recommend you follow the instructions in the KB '[Building and Installing the USRP Open Source Toolchain (UHD and GNU Radio) on Windows](https://kb.ettus.com/Building_and_Installing_the_USRP_Open_Source_Toolchain_(UHD_and_GNU_Radio)_on_Windows)', even so the listed dependency versions are little outdated. Using these instructions, I was successful able to build UHD while using:
- Windows 10
- Visual Studio 2022
- cmake 3.28.1
- boost 1.83.0 (using the precompiled installer msvc-14.3-64.exe)
- libusb 1.0.22
- python 3.12.1
- doxygen 1.8.14
- (optionally) nsis 3.09.

Note: 
- Enable the ENABLE_PYTHON_API box in the cmake_gui application to optionally also build the UHD python API 
- To use the python API please see info in [UHD Manual>>Installing the Python API](https://files.ettus.com/manual/page_python.html#python_install_windows).

If you run into problems when executing the outlined steps, either create a new issue or consider other available support options outlined at https://www.ettus.com/support/.

Paneran, 2024-02-13:
I was able to solve this issue. To revisit the original problem: @mkoop-ni . and to log for others who come across the same problem. 
[image]

This error occurred because I used precompiled installer msvc-14.0-64.exe, when using Visual Studio 2022. Visual studio 2022 is only compatible with msvc-143.