EttusResearch/uhd #645: UHD C API unexpectedly invalidates handles on error

mgw93, 2022-11-05:
## Issue Description ##
The UHD C API seems to invalidate passed handles when encountering an error.

## Setup Details ##
I am trying to write a wrapper around the UHD API in the Julia language. The current effort can be found [here](https://github.com/mgw93/UHDAPI)
As interfacing with C++ is more complicated, I’m using the C-API. While testing my code, I encountered unexpected crashes.
They occured when I tried to acces non-existing sensor-values for the purpose of testing my error handling.

## Expected Behavior ##
The code
> uhd_sensor_value_handle sensor_handle;
> uhd_sensor_value_make(&sensor_handle);
> uhd_usrp_handle usrp_handle;
> uhd_usrp_make(&usrp_handle,"");
> uhd_usrp_get_mboard_sensor(usrp_handle,"nonexisting",0,&sensor_handle); // Fails and gives a UHD_ERROR_LOOKUP:
 "LookupError: Path not found in tree: /mboards/0/sensors/auebd" error message when queried
> uhd_usrp_sensor_value_free(&sensor_handle);
> uhd_usrp_free(&usrp_handle);

should not cause any memory corruption.

## Actual Behaviour ##
The above mentioned code causes a double free. The internal C++ object is freed before the new one is constructed.
When the construction fails, the handle is left invalid.

## Steps to reproduce the problem ##
See code above.

## Additional Information ##
This might happen with other getter functions too. I just happened to find it with the get_mboard_sensor.
The offending code for this particular instance seems to be in [host/lib/usrp/usrp_c.cpp](https://github.com/EttusResearch/uhd/blob/master/host/lib/usrp/usrp_c.cpp) on line 481.

mbr0wn, 2022-11-09:
Yup, confirmed. We have a fix in the pipeline.

closed 2022-11-11 by skooNI

mbr0wn, 2022-11-17:
f82a19af9