EttusResearch/uhd #646: Incorrect Python Directory for UHD install

dylan-baros, 2022-11-19:
## Issue Description ##

Seems to be a path issue when installing UHD with Python support on Ubuntu 22.04.1. The installation directory is now incorrect. 

Short example snippet of installation logs:

Before with 20.04-> 
``` 
-- Installing: /usr/local/lib/python3.8/site-packages/usrp_mpm/sys_utils/__init__.py
-- Installing: /usr/local/lib/python3.8/site-packages/usrp_mpm/sys_utils/gpio.py 
```
After with 22.04.1->
```
-- Installing: /usr/local/local/lib/python3.10/dist-packages/usrp_mpm/sys_utils/__init__.py
-- Installing: /usr/local/local/lib/python3.10/dist-packages/usrp_mpm/sys_utils/gpio.py
```

As you can see, /local/local/ should just be /local/.

## Setup Details ##

x410 Running UHD 4.3
Ubuntu 22.04.1
Git commit: 6c647732253e5ce27f840b472fcdeb9a0ae3b648

## Expected Behavior ##

I expect the local directory to not be copied twice. 

## Actual Behaviour ##

/local/ becomes /local/local/

## Steps to reproduce the problem ##

Install UHD from commit above with Python cmake flag. 

## Additional Information ##

dylan-baros, 2022-11-19:
@michaelld

michaelld, 2022-11-21:
@dylan-baros what is the full `cmake` command you're executing?

dylan-baros, 2022-11-21:
> @dylan-baros what is the full `cmake` command you're executing?

cmake .. -DENABLE_PYTHON_API=ON

LawsonGu, 2022-11-29:
I have switch to python3.9 in ubuntu 22.04.1
CMake command:
```
cmake -DPYTHON_INCLUDE_DIR=/usr/include/python3.9 -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-
gnu/libpython3.9.so -DENABLE_PYTHON_API=ON ..
```
Result:
[image]

jdemel, 2022-12-19:
@dylan-baros #628 should describe the same issue. I'm glad I'm not the only one stumbling over this.

mbr0wn, 2023-01-09:
Dupe of #628. Closing this.

closed 2023-01-09 by mbr0wn