EttusResearch/uhd #650: Use of external GPSDO with B200. Not working :(

tiger762, 2022-12-22:
Hello!

Intel i5-7500k, Centos 8, Ettus B200, UHD 3.15.0, srsRAN 21.4.0, Tektronix TDS2012 oscilloscope

Using an application (pdsch_ue, part of srsRAN 21.4.0), with a 10MHz (1 volt amplitude sine wave) external clock source, device args passed through to the B200: "clock_source=external", locked on to Verizon's EnodeB (751MHz, LTE Band 13)

I was expecting the clock frequency offset to be single-digit Hertz. Instead, it is the usual 2000 to 3000 Hertz it is even without the GPSDO attached.  Given the UHD version, the board type, the particulars of the sine wave clock source (0.71 volts RMS, 10mw across 50 ohm load), what could be the problem?

How do I go about troubleshooting this further? Thanks!

tiger762, 2023-01-13:
> @tiger762 I strongly doubt that the issue here is with UHD itself, so I'm closing this issue. Please send an email to ***@***.***" if you haven't already; if you have, reply to it & tag me on it & I'll reply from there

Hi Michael,
It turned out that the correct device args to use were "clock=external". When I finally figured out that "clock_source=external" was wrong for either my board (B200) and/or my version of UHD (3.15.0)
When I used the correct args, the CFO dropped from 2500 Hz to plus/minus 20 Hz. It was a dramatic improvement. I submit that there is conflicting information floating around the Internet about what args to use. I write this in the hopes that someone searching the archives will find it useful :)

michaelld, 2023-01-13:
@tiger762 interesting ... thanks for the followup info & I'm glad you worked out the correct device args and that they work! It very well could be that the device args for time and clock source changed from UHD 3 to UHD 4 ... lots of things changed!