EttusResearch/uhd #653: rpc::timeout: Timeout 2000ms when using UHD Python API

LawsonGu, 2023-01-06:
## Issue Description ##
When I was using UHD Python API
`import uhd`
`usrp = uhd.usrp.MultiUSRP("addr=192.168.9.2")`
This issue happened

## Setup Details ##
Ubuntu: 20.04
UHD: 4.2
Python: 3.8

## Expected Behavior ##
[image]

## Actual Behaviour ##
[image]

## Steps to reproduce the problem ##
Before you reproduce, please confirm your $home/.bashrc. It should be like this.
[image]

## Additional Information ##
After I change $home/.bashrc file like below, issue disappear and expected behavior goes well.
[image]

Maybe the question is: 
1. Is build and install uhd could change $home/.bashrc file? 
3. If $PYTHONPATH contains 
`/python3/dist-packages`
Seems UHD Python API could not use normally in this case?

wordimont, 2023-01-07:
It looks like get_num_xbars was removed when UHD 4.0 was released. Notice in your actual output the UHD version printed is 3.15, which probably doesn't match your device. Perhaps you have remnants from multiple UHD versions on your system and changing your PYTHONPATH is affecting which one is being used.

mbr0wn, 2024-03-19:
In your logs, you are mixing UHD versions. I expect if you update your devices, you're good to go.

closed 2024-03-19 by mbr0wn